home-assistant/operating-system #597: hassos 3.11 broke external USB SSD

marcgarciamarti, 2020-02-22:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

I upgraded today my hassos from 3.10 to 3.11 and my system has stopped working. When I try to boot up, the system won't even react to ping. I removed the SSD drive and replace it with one other SSD running raspian and ping works just fine, therefore I rule out any hw related problem.
Currently flashing hassos 3.10 hoping to get the system up and running soon. 
My SSD drive is a Crucial BX500 120GB.

niesink, 2020-03-02:
@marcgarciamarti have you seen any progress on this?

I'm currently on 3.9 and hesitant to upgrade because of this.
I've noticed booting-problems on my own system (Pi3B+ booting from a Samsung T5 SSD), which might have something to do with improper unmounting of the SSD before unpowering the Pi. Mounting and unmounting the SSD on a Windows machine seems to 'fix' the SSD, allowing the Pi to boot from it. I'm looking into perhaps using a powered USB-hub to see if this stops the Pi from breaking the SSD everytime it powers down. This theory is mainly based on https://github.com/raspberrypi/firmware/issues/891 

So anyway, I was wondering if this might be what's happening on your end as well or if you've managed to confirm that it really is 3.11 that's causing the issues?

sockmonkey0223, 2020-03-02:
I'm also waiting to hear more info on this before upgrading. Been using an Pi 3B w/ SSD for several months, no issues.

marcgarciamarti, 2020-03-02:
> @marcgarciamarti have you seen any progress on this?
> 
> I'm currently on 3.9 and hesitant to upgrade because of this.
> I've noticed booting-problems on my own system (Pi3B+ booting from a Samsung T5 SSD), which might have something to do with improper unmounting of the SSD before unpowering the Pi. Mounting and unmounting the SSD on a Windows machine seems to 'fix' the SSD, allowing the Pi to boot from it. I'm looking into perhaps using a powered USB-hub to see if this stops the Pi from breaking the SSD everytime it powers down. This theory is mainly based on [raspberrypi/firmware#891](https://github.com/raspberrypi/firmware/issues/891)
> 
> So anyway, I was wondering if this might be what's happening on your end as well or if you've managed to confirm that it really is 3.11 that's causing the issues?

Hi there,
I really can’t tell what the culprit is because I downgraded to 3.10 and I’m still running 3.10 today because of fear to rendering my setup useless again. I always do full backups before doing any major thing, and fortunately I had a snapshot just a few hours old that I used to recover the system more or less quickly (I of course wasted a couple of hours).
I must say that I’m under the impression that I was running 3.11 while some reboots where done. Can’t really tell for sure but I’m more or less positive I did reboot it at least once.
Anyway, as I said it died on me and I had to reinstall 3.10 from scratch. Maybe it has to do with that weird behavior that gets fixed by simply mounting the drive in a PC. 
If somebody could point me in the right direction of what one needs to collect, I could upload my logs somewhere should this happen again, if I can mount the file system to get those logs. But since I could not analyze the logs, I can really say what the culprit was.
@niesik, one need to only plug the SSD, mount, unmount and off you go?

closed 2020-05-09 by stale[bot]