home-assistant/operating-system #598: No GPIO hardware on Raspberry Pi 4

spanishkangaroo, 2020-02-27:
**HassOS release with the issue:**
HassOS 3.11
Home Assistant 0.106.0

**Supervisor logs:**
20-02-27 05:59:30 INFO (MainThread) [__main__] Initialize Hass.io setup
20-02-27 05:59:30 INFO (SyncWorker_0) [hassio.docker.supervisor] Attach to Supervisor homeassistant/armv7-hassio-supervisor with version 201
20-02-27 05:59:30 INFO (MainThread) [__main__] Setup HassIO
20-02-27 05:59:30 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd1
20-02-27 05:59:31 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.hostname1 - /org/freedesktop/hostname1
20-02-27 05:59:31 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: de.pengutronix.rauc - /
20-02-27 05:59:31 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.NetworkManager - /org/freedesktop/NetworkManager/DnsManager
20-02-27 05:59:31 INFO (MainThread) [hassio.host.info] Update local host information
20-02-27 05:59:31 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-02-27 05:59:31 INFO (MainThread) [hassio.host.services] Update service information
20-02-27 05:59:31 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-02-27 05:59:31 INFO (MainThread) [hassio.host.network] Update local network DNS information
20-02-27 05:59:31 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-02-27 05:59:31 INFO (MainThread) [hassio.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
20-02-27 05:59:31 INFO (MainThread) [hassio.host.services] Reload local service hassos-apparmor.service
20-02-27 05:59:31 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1
20-02-27 05:59:31 INFO (SyncWorker_1) [hassio.docker.interface] Attach to homeassistant/armv7-hassio-dns with version 1
20-02-27 05:59:31 INFO (MainThread) [hassio.dns] Start CoreDNS plugin
20-02-27 05:59:31 INFO (MainThread) [hassio.misc.forwarder] Start DNS port forwarding to 172.30.32.3
20-02-27 05:59:31 INFO (SyncWorker_4) [hassio.docker.interface] Clean hassio_dns application
20-02-27 17:25:03 INFO (SyncWorker_4) [hassio.docker.dns] Start DNS homeassistant/armv7-hassio-dns with version 1 - 172.30.32.3
20-02-27 17:25:03 INFO (SyncWorker_3) [hassio.docker.interface] Attach to homeassistant/raspberrypi4-homeassistant with version 0.106.0
20-02-27 17:25:03 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-02-27 17:25:03 INFO (MainThread) [hassio.hassos] Detect HassOS 3.11 / BootSlot B
20-02-27 17:25:03 INFO (SyncWorker_0) [hassio.docker.hassos_cli] Found HassOS CLI homeassistant/armv7-hassio-cli with version 19
20-02-27 17:25:04 INFO (MainThread) [hassio.store.git] Load add-on /data/addons/git/03d23610 repository
20-02-27 17:25:04 INFO (MainThread) [hassio.store.git] Load add-on /data/addons/git/a0d7b954 repository
20-02-27 17:25:04 INFO (MainThread) [hassio.store.git] Load add-on /data/addons/core repository
20-02-27 17:25:04 INFO (MainThread) [hassio.store] Load add-ons from store: 64 all - 64 new - 0 remove
20-02-27 17:25:04 INFO (MainThread) [hassio.addons] Found 10 installed add-ons
20-02-27 17:25:04 INFO (SyncWorker_1) [hassio.docker.interface] Attach to danimtb/armv7-dasshio with version 0.3.8
20-02-27 17:25:04 INFO (SyncWorker_2) [hassio.docker.interface] Attach to hassioaddons/log-viewer-armv7 with version 0.8.0
20-02-27 17:25:04 INFO (SyncWorker_3) [hassio.docker.interface] Attach to homeassistant/armv7-addon-duckdns with version 1.11
20-02-27 17:25:04 INFO (SyncWorker_5) [hassio.docker.interface] Attach to homeassistant/armv7-addon-mosquitto with version 5.1
20-02-27 17:25:04 INFO (SyncWorker_6) [hassio.docker.interface] Attach to homeassistant/armv7-addon-samba with version 9.0
20-02-27 17:25:04 INFO (SyncWorker_5) [hassio.docker.interface] Attach to homeassistant/armv7-addon-configurator with version 4.6
20-02-27 17:25:04 INFO (SyncWorker_11) [hassio.docker.interface] Attach to homeassistant/armv7-addon-ssh with version 8.4
20-02-27 17:25:04 INFO (SyncWorker_0) [hassio.docker.interface] Attach to homeassistant/armv7-addon-letsencrypt with version 4.7
20-02-27 17:25:04 INFO (SyncWorker_4) [hassio.docker.interface] Attach to hassioaddons/motioneye with version 0.6.0
20-02-27 17:25:04 INFO (SyncWorker_2) [hassio.docker.interface] Attach to homeassistant/armv7-addon-check_config with version 3.1
20-02-27 17:25:04 INFO (MainThread) [hassio.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-02-27 17:25:07 INFO (MainThread) [hassio.snapshots] Found 2 snapshot files
20-02-27 17:25:07 INFO (MainThread) [hassio.discovery] Load 1 messages
20-02-27 17:25:07 INFO (MainThread) [hassio.ingress] Load 8 ingress session
20-02-27 17:25:07 INFO (MainThread) [hassio.secrets] Load Home Assistant secrets: 1
20-02-27 17:25:07 INFO (MainThread) [__main__] Run Hass.io
20-02-27 17:25:07 INFO (MainThread) [hassio.api] Start API on 172.30.32.2
20-02-27 17:25:07 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rauc.Installer.Mark on /
20-02-27 17:25:07 INFO (MainThread) [hassio.hassos] Rauc: B - marked slot kernel.1 as good
20-02-27 17:25:07 INFO (MainThread) [hassio.addons] Phase 'initialize' start 0 add-ons
20-02-27 17:25:07 INFO (MainThread) [hassio.addons] Phase 'system' start 1 add-ons
20-02-27 17:25:07 INFO (SyncWorker_15) [hassio.docker.interface] Clean addon_core_mosquitto application
20-02-27 17:25:09 INFO (SyncWorker_15) [hassio.docker.addon] Start Docker add-on homeassistant/armv7-addon-mosquitto with version 5.1
20-02-27 17:25:11 INFO (MainThread) [hassio.services.modules.mqtt] Set core_mosquitto as service provider for mqtt
20-02-27 17:25:14 INFO (MainThread) [hassio.addons] Phase 'services' start 4 add-ons
20-02-27 17:25:14 INFO (SyncWorker_12) [hassio.docker.interface] Clean addon_core_ssh application
20-02-27 17:25:14 INFO (SyncWorker_0) [hassio.docker.interface] Clean addon_core_samba application
20-02-27 17:25:14 INFO (SyncWorker_2) [hassio.docker.interface] Clean addon_core_duckdns application
20-02-27 17:25:14 INFO (SyncWorker_9) [hassio.docker.interface] Clean addon_03d23610_dasshio application
20-02-27 17:25:17 INFO (SyncWorker_9) [hassio.docker.addon] Start Docker add-on danimtb/armv7-dasshio with version 0.3.8
20-02-27 17:25:17 INFO (SyncWorker_0) [hassio.docker.addon] Start Docker add-on homeassistant/armv7-addon-samba with version 9.0
20-02-27 17:25:18 INFO (SyncWorker_2) [hassio.docker.addon] Start Docker add-on homeassistant/armv7-addon-duckdns with version 1.11
20-02-27 17:25:18 INFO (SyncWorker_12) [hassio.docker.addon] Start Docker add-on homeassistant/armv7-addon-ssh with version 8.4
20-02-27 17:25:23 INFO (SyncWorker_15) [hassio.docker.interface] Start homeassistant/raspberrypi4-homeassistant
20-02-27 17:25:40 INFO (MainThread) [hassio.auth] Auth request from core_mosquitto for yi-cam-mqtt
[... 516 lines cut ...]
[  138.818342] device veth2d0a3ef entered promiscuous mode
[  138.818468] audit: type=1700 audit(1582824407.236:157): dev=veth2d0a3ef prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  138.818883] audit: type=1300 audit(1582824407.236:157): arch=40000028 syscall=290 per=800000 success=yes exit=40 a0=c a1=4702480 a2=28 a3=0 items=0 ppid=1 pid=325 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj==unconfined key=(null)
[  138.819295] audit: type=1327 audit(1582824407.236:157): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D73746F726167652D6472697665723D6F7665726C617932002D2D6C6F672D6472697665723D6A6F75726E616C64002D2D646174612D726F6F74002F6D6E742F646174612F646F636B6572
[  138.819779] IPv6: ADDRCONF(NETDEV_UP): veth2d0a3ef: link is not ready
[  138.819799] hassio: port 7(veth2d0a3ef) entered blocking state
[  138.819811] hassio: port 7(veth2d0a3ef) entered forwarding state
[  138.912053] hassio: port 7(veth2d0a3ef) entered disabled state
[  139.406405] audit: type=1325 audit(1582824407.826:158): table=nat family=2 entries=0
[  139.406439] audit: type=1300 audit(1582824407.826:158): arch=40000028 syscall=295 per=800000 success=yes exit=0 a0=4 a1=0 a2=40 a3=bed89a5c items=0 ppid=2363 pid=2386 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid [... cut ...]
[  139.972897] eth0: renamed from vethbd046db
[  140.023437] IPv6: ADDRCONF(NETDEV_CHANGE): veth2d0a3ef: link becomes ready
[  140.023536] hassio: port 7(veth2d0a3ef) entered blocking state
[  140.023548] hassio: port 7(veth2d0a3ef) entered forwarding state

**Description of problem:**
In the Home Assistant's "Supervisor" -> "System" -> "Hardware" screen, my Raspberry Pi 4 GPIOs are not detected at all, even when it seems there is no issue with the GPIO platform on Home Assistant. See attached screenshot.

[image]

sbourgeo, 2020-03-02:
Getting the samething on HassOS 3.10

fjufju, 2020-04-29:
I want to move from Pi3 to Pi4.
This works?

spanishkangaroo, 2020-04-29:
> I want to move from Pi3 to Pi4.
> This works?

No, but I just installed everything with Docker and it works fine.

closed 2020-05-28 by pvizeli