home-assistant/operating-system #599: Unable to flash sd card with latest raspberry images

martinscholz83, 2020-02-28:
**HassOS release with the issue:**
3.11

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
I'm trying to flash sd card with the latest raspberry images. When i insert the card into the raspberry nothing happens. When looking on the sd card it looks like this
[image]
This also happened with 3.10 release.

ptoulouse, 2020-03-06:
Same issue here. Flashed a 64GB  SD with 3.11.  The Pi does not start. No output on HDMI. The Pi is wired to ethernet and does not get an IP from the router. Tested the SD card by flashing Raspbian-lite on it and it boots fine, so not an SD card issue.

ptoulouse, 2020-03-09:
Tried again with 3.12 and I have the same issue.

closed 2020-05-08 by pvizeli