home-assistant/operating-system #600: Versions >3.7 causes system freezes on ASUS TinkerBoard S eMMC installation

LordBoos, 2020-03-02:
**HassOS release with the issue:**
3.8, 3.9, 3.10, 3.11 (anything >3.7)

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
I have ASUS TinkerBoard S with eMMC installation. If I update to any more recent version than HassOS 3.7, everything starts to behave weirdly. Problems I have noticed:
- Updating addons is causing the system to freeze ~50% of times
- Updating Home Assistant is causing the system to freeze ~75% of times
- Restarting Home Assistant is causing the system to freeze ~75% of times
- System freezes after some time. Uptime ranged from about 10 minutes to 48 hours before freezing
- I'm unable to stop or kill containers using docker command in SSH, command just keeps processing indefinitely
- Home Assistant and Supervisor logs show absolutely nothing out of ordinary, Home Assistant log just stops being updated, Supervisor log shows nothing about the problem (only log accessed messages)
- SSH works for some time after the freeze, but after few minutes it also stops working
- Only way to unfreeze the system is to unplug the power supply. Host reboot command also doesn't help

I tested this with HassOS 3.7 updated to 3.8, 3.9, 3.10, 3.11 and even clean installations with no addons and default Home Assistant configuration. Problem occurred every time.
Right now I'm using HassOS 3.7 and it is running without problems.

mbo18, 2020-03-08:
I have this board and I have no issue at all. Could it be defective? Did you test the eMMC?

LordBoos, 2020-03-08:
I did not test it directly, but if the eMMC was defective wouldn't it also affect the versions 3.7 a earlier?

LordBoos, 2020-04-18:
Today I've decided to give updating another shot. So far 3.13 seems stable for me. I¨'ll let it running for some time and if the issue does not reoccur I'll close this issue.

LordBoos, 2020-04-19:
After about 8 hours, the issue reoccurred and I had to downgrade to 3.7 again.

pvizeli, 2020-05-08:
https://github.com/home-assistant/operating-system/issues/650

closed 2020-05-08 by pvizeli