home-assistant/operating-system #605: Intel NUC CPU temperature support

jchasey, 2020-03-08:
When installing HassOS on an Intel NUC there does not currently appear to be a way to access the cpu temperature.

Would the addition of lm_sensors or similar allow this to be supported?

adandl, 2020-04-07:
This works for me - not what you are looking for?
`- platform: command_line
    name: CPU Temp
    command: "cat /sys/class/thermal/thermal_zone2/temp"
    unit_of_measurement: "°C"
    value_template: '{{ value | multiply(0.001) | round(1) }}'`

jchasey, 2020-04-14:
Unfortunately not, the command line sensors are not available, although the temperatures are reported when booting into the BIOS configuration screen.

adandl, 2020-04-18:
Hi,
not sure if I understand exactly what you are looking fore.
For me the command-line sensor works - but I'm only reading one temperature from available three:

Zone0 shows a negative number - that's why I ignore it :)

```
~ $ cat /sys/class/thermal/thermal_zone*/temp
-263200
27800
38000
36000

```

jchasey, 2020-04-18:
Those sensors don't exist on my NUC, although the temperature of the CPU is displayed in the BIOS. So the hardware is there, but HassOS is not picking them up.

adandl, 2020-04-18:
I understand. Have you tried doing a BIOS update of the Nuc. I installed the most recent just a few weeks ago when installed it.

myshiftzz, 2020-05-25:
I am running HassOS in a VirtualBox VM on an Intel NUC running Windows 10 Pro and the sensors don't exist.   I only see cooling_zone0 and cooling_zone1 directories with no data files representing any kind of temperature.

Did anyone find the solution to this?

pattyland, 2020-06-08:
@adandl Thanks for the advice! Working great on my J3455:
```bash
~ cat /sys/class/thermal/thermal_zone*/temp
52000
49000
47000
```
I don't even get weird negative numbers

j9brown, 2020-06-23:
I just went down this rabbit hole.  It seems the only temperature sensor you can read is the CPU temperature.  There aren't any drivers for reading the rest.

At first, I thought that temp2 was the motherboard temperature but it always returns 27.8 °C.

As for getting the data from within a virtual machine, try running snmpd on the host and using the Home Assistant snmp integration to get the values.  I did that just this morning only to find out that the temperature values aren't usable.  :(

closed 2020-08-29 by stale[bot]