home-assistant/operating-system #607: Update fails due to "dirty pages"

hampshire-al, 2020-03-15:
**HassOS release with the issue:**
3.10 - when trying to update to either 3.11 or 3.12

**Supervisor logs:**
20-03-15 11:48:33 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/3.12/hassos_intel-nuc-3.12.raucb
20-03-15 11:48:47 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-3.12.raucb
20-03-15 11:48:47 INFO (MainThread) [supervisor.utils.gdbus] Call de.pengutronix.rauc.Installer.Install on /
20-03-15 11:48:48 INFO (MainThread) [supervisor.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-03-15 11:48:49 INFO (MainThread) [supervisor.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-03-15 11:48:49 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-03-15 11:48:50 ERROR (MainThread) [supervisor.hassos] HassOS update fails with: Failed mounting bundle: failed to run mount: Child process exited with code 32

**Journal logs:**

**Kernel logs:**
loop_set_status: loop0 () has still dirty pages (nrpages=48)
sometimes this says 
loop_set_status: loop0 () has still dirty pages (nrpages=2)

**Description of problem:**
I cannot update beyond Hass.io 3.10 on an Intel NUC.
The same errors have arisen when trying to update to either 3.11 or 3.12.

I am quite a noob at Home Assistant, so I may have missed something!! (Sorry!)
Is there a file that needs deleting etc. that will stop this error and allow the update?

Thanks

lassej01, 2020-03-20:
I restarted my NUC, and that has solved the problem for me twice (different upgrades)

hampshire-al, 2020-03-24:
After many attempts over time the update to 3.12 finally worked yesterday.  I haven't changed anything, so don't know why it has now worked.  Maybe the update to Home Assistant to 0.107.5 helped, but it still was failing as above after that update, so not convinced it was that.

Anyway I'm sorted, but it does seem that these updates are not always going as smoothly as we would like.  Thanks.

pvizeli, 2020-03-31:
Look like that is a kernel bug: https://lkml.org/lkml/2019/5/17/787

I can look if we can use that patch

closed 2020-05-08 by pvizeli

Rtoax, 2021-12-08:
I got the same problem when i booting  a Fedora CoreOS, kernel 4.18.

PR #659 (2020-05-08): Fix issue with loop devices
Fix: #544
Fix: #607