home-assistant/operating-system #610: Home Assistant on Odroid N2 not contactable after network interruption

mhaack, 2020-03-23:
**HassOS release with the issue:**
HassOS 4.47
Supervisor 209
HA Core v0.106.6

**Setup**
Odroid N2 with fixed IP configured
Router & Switch: Unifi

**Description of problem:**
If the network connection gets interrupted and reconnected Home Assistant on Odroid N2 is not reachable anymore. The entire system is not reachable. IP address can not be pinged, also HA Add-On's like MQTT server is not reachable anymore.

While a network cable disconnect is very unlikely to happen this situation can happen for example if the firmware of a router/switch was updated and the switch rebooted.

**Steps to reproduce:**
* Setup HA on Odroid N2
* Configure the system to use fixed IP
* Optionally install MQTT Add-On
* Make sure system is working and can be connected from browser
* Disconnect network cable
* Wait 1 min
* Reconnect network cable
> While the system still might be running fine it does not reestablish the network connection.

pepeEL, 2020-03-23:
I also has the same problem. Good will be if the problem will be resolved. 
SOmetimes we need reboot router and after reboot HA not working on IP.

pvizeli, 2020-05-08:
Fix: 4.6

closed 2020-05-08 by pvizeli