home-assistant/operating-system #612: HassOS - ohci-pci HC died; cleaning up => USB devices disconnecting

boesing, 2020-03-24:
**HassOS release with the issue:**
|arch | x86_64
|-- | --
|dev | false
|docker | true
|hassio | true
|os_name | Linux
|os_version | 4.19.107
|python_version | 3.7.7
|timezone | Europe/Berlin
|version | 0.107.5
|virtualenv | false

**Supervisor logs:**
None.

**Journal logs:**
None as of hassOS

**Kernel logs:**
```
[    0.000000] Linux version 4.19.107 (builder@6e8f57a351c2) (gcc version 7.4.0 (Buildroot 2019.02.9-g834754d9)) #1 SMP Fri Mar 6 14:07:57 UTC 2020
[    0.000000] Command line: apparmor=1 security=apparmor console=tty1 bootchooser.active=A root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait zram.enabled=1 zram.num_devices=3
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003dd63fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003dd64000-0x000000003dd64fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003dd65000-0x000000003de90fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003de91000-0x000000003df1afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003df1b000-0x000000003ef9afff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ef9b000-0x000000003eff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003eff3000-0x000000003effafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003effb000-0x000000003effefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003efff000-0x000000003f46afff] usable
[    0.000000] BIOS-e820: [mem 0x000000003f46b000-0x000000003ffeffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi:  ACPI=0x3effa000  ACPI 2.0=0x3effa014  SMBIOS=0x3effd000  MPS=0x3effc000 
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 9791001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 2817919756 cycles
[    0.000001] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000003] tsc: Detected 2808.000 MHz processor
[    0.001878] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001879] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001882] last_pfn = 0x3f46b max_arch_pfn = 0x400000000
[    0.001890] MTRR default type: uncachable
[    0.001891] MTRR variable ranges disabled:
[    0.001891] Disabled
[    0.001892] x86/PAT: MTRRs disabled, skipping PAT initialization too.
[    0.001893] CPU MTRRs all blank - virtualized system.
[    0.001895] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.001936] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[    0.001948] Scanning 1 areas for low memory corruption
[    0.001968] BRK [0x09a01000, 0x09a01fff] PGTABLE
[    0.001970] BRK [0x09a02000, 0x09a02fff] PGTABLE
[    0.001970] BRK [0x09a03000, 0x09a03fff] PGTABLE
[    0.001987] BRK [0x09a04000, 0x09a04fff] PGTABLE
[    0.002035] BRK [0x09a05000, 0x09a05fff] PGTABLE
[    0.002047] BRK [0x09a06000, 0x09a06fff] PGTABLE
[    0.002092] BRK [0x09a07000, 0x09a07fff] PGTABLE
[... 575 lines cut ...]
[   71.565947] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[   71.573941] usbcore: registered new interface driver cdc_acm
[   71.574529] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  110.255905] ohci-pci 0000:00:06.0: frame counter not updating; disabled
[  110.258272] ohci-pci 0000:00:06.0: HC died; cleaning up
[  110.261072] usb 1-1: USB disconnect, device number 2
[  110.262286] usb 1-2: USB disconnect, device number 3
[  113.465578] udevd[180]: starting version 3.2.8
[  113.511178] udevd[181]: starting eudev-3.2.8
```

**Description of problem:**

Hey there, if I connect my USB device, it gets disconnected after almost exactly 60 seconds.

I am using Virtualbox on OSX Cataline with installed extension pack and forwarding the USB device to my virtual machine running HassOS (used the VMDK, converted to VDI, extended to 32GB, etc.).

I've found this launchpad bug:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1634737

It says that there might be a problem with the kernel version but as I dont have access to HassOS host machine, I cannot verify if I am still on a probably bugged kernel version.

Does anyone had this issue before and could help me solving this issue? I want to run hassio on my Pi 4 which arrives on wednesday but until then, I would love to configure it inside my virtualbox.

My devices are not reconnecting after this. I have to fully restart virtualbox to make it work again.

boesing, 2020-03-26:
Switched to Raspberry Pi4. No issues in there. 
If you're fine with that VMDK on OSX does not provide permanent access to an USB device, feel free to close as I continue working on the Pi now. 👍

closed 2020-03-27 by boesing

nschwermann, 2021-07-01:
Also having this issue on running home assistant with virtual box on mac :(

Warrenvan1, 2022-03-21:
Hi i got this error

c2] ehci-pci 0000:00:1a.0: hc died cleaning up

Warrenvan1, 2022-03-21:
c2] ehci-pci 0000:00:1a.0: fatal error