home-assistant/operating-system #613: Cannot build rpi3: Unable to find image 'hassos-hostapps:latest' locally

lucagiove, 2020-03-25:
Hi,
I wanted to submit a patch for bluetooth issues with the rpi3B.
I followed the documentation but after hours of download/compiling the build stops below:

```
root@f1eaff78249a:/build# make rpi3
config rpi3
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external "rpi3_defconfig"
make[1]: Entering directory '/build/buildroot'
#
# configuration written to /build/buildroot/.config
#
make[1]: Leaving directory '/build/buildroot'
build rpi3
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external
make[1]: Entering directory '/build/buildroot'
/usr/bin/make -j1 O=/build/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
make[2]: Leaving directory '/build/buildroot'
>>> hassos 1.0.0 Installing to target
docker run --rm --privileged -e BUILDER_UID="0" -e BUILDER_GID="0" -v /build/buildroot/output/images:/export hassos-hostapps --supervisor "homeassistant/armv7-hassio-supervisor" --supervisor-version "209" --supervisor-args "-e HOMEASSISTANT_REPOSITORY=homeassistant/raspberrypi3-homeassistant" --supervisor-profile "hassio-supervisor" --supervisor-profile-url "https://version.home-assistant.io/apparmor.txt" --cli "homeassistant/armv7-hassio-cli" --cli-version "21" --cli-args "--network=hassio --add-host supervisor:172.30.32.2" --cli-profile "docker-default" --cli-profile-url "" --apparmor "supervisor/apparmor"
Unable to find image 'hassos-hostapps:latest' locally
docker: Error response from daemon: pull access denied for hassos-hostapps, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
package/pkg-generic.mk:331: recipe for target '/build/buildroot/output/build/hassos-1.0.0/.stamp_target_installed' failed
make[1]: *** [/build/buildroot/output/build/hassos-1.0.0/.stamp_target_installed] Error 125
make[1]: Leaving directory '/build/buildroot'
Makefile:24: recipe for target 'rpi3' failed
make: *** [rpi3] Error 2
```

Any hints?

pvizeli, 2020-03-31:
Look like they fails before but the rerun only if they are not exists on the build folder

lucagiove, 2020-03-31:
Sorry I don't get you the error seem this one:
```
Unable to find image 'hassos-hostapps:latest' locally
docker: Error response from daemon: pull access denied for hassos-hostapps, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
```
If I rerun I get the same error.

pvizeli, 2020-03-31:
You need clean `make clean` and restart the build. The first error message is what the issue represents. If you later run the build again, you show that error above which doesn't say anything about the initial problem.

lucagiove, 2020-04-01:
Ok now the error is:
`failed to start daemon: error initializing graphdriver: driver not supported
`
Going to analyse it...

pvizeli, 2020-04-01:
Oh, if you use MacOSX, you can build the OS. No one was able to do that without a vm on OSX.

lucagiove, 2020-04-01:
No I'm using Ubuntu 18.04

lucagiove, 2020-04-01:
Yeah I made it!
I updated the docker-ce version on my Ubuntu 18.04 now I've docker-ce 19.03.8.

In order to avoid doing `make clean` and then `make rpi3` again (that takes hours on my pc)
I cleaned up the failed step and re-run the build in this way:
```
root@aa954cf592ce:/build/buildroot# make hassos-dirclean
rm -Rf /build/buildroot/output/build/hassos-1.0.0
root@aa954cf592ce:/build/buildroot# cd ..
root@aa954cf592ce:/build# make rpi3
```
and worked!

```
make[1]: Leaving directory '/build/buildroot'
cp -f /build/buildroot/output/images/hassos_* /build/release/
# Do not clean when building for one target
finished rpi3
```

closed 2020-04-01 by lucagiove