home-assistant/operating-system #615: Potential filesystem corruption during failed HassOS upgrade from 3.10 --> 3.12

adamfeldman, 2020-03-30:
**HassOS release with the issue:**
Start: 3.10
Target: 3.12

**Supervisor version: _214_**
**Supervisor logs:**

<details><summary>CLICK ME</summary>
<p>

```
20-03-30 14:26:07 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attach to Supervisor homeassistant/amd64-hassio-supervisor with version 214
20-03-30 14:26:07 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connect Supervisor to hassio Network
20-03-30 14:26:07 INFO (SyncWorker_1) [supervisor.docker.interface] Cleanup images: ['homeassistant/amd64-hassio-supervisor:213']
20-03-30 14:26:08 INFO (MainThread) [__main__] Setup Supervisor
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd1
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.hostname1 - /org/freedesktop/hostname1
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: de.pengutronix.rauc - /
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Connect to dbus: org.freedesktop.NetworkManager - /org/freedesktop/NetworkManager/DnsManager
20-03-30 14:26:08 INFO (MainThread) [supervisor.host.info] Update local host information
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-03-30 14:26:08 INFO (MainThread) [supervisor.host.services] Update service information
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-03-30 14:26:08 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-03-30 14:26:08 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
Failed to load cookie file from cookie: No such file or directory
20-03-30 14:26:08 INFO (MainThread) [supervisor.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
20-03-30 14:26:08 INFO (MainThread) [supervisor.host.services] Reload local service hassos-apparmor.service
20-03-30 14:26:08 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1
20-03-30 14:26:08 INFO (SyncWorker_1) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-dns with version 3
20-03-30 14:26:08 INFO (SyncWorker_2) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-audio with version 9
20-03-30 14:26:08 INFO (SyncWorker_5) [supervisor.docker.interface] Attach to homeassistant/amd64-hassio-cli with version 24
20-03-30 14:26:08 INFO (MainThread) [supervisor.misc.forwarder] Start DNS port forwarding to 172.30.32.3
20-03-30 14:26:08 INFO (SyncWorker_0) [supervisor.docker.interface] Restart homeassistant/amd64-hassio-dns
20-03-30 14:26:12 INFO (MainThread) [supervisor.dns] Updated /etc/resolv.conf
20-03-30 14:26:12 INFO (SyncWorker_5) [supervisor.docker.interface] Attach to homeassistant/qemux86-64-homeassistant with version 0.107.7
20-03-30 14:26:12 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
20-03-30 14:26:12 INFO (MainThread) [supervisor.hassos] Detect HassOS 3.10 / BootSlot B
20-03-30 14:26:12 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/d8ef7a72/smartthings/config.json: required key not provided @ data['arch']. Got None
20-03-30 14:26:12 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/a0d7b954 repository
20-03-30 14:26:12 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/d8ef7a72 repository
20-03-30 14:26:12 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/core repository
20-03-30 14:26:13 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/d8ef7a72/smartthings/config.json: required key not provided @ data['arch']. Got None
20-03-30 14:26:13 INFO (MainThread) [supervisor.store] Load add-ons from store: 65 all - 65 new - 0 remove
20-03-30 14:26:13 INFO (MainThread) [supervisor.addons] Found 17 installed add-ons
20-03-30 14:26:13 INFO (SyncWorker_8) [supervisor.docker.interface] Attach to homeassistant/amd64-addon-letsencrypt with version 4.3
20-03-30 14:26:13 INFO (SyncWorker_17) [supervisor.docker.interface] Attach to homeassistant/amd64-addon-check_config with version 3.1
20-03-30 14:26:13 INFO (SyncWorker_15) [supervisor.docker.interface] Attach to hassioaddons/vscode-amd64 with version 2.1.0
20-03-30 14:26:13 INFO (SyncWorker_6) [supervisor.docker.interface] Attach to hassioaddons/pi-hole-amd64 with version 4.1.1
20-03-30 14:26:13 INFO (SyncWorker_10) [supervisor.docker.interface] Attach to homeassistant/amd64-addon-samba with version 9.0
20-03-30 14:26:13 INFO (SyncWorker_16) [supervisor.docker.interface] Attach to homeassistant/amd64-addon-ssh with version 8.5.3
20-03-30 14:26:13 INFO (SyncWorker_11) [supervisor.docker.interface] Attach to hassioaddons/node-red-amd64 with version 6.1.4
20-03-30 14:26:13 INFO (SyncWorker_19) [supervisor.docker.interface] Attach to hassioaddons/ssh-amd64 with version 7.3.1
20-03-30 14:26:13 INFO (SyncWorker_3) [supervisor.docker.interface] Attach to homeassistant/amd64-addon-mosquitto with version 5.1
20-03-30 14:26:13 INFO (SyncWorker_4) [supervisor.docker.interface] Attach to hassioaddons/log-viewer-amd64 with version 0.8.0
20-03-30 14:26:13 INFO (SyncWorker_12) [supervisor.docker.interface] Attach to hassioaddons/aircast-amd64 with version 2.3.0
20-03-30 14:26:13 INFO (SyncWorker_14) [supervisor.docker.interface] Attach to hassioaddons/influxdb-amd64 with version 3.6.1
20-03-30 14:26:13 INFO (SyncWorker_7) [supervisor.docker.interface] Attach to homeassistant/amd64-addon-deconz with version 5.1
20-03-30 14:26:13 INFO (SyncWorker_9) [supervisor.docker.interface] Attach to homeassistant/amd64-addon-cec_scan with version 2.4
20-03-30 14:26:13 INFO (SyncWorker_17) [supervisor.docker.interface] Attach to hassioaddons/ide-amd64 with version 2.2.0
20-03-30 14:26:13 INFO (SyncWorker_13) [supervisor.docker.interface] Attach to hassioaddons/grafana-amd64 with version 4.2.1
20-03-30 14:26:13 INFO (SyncWorker_15) [supervisor.docker.interface] Attach to afelds/amd64-hassio-smartthings-mqtt-bridge with version 20180714
20-03-30 14:26:13 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-03-30 14:26:13 INFO (MainThread) [supervisor.snapshots] Found 6 snapshot files
20-03-30 14:26:13 ERROR (MainThread) [supervisor.snapshots.snapshot] Can't read snapshot tarfile /data/backup/._Hass_io_0_94_1_pre_hassos_migration.tar: invalid header
20-03-30 14:26:13 INFO (MainThread) [supervisor.discovery] Load 2 messages
20-03-30 14:26:13 INFO (MainThread) [supervisor.ingress] Load 0 ingress session
20-03-30 14:26:13 INFO (MainThread) [supervisor.secrets] Load Home Assistant secrets: 3
20-03-30 14:26:13 INFO (MainThread) [__main__] Run Supervisor
[... 994 lines cut ...]
</details>

**Description of problem:**

Initiated HassOS upgrade from Supervisor UI. VM failed to boot, with what I believe to be filesystem-related errors.

HassOS is running within a Proxmox VM, on an Intel x86 processor.

As I had used Proxmox to take a disk snapshot prior to initiating the HassOS upgrade, I rolled-back to the pre-upgrade snapshot. **The logs above are from the VM _post-rollback_.**

The following screenshots show the VM in its failed state:

<details><summary>**First boot after attempting HassOS upgrade** </summary>

[image]

</details>

<details><summary>**Second boot, after unclean shutdown (using "Stop")** </summary>

[image]

</details>

Please let me know what other information I can collect, or how I can otherwise assist with troubleshooting. Thank you!

adamfeldman, 2020-03-30:
Per https://github.com/home-assistant/supervisor/issues/1620, **I am now having issues running HassOS 3.10 with Supervisor 214 (_latest release as of today_).**

pvizeli, 2020-03-30:
It's carefully to use snapshot because that is not aware of the docker database. I'm just 100% sure that the issue starts with a rollback of a snapshot. Please use Home Assistant snapshots they are docker safe.

adamfeldman, 2020-03-30:
@pvizeli thank you for your assistance!

I only take VM snapshots for rollback in emergencies 🙂. I also create and export regular HA snapshots.

This is the result of logging in to the console (not via Terminal add-on) and attempting cli update:

[image]

The same thing happens when I run the same command over ssh (currently, when I ssh into hassos, I get the `hassio` utlity, not `ha` -- as you pointed out, this is what I need to fix)

adamfeldman, 2020-03-30:
According to http://hassio.local:8123/hassio/addon/core_ssh, I am already running `core_ssh` version `8.2`, which is `latest`. Yet it offers me the `hassio` cli, and not `ha`. @pvizeli I am not sure which other add-on you want me to install (per your comment in the other issue I opened)

adamfeldman, 2020-03-30:
Oops, I see in repo that latest is `8.5.3`: https://github.com/home-assistant/hassio-addons/blob/master/ssh/config.json

I wonder why my Supervisor isn't picking up the latest version? 

[image]

adamfeldman, 2020-03-30:
Anndd running the following over my SSH session into HassOS picked up the latest versions...

`$ hassio addons reload`

Sorry to be too quick to comment here. 😅 will update here with any further steps I end up needing to take to solve the issue.

adamfeldman, 2020-03-30:
So far so good. Thanks again @pvizeli!
!
Note for future readers of this thread: I ended up stopping work on my rolled-back snapshot that was quite old (~5 weeks). I used what I learned to fix-up* my almost-working original VM (which I had rolled-back to the point right after the Supervisor was updated to 214, and right before updating from 3.10 to 3.12).

*by running `ha cli update` or the equivalent for the older cli

closed 2020-03-30 by adamfeldman