home-assistant/operating-system #617: Cannot update  Error: 502: Bad Gateway

dany250, 2020-04-01:
**HassOS release with the issue:**

currently running version 3.11
**Supervisor logs:**

20-04-01 22:43:23 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:43:23 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:43:23 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:43:28 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:43:28 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:43:28 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:43:33 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:43:33 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:43:33 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:43:38 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:43:38 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:43:38 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:43:41 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
20-04-01 22:43:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:43:43 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:43:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
20-04-01 22:44:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-04-01 22:44:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-04-01 22:44:57 ERROR (MainThread) [supervisor.utils.json] Can't read json from /data/addons/git/5f2994cf/repository.json: [Errno 2] No such file or directory: '/data/addons/git/5f2994cf/repository.json'
20-04-01 22:44:57 WARNING (MainThread) [supervisor.store.data] Can't read repository information from /data/addons/git/5f2994cf/repository.json
20-04-01 22:44:57 INFO (MainThread) [supervisor.store] Load add-ons from store: 65 all - 0 new - 0 remove
20-04-01 22:45:27 INFO (SyncWorker_15) [supervisor.docker.interface] Stop addon_a0d7b954_nodered application
20-04-01 22:45:27 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-04-01 22:45:41 INFO (SyncWorker_15) [supervisor.docker.interface] Clean addon_a0d7b954_nodered application
20-04-01 22:46:10 INFO (SyncWorker_12) [supervisor.docker.addon] Start Docker add-on hassioaddons/node-red-armv7 with version 6.1.4
20-04-01 22:46:11 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_nodered
20-04-01 22:46:14 INFO (MainThread) [supervisor.api.security] /dns/info access from a0d7b954_nodered
20-04-01 22:46:19 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.32.1:62289 ssl:None [Connection refused]
20-04-01 22:46:37 INFO (MainThread) [supervisor.auth] Auth request from core_mosquitto for mqttuser2
20-04-01 22:46:37 INFO (MainThread) [supervisor.auth] Success login from mqttuser2
20-04-01 22:46:41 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:46:41 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:46:41 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
20-04-01 22:48:49 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:0 is not WSMsgType.TEXT
20-04-01 22:48:49 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-04-01 22:48:54 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:48:54 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:48:54 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
20-04-01 22:56:44 INFO (SyncWorker_11) [supervisor.docker.interface] Restart homeassistant/raspberrypi4-homeassistant
20-04-01 22:56:49 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:1000 is not str
20-04-01 22:56:49 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-04-01 22:56:52 ERROR (MainThread) [supervisor.homeassistant] Error on call https://172.30.32.1:8123/api/services/homeassistant/restart: None
20-04-01 22:56:52 ERROR (MainThread) [supervisor.api.proxy] Error on API for request services/homeassistant/restart
20-04-01 22:56:54 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:56:54 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:56:54 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:56:59 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:56:59 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:56:59 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:04 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:57:04 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:57:04 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:09 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:57:09 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:57:09 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:14 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:57:14 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:57:14 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:57:19 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:57:19 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:57:24 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:57:24 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:29 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
[... 17 lines cut ...]
20-04-01 22:57:49 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:54 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-04-01 22:57:54 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
20-04-01 22:57:54 ERROR (MainThread) [supervisor.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-04-01 22:57:58 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
[... cut ...]
future: <Task finished coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:129> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 140, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 92, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

**Journal logs:**

**Kernel logs:**

**Description of problem:**
it shows ERROR:502: Bad Gateway every time i try to update from 3.11

markinsutton, 2020-04-19:
having same issue and error. is there an easy way to roll back to 3.11 rather than doing a complete restore from backup?

Semag, 2020-05-22:
I also have this issue, but am on 3.12. Haven't been able to upgrade the OS since 3.13 came out,.

Mediacj, 2020-05-30:
Me to same problem also from 3.12 to 4.8

ctm514, 2020-06-01:
Eu também estou com esse problema...  versão 3.13 pra 4,8. Alguém sabe como resolver?

Bettinaglio, 2020-06-02:
Same problem here when trying to upgrade from 3.13 to 4.8. Any solution?

Semag, 2020-06-02:
I ran a tcpdump while i ran the os update and it seems to grab the entire upgrade, then fails after that - there is definitely a ton of traffic passing. Is there anyway to manually download the OS upgrade an upgrade in place using a local file?

Example:

09:06:18.579391 IP homeassistant.home.35306 > s3-1-w.amazonaws.com.https: Flags [.], ack 65872, win 9774, length 0
09:06:18.579624 IP homeassistant.home.35306 > s3-1-w.amazonaws.com.https: Flags [.], ack 67304, win 9767, length 0
09:06:18.658610 IP s3-1-w.amazonaws.com.https > homeassistant.home.35306: Flags [.], seq 67304:70168, ack 1, win 124, length 2864
09:06:18.847574 IP s3-1-w.amazonaws.com.https > homeassistant.home.35306: Flags [.], seq 70168:71600, ack 1, win 124, length 1432
09:06:18.849707 IP homeassistant.home.35306 > s3-1-w.amazonaws.com.https: Flags [.], ack 67304, win 9782, options [nop,nop,sack 1 {70168:71600}], length 0
09:06:18.977519 IP s3-1-w.amazonaws.com.https > homeassistant.home.35306: Flags [.], seq 67304:68736, ack 1, win 124, length 1432
09:06:18.978939 IP homeassistant.home.35306 > s3-1-w.amazonaws.com.https: Flags [.], ack 68736, win 9773, options [nop,nop,sack 1 {70168:71600}], length 0

Semag, 2020-06-02:
@Mediacj  - are you running on a virtual machine?

Mediacj, 2020-06-03:
Oh yes sorry, I am running it as a VM under Proxmox.

Semag, 2020-06-03:
Thanks @Mediacj  - is your adapter set to NAT in the VM?  I'm running virtual box, but i have the adapter "bridged" from the host machine, so i can't change the device type right now.  I'm going to try playing with changing it to NAT and some other options possibly.

Semag, 2020-06-03:
@Mediacj  - wow that is amazing. It worked!!!

For anyone else finding this, i changed the Virtualbox VM adapter to NAT. Obviously this is going to rip HA off the network and disable SSH access.

I had to login via a VNC window and run 'os update' - but it updated VERY fast, and then rebooted and came up on version 4.8.

I've now moved it back to a bridged ethernet adapter and all is good.

Mediacj, 2020-06-05:
Ok great! My adapter is bridged to the host.

I just updated to version 4.9 without errors.

XxCabbyoO, 2020-06-07:
same error 502 bad gateway from 3.13 to 4.9, using rpi4 2gb

after trying again to start the update:
20-06-07 18:40:14 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.9/hassos_rpi4-4.9.raucb: Cannot connect to host github-production-release-asset-2e65be.s3.amazonaws.com:443 ssl:None [Operation timed out]
20-06-07 18:40:59 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.9/hassos_rpi4-4.9.raucb: 
20-06-07 18:40:59 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=HassOSUpdateError()>

third try: Error: 502: Bad Gateway

any solution?

pvizeli, 2020-06-08:
Geo-blocking? If yes, you need use maybe an VPN?

XxCabbyoO, 2020-06-08:
> 
> 
> Geo-blocking? If yes, you need use maybe an VPN?

how do you come up with geo blocking? how can that be the reason? Confusing for me

balloob, 2020-06-08:
Wait, you've opened issue about that you can't reach the update server but forgot to mention that you're blocking external access in your initial post?

SAOPP, 2020-06-11:
I have exactly the same problem now, I did not see this topic because of the name, and only now I saw a 502 error at my side too.

https://github.com/home-assistant/operating-system/issues/738

XxCabbyoO, 2020-06-16:
today the update from 3.13 to 4.10 suddenly worked. I haven't changed anything

SAOPP, 2020-06-17:
Thanks, I will try today to update my other instance...

SAOPP, 2020-06-17:
All fine now!

morfei1, 2020-06-18:
I've tried to update the supervisor 3.13 to 4.10 several time without success. The installation was cancelled with message: Error: 502: Bad Gateway. Rpi3b+, HA v0.111.4.

radostint, 2020-06-21:
I got the same problem here, updating from 3.13 to 4.10. Tried at least 4-5 times but no luck. Every single try I get "Error: 502: Bad Gateway."

Semag, 2020-06-23:
FYI - i just did the update again.

Had issues (502 bad gateway) updating the OS. So I repeated my previous process:

1)  Set the virtualmachine network adapter to NAT
2)  Connect via VNC to the console
3)  Issue an ha> os update
4)  HA restarts, reset the network adapter to bridge.

Seems to work like a charm.

watermarkhu, 2020-06-25:
Same issue from 3.13 to 4.10

watermarkhu, 2020-06-26:
It may be caused by a time-out due to the long download time (>10 min).
I solved it by performing the update via a USB stick. [Instructions here](https://github.com/home-assistant/operating-system/blob/600eb5e2081afb9dbdfa50b552d1af31e54dcedf/Documentation/configuration.md). 

You'll need to:
1. Format a USB to FAT32/EXT4/NTFS and name it CONFIG (all capitals)
2. Download the correct hassos-xy.raucb for you. You can find the version your device tries to download in the system log. 
3. Place hassos-xy.raucb in the root folder of your USB
4. Restart your device with the USB plugged in.

XxCabbyoO, 2020-07-06:
The error reoccurs with the new update

Error: 502: Bad Gateway

20-07-06 09:23:55 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.11/hassos_rpi4-4.11.raucb: 
20-07-06 09:23:55 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 136, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 88, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

labachev, 2020-07-07:
Same here with the new update from 4.10 to 4.11

cpuks, 2020-07-08:
Same problem - HassOS 4.10 running in VM - can't update, everything else works fine...

balloob, 2020-07-08:
Please use reactions instead of writing me too's.

psycmos, 2020-07-09:
> 
> 
> Please use reactions instead of writing me too's.

You are right.
I'm using VM in proxmox.

The error:
Client timeout error on API request os/update
17:54:57 – Hass.io (ERROR)

facuzestoa, 2020-08-12:
Same happening from 4.11 to 4.12. Is there any way to increase this 10 min time-out?

olli-dot-dev, 2020-10-20:
Still having the same problem running a RPI4:

Here's my recent error logs:
```
20-10-20 07:38:10 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.14/hassos_rpi4-4.14.raucb
20-10-20 07:48:10 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.14/hassos_rpi4-4.14.raucb: 
20-10-20 07:48:10 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-329133' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 136, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 88, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError
```

bibikiwi, 2021-03-04:
When this happens to me (I'm kinda getting used to updates via the web interface being flaky), I do it via the command line. That seems to work more reliably. Login to the device (a RPI in my case) via SSH and (as root) type `ha core update` or `ha supervisor update`.
P.S. It just says 'Command completed successfully.' and returns, so to keep an eye on what's happening, I also open 2 other SSH connections, with `docker logs -f homeassistant` (just to see when it restarts) and `watch docker ps` (to keep an eye on what containers are being restarted and what versions.

closed 2022-01-11 by github-actions[bot]