home-assistant/operating-system #628: Raspberry Pi POE fan control settings in config.txt not taking effect in 3.13

inthehoose, 2020-04-19:
**HassOS release with the issue:**

3.13
**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

I have the following added to the end of my config.txt file:

dtparam=poe_fan_temp0=62000,poe_fan_temp1=65000

In HassOS 3.12 or earlier, this config works as expected.

After upgrading to HassOS 3.13, the POE fan starts to kick in at the default 50 degrees celsius mark and not at my pre-determined 62 degs.  Everything works as expected again after I downgraded back to 3.12 via the command line (ha os downgrade --version 3.12).

123WTF456, 2020-04-20:
I’ve got the same issue.

I noticed this earlier, does it have bearing at all?

https://github.com/raspberrypi/firmware/issues/1369

123WTF456, 2020-04-22:
I added the additional lines e.g.

dtparam=poe_fan_temp0=65000,poe_fan_temp0_hyst=5000
dtparam=poe_fan_temp1=70000,poe_fan_temp1_hyst=4999
dtparam=poe_fan_temp2=75000,poe_fan_temp0_hyst=4999
dtparam=poe_fan_temp3=80000,poe_fan_temp1_hyst=4999

And it seems better now.

inthehoose, 2020-04-22:
thank you @123WTF456.  I will try the suggestion and report back

fjufju, 2020-04-29:
@inthehoose works?
I want buy https://www.argon40.com/argon-fan-hat-for-raspberry-pi-4-raspberry-pi-3b-and-raspberry-pi-3-b.html

123WTF456, 2020-04-29:
It worked fine for me

123WTF456, 2020-04-29:
However that's not the PoE HAT but a fan HAT - not sure if it makes a difference

fjufju, 2020-04-30:
> However that's not the PoE HAT but a fan HAT - not sure if it makes a difference

I know but it also has speed control.
Don't want order if not working with HA on PI4, but I think the best way to test is to order one :)

inthehoose, 2020-05-10:
Finally applied the fix as suggested by @123WTF456, upgraded to 3.13 and all seems to be working as expected!  :)

closed 2020-05-10 by inthehoose

LordMike, 2021-02-13:
> I added the additional lines e.g.
> 
> dtparam=poe_fan_temp0=65000,poe_fan_temp0_hyst=5000
> dtparam=poe_fan_temp1=70000,poe_fan_temp1_hyst=4999
> dtparam=poe_fan_temp2=75000,poe_fan_temp0_hyst=4999
> dtparam=poe_fan_temp3=80000,poe_fan_temp1_hyst=4999
> 
> And it seems better now.

@123WTF456  Note the duplicates in the hysteresis params. You're not setting `poe_fan_temp2_hyst` and `poe_fan_temp3_hyst`