home-assistant/operating-system #648: Odroid N2 not booting with HassOS 4.5 release/update

Multipage, 2020-05-01:
HassOS release with the issue: 4.5

After updating to HassOS my Odroid N2 went dead, no heartbeat, nothing.

To check if my upgrade was bad i got myself other SD cards for testing. Flashed the latest Odroid N2 image.
It also doesn't boot. Black screen after power on and no heartbeat.
Tried fiddling with the SPI switch, doesn't do anything

Got an image of Armbian and that is booting well so it isn't my Odroid.

Hardware used:
Odroid N2
16GB Sandisk SD
32GB Kingston SD
256GB Samsung EVO plus SD

mabaat, 2020-05-01:
I got the same issue with an Odroid N2 and eMMC. After update from 4.4 to 4.5 it went black. Installed 4.4 again and all things are fine. Unfortunately also no logs from my end.

liudger, 2020-05-02:
Same problem here :( A bit too quick to install

liudger, 2020-05-02:
I there a way to install HassOS 4.4 withouth overwritng the other parts? My last back up wasn't that recent

Multipage, 2020-05-02:
Afaik no. And since it's not booting you can't recover or anything. 

You need to reinstall 4.4 and restore your backup, cry and start over again. Had the same thing few months back.  All was lost. Had to start al over again. 

Better install Google drive backup. It's a life saver!

liudger, 2020-05-02:
You can read the mmc card. So in principle you can fix it

frenck, 2020-05-02:
If I'm correct, a couple of restarts should restore the previous version (not 100% sure).

liudger, 2020-05-02:
Thanks I will let you know if that really works.

Multipage, 2020-05-02:
It doesn't even boot so how do you think that will help?  I doubt this will work.

frenck, 2020-05-02:
@Multipage, it is a firmware-based embedded system. If it fails to start multiple times, it will fall back to the previous firmware. That is the theory.

liudger, 2020-05-02:
Hi frenck. Thanks for the advice. It didn't work but I could recover from my backup 5 weeks ago.
Didn't change too much luckily

Is there a certain amount of time you need to turn on and off? And is there some documentation? Also the update is still available maybe disable the beta update for now? Just in case more Odroid N2 users are using the beta (only option for now)

pvizeli, 2020-05-03:
I think the issue is that the official kernel (which we use since 4.5) have an issue with the SD card slot on the DTS file. The solution would be to replace the dts file on the boot partition with one from 4.4 - I need reverse engineer the hardkernel dts file on mondey to see what is fails with the upstream kernel they we use on 4.4 - Since we use the device only with eMMC, I did not test the card slot with the official kernel from hardkernel/odroid.

MattiBaat, 2020-05-03:
@pvizeli Did you manage to update to 4.5 with an Odroid N2 and eMMC? I've got the this issue with eMMC.

Multipage, 2020-05-03:
The problem is with SD card AND eMMC too.
Sd-card uses the eMMC setting of the SPI switch.

pepeEL, 2020-05-04:
But when will be support booting from USB SSD on Odroid N2 ? Now dont working. Godd feature will be add this feature.... Please.

closed 2020-05-05 by pvizeli

alex7614, 2020-05-08:
Hi
I just switched Tinker S eMMC (worked fine until 4.4) to new Odroid N2 4GB + eMMC.

I downloaded and tried to flash HassOS 4.4 and 4.6 to eMMC with WinDisk and Etcher and verification pass. 
But the ODROID don't boot to eMMC. If I use switch to boot directly to eMMC = Only RED LED. 
Otherwise, it boot to PETITBOOT but no on HassOS. 

But If I put eMMC module to MicroSD Reader and put the MicroSD in ODROID N2, HassOS boot fine... 

Someone would have any idea ? Can we start on the eMMC directly as on Tinker S?

Thanks !

pvizeli, 2020-05-08:
99% of the issue is if people do not run etcher as administrator which make that the eMMC is not bootable. I can confirm that the 4.4, 4.5 and 4.6 boot without issues from eMMC

alex7614, 2020-05-08:
> 99% of the issue is if people do not run etcher as administrator which make that the eMMC is not bootable. I can confirm that the 4.4, 4.5 and 4.6 boot without issues from eMMC

Hi, thanks for your reply.
I just tried to run Etcher in Administrator and Flash 4.6 HassOS -> Verification PASS but not boot to eMMC on ODROID. But boot if I put eMMC to MicroSD CARD READER and MicroSD to ODROID.

Just re-tried to flash Odroid Ubuntu 18.04.4 img to eMMC with Etcher -> Verficiation pass and boot directly to eMMC from Odroid. I get the Ubuntu login console on HDMI.

I really don't understand... Possible eMMC module defective ? But I don't unterstand why with Android img or Ubuntu img works and not HassOS....

pvizeli, 2020-05-08:
Sorry, I can't help. I flash now 8 N2 without issue and the flash module from Odroid

alex7614, 2020-05-08:
Maybe the MicroSD Card Reader is the problem 🤷🏻‍♂️ 
I see only that...
I will buy the USB module that you said
Thanks

jrhodk, 2020-05-23:
I have the same problem, and my emmc writer works fine! Version 4.3 of hassio works perfectly, but the moment i update to 4.6 it becomes unresponsible. (I have the suggested eMMC writer from Hardkernel) And on the screen the message "no backend enabled for fe.dai-link-0" is repeated again and again when it reboots after an update.

glenswada, 2020-07-27:
> If I'm correct, a couple of restarts should restore the previous version (not 100% sure).

you gave me bit of hope.  After some 10 reboots waiting 5 minutes between every reboot the lights on my keyboard suddenly lit up.  Thanks you saved my bacon.

tjengbudi, 2020-09-15:
hello, is there no fix for this? i download latest build for odroid n2. flashed into emmc and trying to boot but failed. is this emmc never supported by hasio?

jrhodk, 2020-09-22:
I gave up getting an answer, thou my problem was that it did start, but
crashed after a while.

liudger, 2020-09-22:
> hello, is there no fix for this? i download latest build for odroid n2. flashed into emmc and trying to boot but failed. is this emmc never supported by hasio?

My N2 with emmc seems to work fine now. Which version do you have of the N2? There are different models now. I had problems a few versions earlier with hassos, but now it feels quite stable. Maybe try an earlier version and upgrade from that when home assistance is running?

tjengbudi, 2020-09-22:
mine is odroid N2. would you like to tell me which version you are using? i
always failed when used emmc. it can works when the emmc plug into sdcard
port. but it will be really works nice if hasio can works with emmc at emmc
port

jrhodk, 2020-09-28:
I have changed to a nuc, which works as a charm, and the odroid n2 i use as
a CoreElec musicplayer instead! (o;

liudger, 2020-09-28:
> 
> mine is odroid N2. would you like to tell me which version you are using? i always failed when used emmc. it can works when the emmc plug into sdcard port. but it will be really works nice if hasio can works with emmc at emmc port
> […](#)

ODROID-N2+ is a revision from ODROID-N2, but if you got normal N2 and it's not working it could be either etcher or faulty emmc memory stick? 

other issues with emmc and etcher
#678

tjengbudi, 2020-09-29:
@liudger i already try install with balena etcher 1.5.80 that your link explained it works with old version. but mine is not
i trying used armbian and the rom is working just fine. so i think the emmc is ok
if i use sdcard port it can run, even used emmc plug into the dongle. 
it is really weird cause i did not really understand how kernel or bootloader works. 
i still did not update my petitboot. but i think it is not involved

how do you think? any suggestion?

liudger, 2020-09-29:
> @liudger i already try install with balena etcher 1.5.80 that your link explained it works with old version. but mine is not
>
> i trying used armbian and the rom is working just fine. so i think the [...]

I think discord channel is maybe a better place to find a solution and talk to the developers? It's quite specific and not a common issue for the moment. I am also not a specialist. Also this issue is closed so you get less response I guess

tjengbudi, 2020-09-29:
it might bootloader or dtb issue. i trying 4.4 and it still cannot running well... i might get move on to another device... thanks for your reply

PR #652 (2020-05-05): Use older Kernel for N2
Fix: #648