home-assistant/operating-system #650: Tinker Board S does not boot from eMMC with HassOS 4.5 release/update

mbo18, 2020-05-01:
**HassOS release with the issue:**

HassOS 4.5

**Supervisor logs:**

None, does not boot from eMMC. Does boot from SD card

**Journal logs:**

None

**Kernel logs:**

None

**Description of problem:**
Just tried to upgrade from HassOS 3.13 to 4.5 from 'import from USB' and my Tinker Board S did not boot anymore. Tried to flash 4.5 directly to eMMC and it does not boot. Flash to SD card and it boot fine.
It seems that boot from eMMC is not working anymore with HassOS 4.5

venditdevs, 2020-05-01:
That would make sense  why my Odroid N2 isn't booting anymore from SD. Because the Odroid-N2 must select MMC to get it to boot Home Assistant from SD. I know: that doesn't make sense but it really needs  it ;).
So basically we can say: EMMC booting is broken in HassOS4.5.

Multipage, 2020-05-01:
That would make sense why my Odroid N2 isn't booting anymore from SD. Because the Odroid-N2 must select MMC to get it to boot Home Assistant from SD. I know: that doesn't make sense but it really needs it ;).
So basically we can say: EMMC booting is broken in HassOS4.5.

alex7614, 2020-05-03:
Same problem on Tinker S on eMMC. I flahed back to 4.4 on eMMC until new version.

mbo18, 2020-05-06:
It seems that it will not be solved as Pvizeli has no tinker board S to make tests and rbray89 who worked on it has leaved the project

mbo18, 2020-05-07:
@pvizeli Maybe you should block OS update for this specific board? because people will click on update and it won't boot anymore...
What is the best way now? SD card + offloading data to eMMC?

pvizeli, 2020-05-08:
@Multipage the N2 was just a bug in the official kernel from the manufacture and is fix with 4.6. The Tinker was just designed for usage on Tinker (without S) because for that is needs a new image. It's my wrong that I did not look exactly what Ryan did. But also on the official website, we just list the Tinker board without S (https://www.home-assistant.io/hassio/installation). However, Look like there are some people, maybe someone of you will join the project and spend some days per year into maintenance and responsible for issues.

cybergrimes, 2020-06-01:
What is this? I've had a Tinker Board S running Home Assistant since December 2018. We have had other non-booting OS image issues in the past for the S model that were fixed.

cybergrimes, 2020-06-01:
Here for example, we had issues with booting eMMC on Tinker Board S and the problem was resolved. No mention of S model not being supported: https://github.com/home-assistant/operating-system/issues/344

cybergrimes, 2020-06-01:
> 
> 
> @pvizeli Maybe you should block OS update for this specific board? because people will click on update and it won't boot anymore...
> What is the best way now? SD card + offloading data to eMMC?

What have you decided to do with your setup? I have been on Tinker Board S since December 2019 without issue. We've both posted here in the past with regards to S model not booting after OS updates. I'm a little disappointed to say the least, not expecting to need new hardware.

mbo18, 2020-06-01:
Well, it won't help you :) 
I've bought a used NUC (from 2015) with way more RAM and with an SSD. I've made a good choice: HA restart in 12s, everything is smoother and I can run more addons

cybergrimes, 2020-06-01:
Yeah, I just spied you on forums and saw that. 
I'll consider that option-- no luck with data partition offload?

mbo18, 2020-06-01:
I didn't even tried. I was looking for a NUC before the last HassOS release. When I saw that boot from eMMC was removed, choice was made

cybergrimes, 2020-06-01:
Gotcha, moved mine to a VirtualBox until I figure out something else (if anything)

Cheers

anthfarr, 2020-06-03:
I have the same issue. Is there an easy way to downgrade hassos ?

cybergrimes, 2020-06-03:
@pvizeli any chance you would do take donations to get a Tinker Board S like you did a couple years ago with original Tinker Board? https://github.com/home-assistant/operating-system/issues/33

junologue, 2020-06-05:
> We are not currently expanding supported boards. It takes ongoing maintenance to test every update on every board and we don't have the resources for this right now.

Totally understand. But since Hassos 3.13 works very well for my tinker S still, for how long could i continue to upgrade Home Assistant Core with 3.13 as a base? Does it have a limit?

jpjcrossi, 2020-06-12:
I'm little bit worrying because it was working very well and after an update it just stopped to work.
Now I'm use hassio on SD card and seems to me very unstable.
The Home Assistant just freezing whihout any reason.  :(
Please, I'm beg you. Consider provide support for the eMMC for Hassio on Tinker Board S again.

junologue, 2020-06-12:
> We are not currently expanding supported boards. It takes ongoing maintenance to test every update on every board and we don't have the resources for this right now.

-

I understand. Tinkerboard S is just to good to be left behind though! So please consider it. The emmc is so stable compared to my Pi 3B+ even with the most high end SD-card.

In what version did you drop the support? Is 3.13 really the last one? I’ve heard someone talked about 3.8, but haven’t tried yet.

poelsevev, 2020-07-26:
I'm using 3.9 on my tinkerboard S. Seems to me the issue comes with all version 4.x 
The tinkerboard S has run flawless for me so hope they will fix this issue in the future.

junologue, 2020-07-26:
I am using version 4.4 with on Tinkerboard S without any issues. Works just as fine as 3.13

jdbrookes, 2020-07-26:
> Right now I am using version 4.4 with on Tinkerboard S without any issues. Works just as fine as 3.13
> 
> Anyone who knows how to disable and remove the OS update notification though?
> 
> [image]

I think the issue is with booting directly from eMMC, without an SD card. It would be great if support for this had been reintroduced? 

I've not yet bit the bullet and moved everything to microsd, and then done the datactl command. If anyone has experience with doing this on a tinkerboard s and would be able to do a brief writeup it would be appreciated! I think I would need to wipe the eMMC before flashing to microsd but I'm not sure how to do this.

junologue, 2020-07-26:
I am running on emmc, hassos 4.4
Tinkerboard S

jdbrookes, 2020-07-31:
> I am running on emmc, hassos 4.4
> Tinkerboard S

I don't think I'm brave enough to try updating to the latest version!

poelsevev, 2020-08-04:
I installed the versions backwards until I found one that worked, and for me that was 3.9, but I can't remember if 4.4 was out then. But all versions worked with SD card. So that's why I gues thought 4.x is the issue with emmc boot. junologue, did you do anything else beside run the install for tinkerboard??

junologue, 2020-08-04:
Nope, i just did a clean install, then restored previous snapshots :)

pbalogh77, 2020-08-09:
I'm struggling with the same. Why is this issue closed?

frenck, 2020-08-09:
@pbalogh77 Because the Tinkerboard S is not a supported board. For supported hardware see:

<https://github.com/home-assistant/architecture/blob/master/adr/0015-home-assistant-os.md#supported-boardshardwaremachines>

pbalogh77, 2020-08-09:
The Tinkerboard S is just a Tinkerboard with eMMC and has been a supported board for years. Is this a bug that needs fixing (in which case I'm happy to help) or a conscious decision to abandon support, in which case I should either stay with v4.4 or move to a new HW? At least it would have been nice to communicate this as a breaking change :-(

pbalogh77, 2020-08-10:
> The board isn’t made anymore, and no one in the core team has access to
> one. Therefore we cannot test it or support it. However, the project is
> open source, so feel free to use it the way you like.

Thanks, I understand. Maybe I should be moving to a different HW sooner or
later...

jpjcrossi, 2020-08-10:
I didn't know that board was discontinued. I tried looking for more information about this but I didn't find it.
It's a pity ... :(

poelsevev, 2020-08-18:
is this issue with tinkerboard emmc fixed? i notised there where something doen regarding tinker s on release 4.7

dyrvigk, 2020-08-21:
hi frenck i dont understand you say the tinker board s i discontinued but it dosent say that anywhere and i can find more than a 100 places just here in denmark where you can buy 

kenn

zerblatt007, 2020-08-26:
Amazon seems to think the Tinker board S is not discontinued:
https://www.amazon.com/ASUS-Tinker-Board-connectivity-Motherboards/dp/B00FS83U42?th=1
"Is Discontinued By Manufacturer | No"

If someone donates a board to a developer, would that make them supported or is this something you guys rather would not do?

agners, 2020-09-03:
I do have one of those board now, I am going to have a look.

agners, 2020-09-09:
The good news is the fix landed in `dev`. The bad news is I broke the Tinker build in the process :-(

I already made a merge request to fix it (https://github.com/home-assistant/operating-system/pull/854). So a fixed version will come at one point :-)

jpjcrossi, 2020-09-09:
Great job. I'll test it as soon as available  that new feature.
Many thanks! 👍

zerblatt007, 2020-09-09:
Really glad to hear this Stefan!
This will be Dev 5 build 3 per chance?

junologue, 2020-09-09:
Nice Agners!

So this means Tinkerboard S will soon be supported in an official build again?

pvizeli, 2020-09-12:
Fix 5.2

closed 2020-09-12 by pvizeli

anthfarr, 2020-12-24:
Hi - sorry to comment on a closed issue however I'm confused. This issue is marked as fixed, however the official docs do not mention tinkerboard S is supported (only tinkerboard is mentioned) - Does this mean I can use tinkerboard S (using emmc) or not?

poelsevev, 2021-01-01:
> Hi - sorry to comment on a closed issue however I'm confused. This issue is marked as fixed, however the official docs do not mention tinkerboard S is supported (only tinkerboard is mentioned) - Does this mean I can use tinkerboard S (using emmc) or not?

-----

You can use the Tinkerboard S. I have Been using it for a couple of years and works great with a large nummer of z-wave and zigbee devices and a gold bunch of automations running. I just keep a backup on a SD card just inn case something happens during updates.

PR #658 (2020-05-08): Remove wrong support on the docs for Tinker S
Fix: #650