home-assistant/operating-system #654: Boot from USB on Odroid N2

pepeEL, 2020-05-06:
Hi. Is any plans when will be implement to operating system for Odroid N2 option to boot from USB ? Now we can use SDCard and eMMc. Good option will be also USB to use USB SSD disk which is sometime faster than eMMC.

pvizeli, 2020-05-07:
Read https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md

it will not more support as that

closed 2020-05-07 by pvizeli

pepeEL, 2020-05-17:
I read but dont understand... I am not an exprert.
Summary
It is no possible to implement boot from usb on this Odroid N2?
It is probably the Best hasrdwear for Home Assistant. But we must use eMMC or MicroSD. But SSD is better and faster... 
Also boot is not possible by Petiboot on Odroid N2?

pepeEL, 2020-05-19:
please explaine me ....

bramkragten, 2020-05-29:
You can move your data partition to an external USB hard drive with `datactl move /dev/xxx`

pepeEL, 2020-05-29:
Better will be implement option to install on USB as example for RPI and boot from USB

bramkragten, 2020-05-29:
No

pepeEL, 2020-05-29:
But it is technical not possible ?

frenck, 2020-05-29:
Your question was answered now, multiple times. Locking this issue.