home-assistant/operating-system #660: Cannot Deploy OVA onto ESXi 6.7

dancwilliams, 2020-05-09:
**HassOS release with the issue:**

hassos_ova-4.6.ova

**Supervisor logs:**

N/A

**Journal logs:**

N/A

**Kernel logs:**

N/A

**Description of problem:**

I am attempting to import the beta OVA image onto a VMWare ESXi 6.7 host I have in my lab.  I am receiving the following error when attempting to deploy:

`Issues detected with selected template. Details: - 23:5:PARSE_ERROR: Parse error: Undeclared namespace prefix "vmw" (for attribute "osType") at [row,col,system-id]: [23,93,"descriptor.ovf"].`

I am using the following VMware Version: `VMware ESXi, 6.7.0, 14320388`

If there are any further details needed just let me know.  Thanks!

fabaff, 2020-05-14:
Ok, just to be sure, the hashes in the ova file are a match.

```bash
$ tar -xvf hassos_ova-4.6.ova
home-assistant.ovf
home-assistant.vmdk
home-assistant.mf
$ cat home-assistant.mf 
SHA256 (home-assistant.ovf) = aa59cd56f4712c871b507e5d148c273bb071734934d9747dbe44e90cd412123b
SHA256 (home-assistant.vmdk) = 6d8895600afcf44144100c56a9d61c5451dfd9673dcb91c74b331eaf307fc17f
$ sha256sum home-assistant.ovf home-assistant.vmdk
aa59cd56f4712c871b507e5d148c273bb071734934d9747dbe44e90cd412123b  home-assistant.ovf
6d8895600afcf44144100c56a9d61c5451dfd9673dcb91c74b331eaf307fc17f  home-assistant.vmdk
```

Looking at the `home-assistant.ovf` file. It seems that there is a mismatch between the `id` and the `osType`. I don't know if ESX 6.7 has a strong validation or if those details are right/still correct: 

```text
    <OperatingSystemSection ovf:id="102" vmw:osType="otherLinux64Guest" ovf:required="false">
```

The id 102 is "otherGuest64" according to a pretty old document I found. The id 101 is "otherLinux64Guest"

```text
    <OperatingSystemSection ovf:id="101" vmw:osType="otherLinux64Guest" ovf:required="false">
```

MaartenMees, 2020-05-15:
Hi,

I having also a problem when deploying the ova to esxi 6.7. with ovftools.

acederlund, 2020-05-16:
I'm having the same issue as well.

ESXi version: 6.7.0 Update 2 (Build 13006603)

It's stuck at "Extracting OVA, this could take some time..." when using the Host UI.

Tried to unpack the OVA and upload it, but then I cannot get past the target datastore selection dialog. Unpacking the OVA and editing ovf:id="102" to "101" in the ovf-file as @fabaff pointed out does not help either.

Any pointers to get in the righ direction is highly appreciated.

MaartenMees, 2020-05-17:
Hi,

I just used the vmdk.

Greetings.

apoc4lyps, 2020-05-17:
Hi, I was able to modify the ovf file to work on VMware 7.0 (6.5 / 6.7 should also work)

```xml
<?xml version='1.0' encoding='UTF-8'?>
<Envelope xmlns="http://schemas.dmtf.org/ovf/envelope/1" xmlns:ovf="http://schemas.dmtf.org/ovf/envelope/1" xmlns:vmw="http://www.vmware.com/schema/ovf" xmlns:rasd="http://schemas.dmtf.org/wbem/wscim/1/cim-schema/2/CIM_ResourceAllocationSettingData" xmlns:vssd="http://schemas.dmtf.org/wbem/wscim/1/cim-schema/2/CIM_VirtualSystemSettingData">
  <References>
    <File ovf:id="file1" ovf:href="home-assistant.vmdk"/>
  </References>
  <DiskSection>
    <Info>List of the virtual disks used in the package</Info>
    <Disk ovf:capacity="6442450944" ovf:diskId="vmdisk1" ovf:fileRef="file1" ovf:format="http://www.vmware.com/interfaces/specifications/vmdk.html#streamOptimized" ovf:capacityAllocationUnits="byte" />
  </DiskSection>
  <NetworkSection>
    <Info>Logical networks used in the package</Info>
    <Network ovf:name="Bridged">
      <Description>Logical network used by this appliance.</Description>
    </Network>
  </NetworkSection>
  <VirtualSystem ovf:id="Home Assistant">
    <Info>A virtual machine</Info>
    <ProductSection>
      <Info>Meta-information about the installed software</Info>
      <Product>Operating-System</Product>
      <ProductUrl>https://home-assistant.io</ProductUrl>
    </ProductSection>
    <OperatingSystemSection ovf:id="100" vmw:osType="other3xLinux64Guest">
      <Info>The operating system installed</Info>
      <Description>Other 3.x or later Linux (64-bit)</Description>
    </OperatingSystemSection>
    <VirtualHardwareSection>
      <Info>Virtual hardware requirements for a virtual machine</Info>
      <System>
        <vssd:ElementName>Virtual Hardware Family</vssd:ElementName>
        <vssd:InstanceID>0</vssd:InstanceID>
        <vssd:VirtualSystemIdentifier>Home Assistant</vssd:VirtualSystemIdentifier>
        <vssd:VirtualSystemType>vmx-13</vssd:VirtualSystemType>
      </System>
      <Item>
        <rasd:Caption>2 virtual CPU</rasd:Caption>
        <rasd:Description>Number of virtual CPUs</rasd:Description>
        <rasd:ElementName>2 virtual CPU</rasd:ElementName>
        <rasd:InstanceID>1</rasd:InstanceID>
        <rasd:ResourceType>3</rasd:ResourceType>
        <rasd:VirtualQuantity>2</rasd:VirtualQuantity>
        <vmw:CoresPerSocket ovf:required="false">2</vmw:CoresPerSocket>
      </Item>
      <Item>
        <rasd:AllocationUnits>MegaBytes</rasd:AllocationUnits>
        <rasd:Caption>2048 MB of memory</rasd:Caption>
        <rasd:Description>Memory Size</rasd:Description>
        <rasd:ElementName>2048 MB of memory</rasd:ElementName>
        <rasd:InstanceID>2</rasd:InstanceID>
        <rasd:ResourceType>4</rasd:ResourceType>
        <rasd:VirtualQuantity>2048</rasd:VirtualQuantity>
      </Item>
      <Item>
        <rasd:Address>0</rasd:Address>
        <rasd:Caption>sataController0</rasd:Caption>
        <rasd:Description>SATA Controller</rasd:Description>
        <rasd:ElementName>sataController0</rasd:ElementName>
        <rasd:InstanceID>3</rasd:InstanceID>
        <rasd:ResourceSubType>vmware.sata.ahci</rasd:ResourceSubType>
        <rasd:ResourceType>20</rasd:ResourceType>
      </Item>
      <Item>
        <rasd:Address>0</rasd:Address>
        <rasd:Caption>usb</rasd:Caption>
        <rasd:Description>USB Controller (XHCI)</rasd:Description>
        <rasd:ElementName>usb</rasd:ElementName>
        <rasd:InstanceID>4</rasd:InstanceID>
        <rasd:ResourceSubType>vmware.usb.xhci</rasd:ResourceSubType>
        <rasd:ResourceType>23</rasd:ResourceType>
      </Item>
      <Item>
        <rasd:AddressOnParent>0</rasd:AddressOnParent>
        <rasd:Caption>disk1</rasd:Caption>
        <rasd:Description>Disk Image</rasd:Description>
        <rasd:ElementName>disk1</rasd:ElementName>
        <rasd:HostResource>/disk/vmdisk1</rasd:HostResource>
        <rasd:InstanceID>5</rasd:InstanceID>
        <rasd:Parent>3</rasd:Parent>
        <rasd:ResourceType>17</rasd:ResourceType>
        <vmw:Config ovf:required="false" vmw:key="backing.writeThrough" vmw:value="false"/>
      </Item>
      <Item>
        <rasd:AutomaticAllocation>true</rasd:AutomaticAllocation>
        <rasd:Caption>eth0</rasd:Caption>
        <rasd:Description>Ethernet adapter</rasd:Description>
        <rasd:Connection>Bridged</rasd:Connection>
        <rasd:ElementName>eth0</rasd:ElementName>
        <rasd:InstanceID>6</rasd:InstanceID>
        <rasd:ResourceSubType>E1000</rasd:ResourceSubType>
        <rasd:ResourceType>10</rasd:ResourceType>
      </Item>
      <vmw:Config ovf:required="false" vmw:key="firmware" vmw:value="efi"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="cpuid.coresPerSocket" vmw:value="2"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="svga.present" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.speed" vmw:value="2"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.present" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.deviceType" vmw:value="hub"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.port" vmw:value="1"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.parent" vmw:value="-1"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="ctkEnabled" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="sata0:0.ctkEnabled" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.present" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.deviceType" vmw:value="hid"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.port" vmw:value="0"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.parent" vmw:value="-1"/>
    </VirtualHardwareSection>
    
  </VirtualSystem>
</Envelope>
```

acederlund, 2020-05-17:
Thanks @apoc4lyps!

I tried your setting, and got this error:
"Line 86: Unsupported element 'Connection'"

Will try and fiddle some and see if I can get it working :-)

acederlund, 2020-05-17:
So, I moved
`<rasd:Connection>Bridged</rasd:Connection>`
before
`<rasd:Description>Ethernet adapter</rasd:Description>`

and that seems to have worked.

closed 2020-05-18 by pvizeli

tiempososcuros, 2020-06-06:
I have tried to deploy the hassos_ova-4.9 version and I get the same result in ESXi 6.7.
The error message:
`Issues detected with selected template. Details: - 23:5:PARSE_ERROR: Parse error: Undeclared namespace prefix "vmw" (for attribute "osType") at [row,col,system-id]: [23,74,"descriptor.ovf"].`

PR #677 (2020-05-18): Fix vmware OS-Type for OVA
Fix: #660