home-assistant/operating-system #661: Deconz coordinator couldn't start

Nemesis24, 2020-05-10:
Home Assistant 0.109.6
Operating system version 4.6

**Description of problem:**
```
2020-05-10 20:49:16 ERROR (MainThread) [homeassistant.components.zha.core.gateway] Couldn't start Deconz coordinator
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
    res = await self.application_controller.startup(auto_form=True)
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/zigbee/application.py", line 52, in startup
    self.version = await self._api.version()
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 403, in version
    (self._proto_ver,) = await self[NetworkParameter.protocol_version]
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 368, in read_parameter
    r = await self._command(Command.read_parameter, 1 + len(data), param, data)
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 276, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

Deconz coordinator couldn't start
Conbee
/dev/ttyS0

frenck, 2020-05-10:
Hi, @Nemesis24!

This is the issue tracker for the Home Assistant Operating System. Your stack trace shows an issue with the Home Assistant ZHA integration.

Please file an issue on the Home Assistant Core issues tracker for this:

<https://github.com/home-assistant/core/issues>

Thanks! 👍

closed 2020-05-10 by frenck

Nemesis24, 2020-05-11:
Ok but i'm not sure it was a zha issue, because after dowgrading, the coordinator works again... i try again with operating system 4.6 and the coordinator don't start.

Nemesis24, 2020-05-12:
Ok i find the solution i had to revert this 
https://github.com/home-assistant/operating-system/issues/480#issuecomment-532299787

In fact:

i must have now

```
dtoverlay=pi3-disable-bt
enable_uart=1
```

And change /dev/ttyS0 for /dev/ttyAMA0 in zha configuration

https://github.com/home-assistant/core/issues/35489