home-assistant/operating-system #662: Cannot upgrade from HassOS 3.13 to 4.6 on ODroid HC1 (XU4) with external data partition

TonyApuzzo, 2020-05-10:
**HassOS release with the issue:**

HassOS 4.6 / hassos_odroid-xu4-4.6.img.gz

**Supervisor logs:**

Not Available

**Journal logs:**

Not Available

**Kernel logs:**

This is the output of the serial console when booting from 4.6 on my HC1 with the external SSD attached:
NOTE: The first time I encountered this problem I got an error saying that hassos_cli was not running, but this time when I captured the output, the error message refers to a container hash code instead of the name.

https://gist.github.com/TonyApuzzo/646e8074e6fc8a0301cbaa80dcede397

**Description of problem:**
Cannot upgrade from HassOS 3.13 to 4.6 on ODroid HC1 with external data partition.

Note that I **can** boot and use HassOS 4.6 without the external drive -- I was able to restore from a snapshot and have everything work all-in-one on the sdcard.

The issue seems to be something wrong with locating the either the data partition or docker images. When I boot up with the external SSD plugged into the USB, HassOS seems to locate the data partition, but something goes wrong and seemingly none of the docker containers can be launched.  I don't know how to get more information than what I've provided since I cannot log in.

I tried supplying `authorized_keys` in /boot/CONFIG and it works to enable sshd, but SSH to port 22222 fails with the same error as on the serial console.

Here is the information from my working 3.13 environment:

```
# lsblk -o name,mountpoint,label,size,uuid
NAME       MOUNTPOINT LABEL            SIZE UUID
sda                                  223.6G
|-sda1     /mnt/data  hassos-data    167.7G e9535aea-c9e2-42a9-a91b-dd6ef5365543
`-sda2                                55.9G 1096115b-ef70-4198-95e2-4bdd64cc06b0
mmcblk1                               59.5G
|-mmcblk1p1
|          /mnt/boot  hassos-boot       32M DB45-285A
|-mmcblk1p2
|                                        1K
|-mmcblk1p3
|          /mnt/overl hassos-overlay    96M 4b59f6a4-a014-4fc6-9f35-6e3821b9cde5
|-mmcblk1p4
|                     hassos-data        1G 50d3ca72-24b0-44d1-bb46-204c5f142e65
|-mmcblk1p5
|                     hassos-kernel     24M 293abb78-8618-4516-bea6-3a7d54372e51
|-mmcblk1p6
|          /                           256M
|-mmcblk1p7
|                                       24M
|-mmcblk1p8
|                                      256M
`-mmcblk1p9
                                         8M
zram0      [SWAP]     hassos-zramswap
                                     497.9M 015638cc-fafd-45e1-a3b0-ea1f32e0b9da
zram1      /var       hassos-var        32M 812a1c62-5afa-4ebe-b2ae-34c0e2b8e971
zram2      /tmp       hassos-tmp        16M 80f8e2c2-dec7-4b0e-9bd5-bef5366427ab
```

Megachip, 2020-05-11:
Same on RPi3, out of the box installation. See [here](https://community.home-assistant.io/t/update-hassos-3-13-4-6-failing-on-rpi-3/194736)

TonyApuzzo, 2020-05-15:
I will try that for a manual upgrade -- but it won't change the situation when trying to do an automated upgrade.  In the relatively recent past (from 3.0-betas until now) the HassOS automatic upgrades have worked with 2 partitions named `hassos-data` and it has transparently re-attached the external data partition and ignored the pre-existing one on the sd card.

dpackham, 2020-05-17:
I have the same problem running on a ProMox VM with the qcow2 image.  it says I am running the 3.13 OS and that I need to upgrade to 4.6. when I hit the upgrade button it takes a bit and reboots but says the same thing when rebooted

pvizeli, 2020-05-18:
That is not our issue if you offload data partition in wrong why before instead to help to make that feature what we have now in REL-4

closed 2020-05-18 by pvizeli

TonyApuzzo, 2020-05-18:
I was using the suggested way that was working in 3.x -- any suggestion for how to safely migrate? What is a better forum than this issue to discuss?