home-assistant/operating-system #663: HassOS - cannot be restarted on ESXi

Dinth, 2020-05-11:
**HassOS release with the issue:**

**Kernel logs:**

I do apologize for the screenshot, but im getting a segfault before the system boots up.
[image]

**Description of problem:**

When i start my ESXi host, HassOS starts fine, but whenever i want to reboot Hass, whether using shutdown -r now, ha host reboot or through ESXi, HassOS starts segfaulting on boot. The only way of fixing it is restarting whole ESXi host. 

Ive got multiple other VMs running Linux, BSD and Windows on same host, and those are completely unaffected, can be freely rebooted, dont show any signs of instability, before or after HassOS crash.

ericlamotte, 2020-05-14:
I'm not so sure that this is specific to VMs on ESxi. I am noticing similar behavior with the NUC image running on bare metal on my NUC (https://community.home-assistant.io/t/hassos-3-13-installation-becomes-corrupted-on-nuc-after-supervisor-reboot/194937/4)

pvizeli, 2020-05-18:
Can you test 4.6?

Dinth, 2020-05-19:
Sure i can try. 
Im assuming that i need to rebuild VM from scratch, cannot just update it from within the OS? 
Can i use OVA then, is it working again?

pvizeli, 2020-05-19:
`ha os update --version 4.6`

Dinth, 2020-05-21:
> `ha os update --version 4.6`

Thanks for that. So far i was unable to reproduce the problem on 4.6.
This being said i see a lot of glitches with ver 4.6 - for example im getting an update dialog in Supervisor tab asking me to update 4.6.to 3.13.
Also i cannot update or snapshot addons.

pvizeli, 2020-05-25:
Look like somethings with network adapter. Do you have also this issue, if you use our internal shutdown/reboot commands from CLI because that would be the correct way for a clean shutdown.

Dinth, 2020-05-25:
Im using VMXNET 3 as according to many guides it should be fully working for HassOS (after converting HASSOS VMDK) and also i its the only one which utilizes my 10GbE connectivity. 
Im getting this error when rebooting for CLI or "Reboot host" from GUI.

Dinth, 2020-07-25:
Just wanted to update, that recently, after HASSOS updates, the issue stopped occuring on every restart, but rarely (maybe every tenth restart?) its still happening.

closed 2020-10-12 by stale[bot]