home-assistant/operating-system #676: Enable 9p VIRTIO kernel module

JumpMaster, 2020-05-17:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
As part of a migration from a Supervised Linux install to HASSOS on KVM I am missing the ability to share a directory with the host os.  This is possible within KVM but the OS is missing the required 9p kernel module.  This leads to the following error when trying to mount:

mount -t 9p -o trans=virtio,version=9p2000.L hostshare /tmp/host_files
mount: unknown filesystem type '9p'

JumpMaster, 2020-05-17:
I've been thinking more about this and it's probably a lot more work then I originally anticipated to have this fully supported.  ie. Been able to mount directories from within the supervisor ui.

closed 2020-05-21 by JumpMaster

xiki808, 2021-07-01:
I would love to see this. I am currently trying to setup HAOS in an LXD VM and it works, but I cannot run any commands inside the VM from my bash script running at the host, since the agent is missing in the OS

PR #2145 (2022-10-02): Add LXD VM Guest Agent Support
Fixes #676