home-assistant/operating-system #678: Odroid N2 - eMMC

pvizeli, 2020-05-18:
I'm having a similar problem with the Odroid N2

```
G12B:BL:6e7c85:7898ac;FEAT:E0F83180:2000;POC:F;RCY:0;EMMC:0;READ:0;0.0
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

L0:00000000
L1:00000703
L2:00008067
L3:04000000
B2:00002000
B1:e0f83180

TE: 437975

BL2 Built : 06:17:13, Jun 28 2019. g12b gf0505d7-dirty - qi.duan@droid13

Board ID = 4
Set A53 clk to 24M
Set A73 clk to 24M
Set clk81 to 24M
A53 clk: 1200 MHz
A73 clk: 1200 MHz
CLK81: 166.6M
smccc: 0006f70e
eMMC boot @ 0
sw8 s
DDR driver_vesion: LPDDR4_PHY_V_0_1_14 build time: Jun 28 2019 06:17:09
board id: 4
Load FIP HDR from eMMC, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from eMMC, src: 0x00054200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from eMMC, src: 0x0002c200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
00000000
emmc switch 1 ok
00000000
emmc switch 2 ok
fastboot data verify
verify result: 255
Cfg max: 1, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from eMMC, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0
00000000
emmc switch 0 ok
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from eMMC, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

R0_RxClkDly_Margin==106 ps 9
R0_TxDqDly_Margi==106 ps 9

R1_RxClkDly_Margin==0 ps 0
R1_TxDqDly_Margi==0 ps 0

 dwc_ddrphy_apb_wr((0<<20)|(2<<16)|(0<<12)|(0xb0):0001 
2D training succeed
auto size-- 65535DDR cs0 size: 2048MB
DDR cs1 size: 2048MB
DMC_DDR_CTRL: 00600024DDR size: 3928MB
cs0 DataBus test pass
cs1 DataBus test pass
cs0 AddrBus test pass
cs1 AddrBus test pass
 pre test  bdlr_100_average==450 bdlr_100_min==450 bdlr_100_max==450 bdlr_100_cur==450
 aft test  bdlr_100_average==450 bdlr_100_min==450 bdlr_100_max==450 bdlr_100_cur==450
non-sec scramble use zero key
ddr scramble enabled

100bdlr_step_size ps== 450
result report
boot times 0Enable ddr reg access
00000000
emmc switch 3 ok
Authentication key not yet programmed
get rpmb counter error 0x00000007
00000000
emmc switch 0 ok
Load FIP HDR from eMMC, src: 0x00010200, des: 0x01700000, size: 0x00004000, part: 0
Load BL3X from eMMC, src: 0x00060200, des: 0x01750000, size: 0x0007b400, part: 0
0.0;M3 CHK:0;cm4_sp_mode 0

E30HDR

MVN_1=0x00000000

MVN_2=0x00000000

[Image: g12b_v1.1.3375-8f9c8a7 2019-01-24 10:44:46 guotai.shen@droid11-sz]

OPS=0x40

ring efuse init

chipver efuse init

29 0a 40 00 01 0a 1c 00 00 19 34 37 57 4e 4b 50 

[0.019858 Inits done]

secure task start!
high task start!
low task start!
run into bl31
NOTICE:  BL31: v1.3(release):ab8811b
NOTICE:  BL31: Built : 15:03:31, Feb 12 2019
NOTICE:  BL31: G12A normal boot!
NOTICE:  BL31: BL33 decompress pass
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2020.01 (Apr 29 2020 - 17:05:24 +0000) odroid-n2

DRAM:  3.8 GiB
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
In:    serial@3000
Out:   serial@3000
Err:   serial@3000
Net:   eth0: ethernet@ff3f0000
Card did not respond to voltage select!
unable to select a mode
switch to partitions #0, OK
mmc1(part 0) is current device
** No partition table - mmc 1 **
MMC Device 2 not found
no mmc device at slot 2
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 1 USB Device(s) found
usb - USB sub-system

Usage:
usb start - start (scan) USB controller
usb reset - reset (rescan) USB controller
usb stop [f] - stop USB [f]=force stop
usb tree - show USB device tree
usb info [dev] - show available USB devices
usb test [dev] [port] [mode] - set USB 2.0 test mode
    (specify port 0 to indicate the device's upstream port)
    Available modes: J, K, S[E0_NAK], P[acket], F[orce_Enable]

ethernet@ff3f0000 Waiting for PHY auto negotiation to complete.... done
Speed: 1000, full duplex
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/01-00-1e-06-42-11-6c
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-meson-w400
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-meson
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
Speed: 1000, full duplex
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Config file not found
Speed: 1000, full duplex
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
HassOS> 
```

_Originally posted by @catchra in https://github.com/home-assistant/operating-system/issues/675#issuecomment-629831686_

pvizeli, 2020-05-18:
@catchra look like you boot partition is wrong flashed

```
** No partition table - mmc 1 **
```

There is something wrong on your flash process. This topic is for people to find out why there flash not work.

I can say, it works on all our test devices and my productive devices which are 10 devices. I use etcher 1.5.80 on Windows 10 with the USB 3 flasher on a USB 3 port of my computer.

pvizeli, 2020-05-18:
[image]

catchra, 2020-05-21:
Hi @pvizeli i have tried doing as you have said but i am still getting the same problem.
here are some pics from gparted and fdisk.

[image]

[image]

i have tried etcher on windows10 and on Linux and i am using the USB 3 flasher

pvizeli, 2020-05-25:
What is the output of the serial console?

catchra, 2020-05-25:
ok this is weird so i was using an old version of etcher, when i made my 1st post after reading your post i updated etcher but to version 1.5.91 and all i got from the serial console was this

`G12B:BL:6e7c85:7898ac;FEAT:E0F83180:2000;POC:F;RCY:0;EMMC:0`
repeating.

so just now i tried balenaEtcher-Portable-1.5.80 and it is working now.

so Etcher seems to be the problem. at the time of this post their is a new version of Etcher now at version 1.5.92. i will give this version a try later and see if it gives the same problems

closed 2020-06-03 by balloob