home-assistant/operating-system #683: Wrong HassOS version reported by Proxmox RAW and QCOW2 images

insanemal, 2020-05-21:
**HassOS release with the issue:**
~ $ ha info
arch: amd64
channel: stable
hassos: "4.6"
homeassistant: 0.109.6
hostname: homeassistant
logging: info
machine: qemux86-64
supervisor: "222"
supported_arch:
- amd64
- i386
timezone: Australia/Brisbane

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Running the proxmox release it seems it doesn't know which version of hassOS it's running and reports version 4.6.

I'm not sure if it's an issue or not but I am starting to set things up and would prefer to know if this is going to be an issue before I get too far into the setup

I'll get other logs if need be, but I've recreated from the proxmox raw and qcow2 image more than once and it always reports 4.6 for hassos

Thanks in advance

peng1can, 2020-05-21:
That confused me at first, too.  But, it's showing you that it's running HassOS (beta) 4.6 which is correct if you note that the installation page says it's a beta and you look at the list of beta releases: https://github.com/home-assistant/operating-system/releases
One of the developers mentioned on the forum that because it's not the same as the release version, it always thinks there is an upgrade available, but since the "upgrade" is older than what you're running, that notification is just something you have to deal with for now.

frenck, 2020-05-21:
HassOS 4.6 is correct and not an error. It is a beta image and thus OS.
However, if your system is not on the beta channel, it will offer an upgrade (which is actually a downgrade) to stable.

Alternatively you can join the beta channel in Supervisor -> System -> Join beta channel.

closed 2020-05-21 by frenck

insanemal, 2020-05-21:
Thanks all and my apologies for not realizing sooner!