home-assistant/operating-system #684: Losing LAN IP on ESXi

nonyhaha, 2020-05-25:
**Home Assistant release with the issue:**
0.110.2

**Operating environment (HassOS/Generic):**
Home Assistant Supervised;
ESXi hosted vm, followed tutorial from https://community.home-assistant.io/t/hass-io-on-vmware-esxi-6-7-step-by-step/151419
**Supervisor logs:**

https://drive.google.com/open?id=1Nv3v9QItk2Ax3opBHweMZVYudszCe5o0 

**Description of problem:**
Also described in here: https://community.home-assistant.io/t/hassio-on-esxi-losing-ip-address/198955 

a few days ago I moved from an older install (running on virtualbox) to a fresh new install on esxi. I backed up /config folder to keep as much data from the old installation as possible. 
After I setup everything and checked that all works, I found out that my server was inaccessible. I checked local console and the server is running. But with the local IP gone. I set the static IP on a pfsense machine. 
After the first incidents, I tried setting a static IP using USB network config file. 
Still, the local IP goes missing and I can't access the server from LAN. 
core restart command in console completes successfully but IP does not get back up. 
Installing non supervised on debian CLI host will end up in inaccessibility too, only that the debian host is holding its LAN IP. 
Simple reset from ESXi will not resolve the issue, only power off command, wait for hassos guest to power off, then start it again. 
I really do not know how to get the supervisor logs. the command provided in the tutorial is not working in the console. In a ssh session, it returns what I have pasted in the supervisor logs.txt. 
[core log.txt](https://github.com/home-assistant/supervisor/files/4676550/core.log.txt)
[host log.txt](https://github.com/home-assistant/supervisor/files/4676551/host.log.txt)
[supervisor logs.txt](https://github.com/home-assistant/supervisor/files/4676552/supervisor.logs.txt)

nonyhaha, 2020-05-27:
I just started another vm running windowa aervwr to test if esxi is the culprit.

pvizeli, 2020-06-08:
We use internal UTC

closed 2020-06-08 by pvizeli