home-assistant/operating-system #687: ESXI 6.7 - Stuck at Extracting OVA, this could take some time...

sumur1au, 2020-05-26:
**HassOS release with the issue:**
N/A

**Supervisor logs:**
N/A

**Journal logs:**
N/A

**Kernel logs:**

**Description of problem:**
Currently using EsXi 6.7 and Hassos 4.7 OVA is stuck at Extracting OVA, this could take some time...

JumpMaster, 2020-05-27:
An ova is a zipped ovf, vmdk, etc. Can you unzip the ova and import the ovf and vmdk?

n3o59hf, 2020-05-28:
I had similar issue with ESXi 7.0 and 4.6 OVA. Extracting OVA and then removing all lines starting with `<vmw:ExtraConfig ` and item for Ethernet from ovf worked for me (I then re-packed OVA without mf and was able to deploy it successfully). Afterwards I adjusted needed configs and it booted successfully and still works without problems. I don't remember if all lines caused problem or just few of those.

Here are all lines I removed:
```      <Item>
        <rasd:AutomaticAllocation>true</rasd:AutomaticAllocation>
        <rasd:Caption>eth0</rasd:Caption>
        <rasd:Description>Ethernet adapter</rasd:Description>
        <rasd:Connection>Bridged</rasd:Connection>
        <rasd:ElementName>eth0</rasd:ElementName>
        <rasd:InstanceID>6</rasd:InstanceID>
        <rasd:ResourceSubType>E1000</rasd:ResourceSubType>
        <rasd:ResourceType>10</rasd:ResourceType>
      </Item>
```
and
```
      <vmw:ExtraConfig ovf:required="false" vmw:key="cpuid.coresPerSocket" vmw:value="2"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="svga.present" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.speed" vmw:value="2"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.present" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.deviceType" vmw:value="hub"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.port" vmw:value="1"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.parent" vmw:value="-1"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="ctkEnabled" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="sata0:0.ctkEnabled" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.present" vmw:value="TRUE"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.deviceType" vmw:value="hid"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.port" vmw:value="0"/>
      <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.parent" vmw:value="-1"/>
```

vgracanin2, 2020-06-01:
> I had similar issue with ESXi 7.0 and 4.6 OVA. Extracting OVA and then removing all lines starting with `<vmw:ExtraConfig ` and item for Ethernet from ovf worked for me (I then re-packed OVA without mf and was able to deploy it successfully). Afterwards I adjusted needed configs and it booted successfully and still works without problems. I don't remember if all lines caused problem or just few of those.
> 
> Here are all lines I removed:
> 
> ```
>       <Item>
>         <rasd:AutomaticAllocation>true</rasd:AutomaticAllocation>
>         <rasd:Caption>eth0</rasd:Caption>
>         <rasd:Description>Ethernet adapter</rasd:Description>
>         <rasd:Connection>Bridged</rasd:Connection>
>         <rasd:ElementName>eth0</rasd:ElementName>
>         <rasd:InstanceID>6</rasd:InstanceID>
>         <rasd:ResourceSubType>E1000</rasd:ResourceSubType>
>         <rasd:ResourceType>10</rasd:ResourceType>
>       </Item>
> ```
> 
> and
> 
> ```
>       <vmw:ExtraConfig ovf:required="false" vmw:key="cpuid.coresPerSocket" vmw:value="2"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="svga.present" vmw:value="TRUE"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.speed" vmw:value="2"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.present" vmw:value="TRUE"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.deviceType" vmw:value="hub"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.port" vw:value="1"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:1.parent" vmw:value="-1"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="ctkEnabled" vmw:value="TRUE"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="sata0:0.ctkEnabled" vmw:value="TRUE"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.present" vmw:value="TRUE"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.deviceType" vmw:value="hid"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.port" vmw:value="0"/>
>       <vmw:ExtraConfig ovf:required="false" vmw:key="usb:0.parent" vmw:value="-1"/>
> ```

I did the same. Thanks.

pvizeli, 2020-06-03:
Fix with next update

closed 2020-06-03 by pvizeli