home-assistant/operating-system #688: Inconsistent upgrade advice from 4.7 to 3.13

PieBru, 2020-05-26:
**HassOS release with the issue:** 4.7 OVA

Frontend version: 20200519.5 - latest
arch | x86_64
chassis | vm
dev | false
docker | true
hassio | true
host_os | HassOS 4.7
installation_type | Home Assistant
os_name | Linux
os_version | 5.4.42
python_version | 3.7.7
supervisor | 224
timezone | Europe/Rome
version | 0.110.3
virtualenv | false

**Supervisor logs:**

All green :)

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Supervisor dashboard notifies and offers to upgrade from 4.7 to 3.13

[image]

Adorem, 2020-05-26:
https://github.com/home-assistant/operating-system/issues/686#issuecomment-633980495

pvizeli, 2020-05-27:
Yeah, 4.7 is not official released

PieBru, 2020-05-27:
Sorry for replying after frenck closed, I'm writing because I'm a great fan of the HASS ecosystem and I am trying the HASS-as-an-appliance SBC & VM concept.

**It's may be silly matter of aesthetics**, but particularly now that HassIO (docker) has been just deprecated, IMO who approaches HassOS without knowing the great work running underneat will be confused by the github project release naming convention.

Yes, the [official doc](https://www.home-assistant.io/hassio/installation/) points to the "released" 3.13 files (except "beta" that points to 4.6), but who visits the [github releases project page](https://github.com/home-assistant/operating-system/releases) reads:
`HassOS Release-4 build 7 (Stable) - release-drafter released this yesterday`
and below:
`HassOS Release-3 build 13 (Stable) - release-drafter released this on Apr 15`

Given this naming, a HassOS noob will obviously choose the 4.7, therefore getting a confusing advice from the WEB UI.

Just my 2 cents...
Thank you all for your great project,
Piero