home-assistant/operating-system #689: DHCP-provided NTP not used

DaAwesomeP, 2020-05-26:
**HassOS release with the issue:**

HassOS 3.13

**Supervisor logs:**

N/A

**Journal logs:**

N/A

**Kernel logs:**

N/A

**Description of problem:**
Network manager sees the NTP server served by DHCP:
```
# nmcli con show "HassOS default" | grep ntp
DHCP4.OPTION[13]:                       ntp_servers = <REDACTED>
DHCP4.OPTION[21]:                       requested_ntp_servers = 1
```
timesyncd settings haven't changed:
```
# cat /etc/systemd/timesyncd.conf
[Time]
NTP=time1.google.com time2.google.com time3.google.com
FallbackNTP=0.pool.ntp.org 1.pool.ntp.org 2.pool.ntp.org 3.pool.ntp.org
```
[The docs say](https://www.freedesktop.org/software/systemd/man/timesyncd.conf.html) that these options should not override network-provided options but will be used in combination with them, so this should work. But timedatectl is using the configured NTP servers only, not the DHCP-provided one:
```
# timedatectl show-timesync --all
LinkNTPServers=
SystemNTPServers=time1.google.com time2.google.com time3.google.com
FallbackNTPServers=0.pool.ntp.org 1.pool.ntp.org 2.pool.ntp.org 3.pool.ntp.org
ServerName=time2.google.com
ServerAddress=216.239.35.4
RootDistanceMaxUSec=5s
PollIntervalMinUSec=32s
PollIntervalMaxUSec=34min 8s
PollIntervalUSec=34min 8s
NTPMessage={ Leap=0, Version=4, Mode=4, Stratum=1, Precision=-20, RootDelay=0, RootDispersion=228us, Reference=GOOG, OriginateTimestamp=Tue 2020-05-26 15:19:06 UTC, ReceiveTimestamp=Tue 2020-05-26 15:19:06 UTC, TransmitTimestamp=Tue 2020-05-26 15:19:06 UTC, DestinationTimestamp=Tue 2020-05-26 15:19:06 UTC, Ignored=no PacketCount=354, Jitter=3.505ms }
Frequency=-1730678
# systemctl -n 200 status systemd-timesyncd.service
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/usr/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/systemd-timesyncd.service.d
           └─hassos.conf, ro.conf
   Active: active (running) since Mon 2020-05-18 17:18:42 UTC; 1 weeks 0 days ago
     Docs: man:systemd-timesyncd.service(8)
 Main PID: 2881 (systemd-timesyn)
   Status: "Synchronized to time server 216.239.35.4:123 (time2.google.com)."
    Tasks: 2 (limit: 4551)
   Memory: 1.1M
   CGroup: /system.slice/systemd-timesyncd.service
           └─2881 /usr/lib/systemd/systemd-timesyncd
```
It appears that it normally receives the NTP server from `systemd-networkd`, but since Network Manager is being used there maybe needs to be some manual hook? [This similar Ubuntu issue](https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1685484) hints that there should be a hook `/etc/dhcp/dhclient-exit-hooks.d/timesyncd` that creates a file `/run/systemd/timesyncd.conf.d/01-dhclient.conf`.
I tried manually creating this file and restarting the timesync service:
```
# cat /run/systemd/timesyncd.conf.d/01-dhclient.conf
[TIME]
NTP=<REDACTED>
```
And I see that it now includes it:
```
# timedatectl show-timesync --all
LinkNTPServers=
SystemNTPServers=time1.google.com time2.google.com time3.google.com <REDACTED>
FallbackNTPServers=0.pool.ntp.org 1.pool.ntp.org 2.pool.ntp.org 3.pool.ntp.org
ServerName=time1.google.com
ServerAddress=216.239.35.0
...
```
However, it does not prefer it:
```
# systemctl -n 200 status systemd-timesyncd.service
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/usr/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/systemd-timesyncd.service.d
           └─hassos.conf, ro.conf
   Active: active (running) since Tue 2020-05-26 16:04:20 UTC; 15s ago
     Docs: man:systemd-timesyncd.service(8)
 Main PID: 19486 (systemd-timesyn)
   Status: "Synchronized to time server 216.239.35.0:123 (time1.google.com)."
    Tasks: 2 (limit: 4551)
   Memory: 868.0K
   CGroup: /system.slice/systemd-timesyncd.service
           └─19486 /usr/lib/systemd/systemd-timesyncd
```
Looking at several other OSes, it seems that the most common configuration is to only specify the fallback servers so that the local server is selected first like so (I used Cloudflare to test like in the newer pulls):
```
# cat /etc/systemd/timesyncd.conf
[Time]
#NTP=
FallbackNTP=time.cloudflare.com
```
This causes the local NTP to be preferred:
```
# timedatectl show-timesync --all
LinkNTPServers=
SystemNTPServers=<REDACTED>
FallbackNTPServers=time.cloudflare.com
ServerName=<REDACTED>
ServerAddress=<REDACTED>
...
# systemctl -n 200 status systemd-timesyncd.service
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/usr/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/systemd-timesyncd.service.d
           └─hassos.conf, ro.conf
   Active: active (running) since Tue 2020-05-26 16:13:07 UTC; 1s ago
     Docs: man:systemd-timesyncd.service(8)
 Main PID: 19710 (systemd-timesyn)
   Status: "Synchronized to time server <REDACTED>:123 (<REDACTED>)."
    Tasks: 2 (limit: 4551)
   Memory: 804.0K
   CGroup: /system.slice/systemd-timesyncd.service
           └─19710 /usr/lib/systemd/systemd-timesyncd
```
It seems that `dhclient-script` is available so creating the hook may work. I could not test this live since the filesystem does not allow me to create new directories in `/etc`.

It appears that this file is provided by `systemd`, so maybe there is a compilation flag or something to include it. 
<details>
  <summary>This is how it appears in a Debian install I have (click this text to show)</summary>
  
  ```
  $ cat /etc/dhcp/dhclient-exit-hooks.d/timesyncd
  TIMESYNCD_CONF=/run/systemd/timesyncd.conf.d/01-dhclient.conf
  
  timesyncd_servers_setup_remove() {
          if [ -e $TIMESYNCD_CONF ]; then
              rm -f $TIMESYNCD_CONF
              systemctl try-restart systemd-timesyncd.service || true
          fi
  }
  
  timesyncd_servers_setup_add() {
          if [ ! -d /run/systemd/system ]; then
                  return
          fi
  
          if [ -e $TIMESYNCD_CONF ] && [ "$new_ntp_servers" = "$old_ntp_servers" ]; then
                  return
          fi
  
          if [ -z "$new_ntp_servers" ]; then
                  timesyncd_servers_setup_remove
                  return
          fi
  
          mkdir -p $(dirname $TIMESYNCD_CONF)
          cat <<EOF > ${TIMESYNCD_CONF}.new
  # NTP server entries received from DHCP server
  [Time]
  NTP=$new_ntp_servers
  EOF
          mv ${TIMESYNCD_CONF}.new ${TIMESYNCD_CONF}
          systemctl try-restart systemd-timesyncd.service || true
  }

  
  case $reason in
          BOUND|RENEW|REBIND|REBOOT)
                  timesyncd_servers_setup_add
                  ;;
          EXPIRE|FAIL|RELEASE|STOP)
                  timesyncd_servers_setup_remove
                  ;;
  esac
  ```
  
</details>

I can work on a pull and testing the hook if this is the preferred solution.

pvizeli, 2020-05-28:
Feel free to adapt this to HassOS.

sdwilsh, 2020-07-13:
This would be useful as a way to workaround the lack of details in the documentation on how to change the NTP settings as well (as noted by https://github.com/home-assistant/supervisor/issues/957#issuecomment-528948200)

DaAwesomeP, 2020-07-13:
I am working on another Home Assistant pull request and will work on this one shortly after that is completed.

DaAwesomeP, 2020-09-13:
Sorry for needless message but just commenting to un-stale since I am working on a pull.

LearnLinuxTV, 2021-02-22:
Why is this marked wontfix? Still seems to be a problem. On my network I mandate a specific NTP server. Home Assistant is having errors because the time is wrong, and it cannot correct it. It's not desirable to create a firewall rule for Home Assistant to reach an NTP server that goes against the design of the network it's running on, when all other devices on the network happily use the provided NTP server.

DaAwesomeP, 2021-02-22:
@LearnLinuxTV it seems the stale bot added it, but @agners removed the wontfix tag. I was working on https://github.com/home-assistant/operating-system/pull/827 but unfortunately I no longer have time to work on it. Recompiling the kernel is painfully slow on my machine and I do not have the ability to test building and trying this overnight or throughout the day at the moment.

I am happy to provide feedback on my findings and assist with further pulls.

DaAwesomeP, 2021-02-22:
A good starting point would be to fork my original pull: https://github.com/DaAwesomeP/hassos/tree/DaAwesomeP-DHCP-based-NTP

See the comments in #689 to see what I was still testing.

redgryphon, 2022-03-16:
I am pretty confident in the next few days, if not hours, I will be able to prepare a PR for this bug. It works in my lab and I need to clear the code a little. Stay tuned.

PR #1798 (2022-03-20): Add support for NTP configuration via DHCP (fixes #689)
I managed to create a working solution that fixes #689. It works in my lab but I think it is pretty straightforward to be understood and to be reproduced as well.
commit f62fee2f (2022-03-20): Add support for NTP configuration via DHCP (fixes #689) (#1798)

* Add support for NTP configuration via DHCP.

* Default fallback NTP pool is the Cloudflare's one
closed 2022-03-20 by agners