home-assistant/operating-system #691: Update from 3.13 to 4.6 and 4.7 failed

Adorem, 2020-05-27:
**HassOS release with the issue:**

_ | _
-- | --
arch | armv7l
chassis | embedded
dev | false
docker | true
hassio | true
host_os | **HassOS 3.13**
installation_type | Home Assistant
os_name | Linux
os_version | 4.19.114-v7
python_version | 3.7.7
supervisor | 224
timezone | Europe/Paris
version | 0.110.3
virtualenv | false

**Supervisor logs:**

```
20-05-27 10:11:01 INFO (MainThread) [supervisor.hwmon] Stop Supervisor hardware monitor
20-05-27 10:11:01 INFO (MainThread) [supervisor.misc.forwarder] Stop DNS forwarding
20-05-27 10:11:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 257:None is not str
20-05-27 10:11:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-05-27 10:11:01 INFO (MainThread) [supervisor.api] Stop API on 172.30.32.2
20-05-27 10:11:01 INFO (MainThread) [supervisor.core] Supervisor is down
20-05-27 10:11:01 INFO (MainThread) [__main__] Close Supervisor
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
20-05-27 10:12:07 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/mopidy_alpine/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:07 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/snapcastserver/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:07 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/nextcloud/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:07 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/kanboard/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:07 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/core repository
20-05-27 10:12:07 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/53caca22 repository
20-05-27 10:12:07 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/a0d7b954 repository
20-05-27 10:12:07 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/15ef4d2f repository
20-05-27 10:12:07 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/mopidy_alpine/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:08 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/snapcastserver/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:08 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/nextcloud/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:08 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/53caca22/kanboard/config.json: required key not provided @ data['arch']. Got None
20-05-27 10:12:08 INFO (MainThread) [supervisor.store] Load add-ons from store: 84 all - 84 new - 0 remove
20-05-27 10:12:08 INFO (MainThread) [supervisor.addons] Found 10 installed add-ons
20-05-27 10:12:08 INFO (SyncWorker_9) [supervisor.docker.interface] Attach to hassioaddons/node-red-armv7 with version 6.2.0
20-05-27 10:12:08 INFO (SyncWorker_5) [supervisor.docker.interface] Attach to hassioaddons/portainer-armv7 with version 1.1.0
20-05-27 10:12:08 INFO (SyncWorker_7) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-ssh with version 8.5.4
20-05-27 10:12:08 INFO (SyncWorker_3) [supervisor.docker.interface] Attach to esphome/esphome-hassio-armv7 with version 1.14.3
20-05-27 10:12:08 INFO (SyncWorker_2) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-mosquitto with version 5.1
20-05-27 10:12:08 INFO (SyncWorker_1) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-configurator with version 5.0.0
20-05-27 10:12:08 INFO (SyncWorker_14) [supervisor.docker.interface] Attach to bestlibre/armv7-nginx-proxy with version 0.1.14
20-05-27 10:12:08 INFO (SyncWorker_8) [supervisor.docker.interface] Attach to bestlibre/armv7-certbot with version 0.1.10
20-05-27 10:12:08 INFO (SyncWorker_15) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-check_config with version 3.3.0
20-05-27 10:12:08 INFO (SyncWorker_9) [supervisor.docker.interface] Attach to hassioaddons/sqlite-web-armv7 with version 2.2.1
20-05-27 10:12:08 INFO (MainThread) [supervisor.snapshots] Found 6 snapshot files
20-05-27 10:12:08 INFO (MainThread) [supervisor.discovery] Load 1 messages
20-05-27 10:12:08 INFO (MainThread) [supervisor.ingress] Load 0 ingress session
20-05-27 10:12:08 INFO (MainThread) [supervisor.secrets] Load Home Assistant secrets: 2
20-05-27 10:12:08 INFO (MainThread) [__main__] Run Supervisor
20-05-27 10:12:08 INFO (MainThread) [supervisor.api] Start API on 172.30.32.2
20-05-27 10:12:08 INFO (MainThread) [supervisor.utils.gdbus] Call de.pengutronix.rauc.Installer.Mark on /
20-05-27 10:12:08 INFO (MainThread) [supervisor.hassos] Rauc: B - marked slot kernel.1 as good
20-05-27 10:12:08 INFO (MainThread) [supervisor.addons] Phase 'initialize' start 0 add-ons
20-05-27 10:12:08 INFO (MainThread) [supervisor.addons] Phase 'system' start 1 add-ons
20-05-27 10:12:08 INFO (SyncWorker_7) [supervisor.docker.interface] Clean addon_core_mosquitto application
20-05-27 10:12:11 INFO (SyncWorker_7) [supervisor.docker.addon] Start Docker add-on homeassistant/armv7-addon-mosquitto with version 5.1
[... 617 lines cut ...]
[ 1082.458911] audit: type=1325 audit(1590575372.012:184): table=nat family=2 entries=5
[ 1082.627238] eth0: renamed from vethe75f195
[ 1082.687757] IPv6: ADDRCONF(NETDEV_CHANGE): vethf7dccd5: link becomes ready
[ 1082.703573] hassio: port 9(vethf7dccd5) entered blocking state
[ 1082.713623] hassio: port 9(vethf7dccd5) entered forwarding state
[ 1086.534792] hassio: port 9(vethf7dccd5) entered disabled state
[ 1086.540050] vethe75f195: renamed from eth0
[ 1086.710952] hassio: port 9(vethf7dccd5) entered disabled state
[ 1086.716835] device vethf7dccd5 left promiscuous mode
[ 1086.721828] hassio: port 9(vethf7dccd5) entered disabled state
[ 1086.721868] kauditd_printk_skb: 23 callbacks suppressed
[ 1086.721872] audit: type=1700 audit(1590575376.262:192): dev=vethf7dccd5 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[ 1086.785129] audit: type=1300 audit(1590575376.262:192): arch=40000028 syscall=290 per=800000 success=yes exit=32 a0=c a1=464afc0 a2=20 a3=0 items=0 ppid=1 pid=341 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj==unconfined key=(null)
[ 1086.803419] audit: type=1327 audit(1590575376.262:192): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D73746F726167652D6472697665723D6F7665726C617932002D2D6C6F672D6472697665723D6A6F75726E616C64002D2D646174612D726F6F74002F6D6E742F646174612F646F636B6572
```

**Description of problem:**
Hello,

I'm using RPI3 and try to update from 3.13 to 4.7 (Stable) without success. I also test previously with 4.6 (RC) with the same results.
After the update I have no response from the device. And after 2 or 3 reboots I'm back to 3.13.

Is there some other informations I need me to provide?

Thanks.

pvizeli, 2020-05-27:
I can not reproduce it. Could it be that it changes the IP?

Adorem, 2020-05-29:
Thanks for the try.

When I reboot the system the first time he still reply to ping but nothing on the http port 8123.
During the install I can see this on the screen:
[image]

And after the reboot:
[image]

abba232, 2020-05-30:
I also encountered the same problem, because I am in China,
After downloading the latest version hassos_rpi3-4.8.img.gz
Create a folder named network in the new U disk
Then created a new my-network file

```
1. [connection]
2. id=my-network
3. uuid=d55162b4-6152-4310-9312-8f4c54d86afa
4. type=802-3-ethernet
5. 
6. [ipv4]
7. method=auto
8. 
9. [ipv6]
10. addr-gen-mode=stable-privacy
11. method=auto

Create a timesyncd.cofn file in the root directory

[Time]
NTP=time1.aliyun.com time2.aliyun.com time3.aliyun.com 
FallbackNTP=ntp1.aliyun.com ntp2.aliyun.com ntp3.aliyun.com 

Upgrade after installation

Operating System 3.13
You are currently running version 4.8

502 gateway error

20-05-30 02:52:56 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information

Adorem, 2020-05-30:
Thanks @abba232 but I do not see how I do this?
For the installation I start it from the menu Supervisor > uptade. And when the installation is in this phase I can't ssh through port 22222 and in the console I get the error "No such container: hassio_cli"

abba232, 2020-05-30:
@Adorem Thanks for your advice

jzee, 2020-06-02:
Similar problem going from 3.13 to 4.8 on an RPI4, hassio. 
I click "update" on on the Supervisor > Dashboard, observing one log entry in the system log: 

> 20-06-02 08:54:13 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.8/hassos_rpi4-4.8.raucb

then nothing, no error message. A reboot doesn't change anything. I'd be happy to provide more info, please help with guidance as to which log can be of help.

markgaze, 2020-06-02:
Had this happen to me last night too, almost identical set up to the original issue, Raspberry Pi 3 running 3.13 but upgrading to 4.8. The Pi rebooted, came back up to respond to pings but nothing ran in HA and I was getting `Error: No such container: hassio_cli`.

After multiple reboots, it's back up and running with 3.13

nalajcie, 2020-06-02:
Hardware: Raspberry 2+ B.
Same problem with updating from 3.13 to 4.8

After trying to login on the console or via SSH the same error message:
```
Error: No such container: hassio_cli
```

I've pulled the SD card from the device and it seems that the hassio-supervisor container can't be run:
```
cze 02 10:48:56 hassio systemd[1]: Starting HassOS supervisor...
cze 02 10:48:56 hassio docker[677]: hassos_supervisor
cze 02 10:48:56 hassio systemd[1]: Started HassOS supervisor.
cze 02 10:48:56 hassio systemd[1]: Started Getty on tty1.
cze 02 10:48:57 hassio hassos-supervisor[689]: Error: No such object: homeassistant/armhf-hassio-supervisor
cze 02 10:48:57 hassio systemd[1]: hassos-supervisor.service: Main process exited, code=exited, status=1/FAILURE
cze 02 10:48:57 hassio systemd[1]: hassos-supervisor.service: Failed with result 'exit-code'.
```

From what I'm seeing the architecture of hassio-supervisor has been changed (inspecting `/usr/sbin/hassos-supervisor` startup script) - previously it was taken from `hassio.conf`, now directly from `/etc/os-release`.

Previously installed image: `armv7-hassio-supervisor`
Currently requested image: `armhf-hassio-supervisor`

Unfortunately I don't know how to migrate form one image to the other, I hope the developers would solve the problem or fall back to `armv7` for rPi 2 platform.

For now going back to 3.13.

PeterBainbridge, 2020-06-02:
> Had this happen to me last night too, almost identical set up to the original issue, Raspberry Pi 3 running 3.13 but upgrading to 4.8. The Pi rebooted, came back up to respond to pings but nothing [...]

I had the same. A few reboots brought me back.

CuriouslyCory, 2020-06-02:
Having the same issue. Updating from 3.12 to 4.x results in Error: No such container: hassio_cli when trying to log in through cli on my rpi 3b. 3 reboots brings it back to 3.x.

thomasschurr, 2020-06-02:
Same issue as described above when trying to upgrade my "production" instance from 3.13 to 4.8 (0.110.4, Raspberry Pi 3 Model B+, eth0). The upgrade of my much simpler test instance (0.110.4, Raspberry Pi 3 Model B+, wifi) ran without any issues.

pvizeli, 2020-06-02:
Fix by https://github.com/home-assistant/operating-system/commit/917f0226147ed0f0f081e7407bff00963dedd99e

Adorem, 2020-06-03:
Cool and thanks @pvizeli. Thanks also to @nalajcie that probably helps to understand the root cause.
Waiting for the next release including this fix 👍

pvizeli, 2020-06-03:
4.9 is on build

closed 2020-06-03 by pvizeli

fbradyirl, 2020-06-03:
[image]

For those of us stuck (RP 3b), how do we get it to update or rollback?

markgaze, 2020-06-03:
> [image]
> 
> For those of us stuck (RP 3b), how do we get it to update or rollback?

You may have already fixed this by now but if you reboot three times, it rolls back to the old version.

nalajcie, 2020-06-04:
> 4.9 is on build

When 4.9 will be available for update via supervisor? I would like to test it against rpi2 target.

For rPi2:
 - stable version points to 3.13 (https://version.home-assistant.io/stable.json)
 - beta version points to 4.8 - **which does not work** (https://version.home-assistant.io/beta.json)