home-assistant/operating-system #695: Version 4.7 asks for downgrade to 3.13

alexeyzel, 2020-05-27:
**HassOS release with the issue:** HassOS 4.7

**Supervisor logs:**

```
20-05-26 21:40:56 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-05-26 23:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/hassio repository
20-05-26 23:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-05-26 23:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-05-26 23:27:35 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-26 23:27:37 INFO (MainThread) [supervisor.store] Load add-ons from store: 68 all - 0 new - 0 remove
20-05-26 23:47:35 INFO (MainThread) [supervisor.host.info] Update local host information
20-05-26 23:47:35 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-05-26 23:47:35 INFO (MainThread) [supervisor.host.services] Update service information
20-05-26 23:47:35 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-05-26 23:47:36 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-05-26 23:47:36 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-05-26 23:47:36 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-05-27 01:27:35 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-27 01:54:15 INFO (MainThread) [supervisor.host.info] Update local host information
20-05-27 01:54:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-05-27 01:54:15 INFO (MainThread) [supervisor.host.services] Update service information
20-05-27 01:54:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-05-27 01:54:16 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-05-27 01:54:16 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-05-27 01:54:16 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-05-27 02:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-05-27 02:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-05-27 02:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/hassio repository
20-05-27 02:27:37 INFO (MainThread) [supervisor.store] Load add-ons from store: 68 all - 0 new - 0 remove
20-05-27 03:27:35 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-27 04:00:55 INFO (MainThread) [supervisor.host.info] Update local host information
20-05-27 04:00:55 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-05-27 04:00:55 INFO (MainThread) [supervisor.host.services] Update service information
20-05-27 04:00:55 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-05-27 04:00:56 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-05-27 04:00:56 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-05-27 04:00:56 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-05-27 05:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/hassio repository
20-05-27 05:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-05-27 05:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-05-27 05:27:35 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-27 05:27:37 INFO (MainThread) [supervisor.store] Load add-ons from store: 68 all - 0 new - 0 remove
20-05-27 06:07:35 INFO (MainThread) [supervisor.host.info] Update local host information
20-05-27 06:07:35 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-05-27 06:07:35 INFO (MainThread) [supervisor.host.services] Update service information
20-05-27 06:07:35 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-05-27 06:07:36 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-05-27 06:07:36 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-05-27 06:07:36 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-05-27 07:27:35 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-27 08:14:15 INFO (MainThread) [supervisor.host.info] Update local host information
20-05-27 08:14:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-05-27 08:14:15 INFO (MainThread) [supervisor.host.services] Update service information
20-05-27 08:14:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-05-27 08:14:16 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-05-27 08:14:16 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-05-27 08:14:16 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-05-27 08:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-05-27 08:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/hassio repository
20-05-27 08:27:35 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-05-27 08:27:37 INFO (MainThread) [supervisor.store] Load add-ons from store: 68 all - 0 new - 0 remove
20-05-27 09:27:35 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-27 10:20:55 INFO (MainThread) [supervisor.host.info] Update local host information
20-05-27 10:20:55 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
20-05-27 10:20:55 INFO (MainThread) [supervisor.host.services] Update service information
20-05-27 10:20:55 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-05-27 10:20:56 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-05-27 10:20:56 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-05-27 10:20:56 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
[... 20 lines cut ...]
20-05-27 14:34:15 INFO (MainThread) [supervisor.host.services] Update service information
20-05-27 14:34:15 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
20-05-27 14:34:16 INFO (MainThread) [supervisor.host.network] Update local network DNS information
20-05-27 14:34:16 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
20-05-27 14:34:16 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-05-27 15:27:35 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-05-27 16:40:55 INFO (MainThread) [supervisor.host.info] Update local host information
[... cut ...]
client_connection: Connection<ConnectionKey(host='172.30.32.1', port=63951, is_ssl=False, ssl=None, proxy=None, proxy_auth=None, proxy_headers_hash=None)>
```

**Journal logs:**

**Kernel logs:**

**Description of problem:**
I installed latest version of Home Assistant based on HassOS 4.7 in Proxmox using this script https://github.com/whiskerz007/proxmox_hassos_install
Everything is working well, but on "Supervisor" tab the system is showing that I need to upgrade to 3.13 (i have latest version 4.7):
[image]

allanpersson, 2020-05-27:
Os 4.7 is currently available in beta only, so when you are not part of the beta Channel it will offer downgrade to 3.13

allanpersson, 2020-05-27:
You can click join beta in Supervisor menu and no downgrade is offered 🙏

frenck, 2020-05-27:
The above answer is correct. It will offer you a downgrade to a stable release.

closed 2020-05-27 by frenck

DaAwesomeP, 2020-06-03:
May I ask why beta releases are offered if you are not in the beta channel? This seems confusing to most users since it just looks like a normal software update.

balloob, 2020-06-03:
You should ask that question to the author of the script that OP used: https://github.com/whiskerz007/proxmox_hassos_install

DaAwesomeP, 2020-06-04:
I guess I should open a new issue—I am being offered this beta version but I installed HA directly on hardware, the recommended/official way.