home-assistant/operating-system #697: intel NUC i219-v Not supported in Hassio 4.8

dlo747, 2020-05-29:
HassOS release with the issue:
0.1010.3

Operating environment (HassOS/Generic):
HassOS, Intel NUC10FNH with NVME SSD

Supervisor logs:
N/A

Journal logs:
N/A

Kernel logs:
N/A

Description of problem:
Tried running Hassio 4.5 and 4.8 NUC install image and the network adapter isn’t being recognized.

Ubuntu VirtualBox install it works fine

dlo747, 2020-05-30:
Ubuntu uses the e1000 driver and it works fine with a docker

dick902, 2020-05-30:
Was that a way to update the e1000 driver?
I had the same issue with my NUC10.

dlo747, 2020-05-30:
> Was that a way to update the e1000 driver?
> I had the same issue with my NUC10.

Thats what Ubuntu 20.04 uses and it works fine (was saying, I’m not sure what they use in Hassio), the Hassio 4.8 NUC iso doesn’t recognize the NIC or get a ip for some reason

dlo747, 2020-05-30:
Seems #618 has similar issue as well

dlo747, 2020-05-30:
https://downloadcenter.intel.com/download/15817/Intel-Network-Adapter-Driver-for-PCI-E-Gigabit-Network-Connections-under-Linux-?product=71307

e1000e driver is the one that can run the I2xx intel Ethernet Controllers. And the latest e1000e driver (as of this writing) is capable of running the I219 chip.

dick902, 2020-05-30:
With Ubuntu 19.xx, I got the same issue but I was able to manual update the e1000 driver that make it work!
May be that could be a way to do the same with HASSOS!
Someone had a hints?

pvizeli, 2020-06-03:
You need to create a buildroot package with that driver, so buildroot can pick the driver, build and distribute it to the image

dlo747, 2020-06-03:
Any tutorials or guide how to do that? Thanks you

dlo747, 2020-06-07:
Still exists

pvizeli, 2020-06-08:
https://buildroot.org/downloads/manual/ the manual above depend on how the software was written.

rileyskyler, 2020-06-09:
Whoever is having this issue with a NUC, something worth a try, using an USB ethernet NIC.

I was having ethernet issues w/ my NUC 10 running baremetal HA.

I haven't yet tested this with baremetal HA but found it to be successful with my NUC 10 using Proxmox when when Ubuntu Desktop had no issues connecting to the onboard NIC.

dlo747, 2020-06-09:
> Whoever is having this issue with a NUC, something worth a try, using an USB ethernet NIC.
>
> I was having ethernet issues w/ my NUC 10 running baremetal HA.
>
> I haven't yet tested this with baremetal [...]

Good idea. I just don’t want to buy a USB NIC just to use till the regular one is supported 😕

pvizeli, 2020-06-12:
@jvanderneutstulen do you want to look into this as maintainer of intel-nuc port?

dlo747, 2020-06-18:
Still not working :(

PR #765 (2020-07-06): add out-of-tree intel e1000e ethernet module
Fixes #697 (I219-V for NUC 10)
closed 2020-07-06 by pvizeli