home-assistant/operating-system #703: After upgrade to 4.8 RPI4 don't boot with UART2 overlay

dianlight, 2020-06-01:
**HassOS release with the issue:**
Hassios 4.3

**Description of problem:**

I had several problems after upgrading to version 4.8 from version 3.13.
After several attempts I managed to isolate the problem. If in config.txt I ask to enable uart2 then the RPI does not even boot (black screen on the video in HDMI and activity led off).
Simply by removing the uart2 everything works properly.

Questo è il mio config.txt 

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# HassOS - don't change it!
disable_splash=1
kernel=u-boot.bin

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

#uart2 on PIN 27/28
dtoverlay=uart2
```

DanielNL, 2020-06-01:
I'm having trouble as well, tried updating to 4.8 from 3.13 and i can't acces my RPI 3B+ anymore. I can only ping the RPI, no file acces via smb nor ssh via port 22222 works.. guess i have to go with a new install..

dianlight, 2020-06-01:
You don't need a new install.

It happened to me too. You just have to reconfigure the network and authorized_keys it. Create the USB CONFIG stick and boot again.

It takes 3/4 boot before everything is up and running again.

L.

martymarty004, 2020-06-01:
@dianlight can you explain how to do it?

IceOnly, 2020-06-01:
> You don't need a new install.
> 
> It happened to me too. You just have to reconfigure the network and authorized_keys it. Create the USB CONFIG stick and boot again.
> 
> It takes 3/4 boot before everything is up and running again.
> 
> L.

Are you sure there was no rollback to 3.x?

DanielNL, 2020-06-01:
@dianlight Thank you for the quick response, up and running again. Albeit on version 3.13 but I'll leave it at that for now.

@martymarty004 just format a usb stick to FAT32 and put the config.txt file on it with the content of dainlight's first post, without the additional lines. Then reboot your rpi.

dianlight, 2020-06-01:
> @dianlight can you explain how to do it?

Follow the instruction to configure the network and the ssh keys for hassio as is a new install. Create a FAT32 USB KEY with name "CONFIG" in the root put authorization_key file with your pub key and a directory network with a file my-network with the config.

dianlight, 2020-06-01:
@IceOnly 

> Are you sure there was no rollback to 3.x?

No I'm on 4.8 now. But UART2 don't work. Do you know how rollback?

martymarty004, 2020-06-01:
> Follow the instruction to configure the network and the ssh keys for hassio as is a new install. Create a FAT32 USB KEY with name "CONFIG" in the root put authorization_key file with your pub key [...]

Thank you!

IceOnly, 2020-06-01:
> @IceOnly
> 
> > Are you sure there was no rollback to 3.x?
> 
> No I'm on 4.8 now. But UART2 don't work. Do you know how rollback?

No. I have the same issues. After rebooting 3/4 Times it runs on 3.18 again...

dianlight, 2020-06-01:
Meantime Rollback done! :)

For everyone:

- Use the CONFIG USB KEY.
- Download the old version .raucb from [release page](https://github.com/home-assistant/hassos/releases/)
- Rename the .raucb to hassos-4.9.raucb (the version need to be upper to perform the process...) and put in the root of USB KEY.
- Connect the USB key and reboot the host
- ....wait....
- after the first boot remove the key
- ..wait.. ( a lot! )

I saved the 4.8 image to perform test if a solution for UART2 is found.

L.

inverse, 2020-06-01:
Just writing to say - updated from 3.13 -> 4.8 and was seemingly dead. Did what others suggested rebooting 3/4 times and then back to life on 3.13.

andker87, 2020-06-01:
same on rpi3b+ with 3.13. after upgrading the rpi seemed dead. Done what suggested above and rpi is finally resurrected. hope for a fix soon.

skitfull, 2020-06-01:
> Just writing to say - updated from 3.13 -> 4.8 and was seemingly dead. Did what others suggested rebooting 3/4 times and then back to life on 3.13.

What exactly did you do? I´ve tried three different options from various forums and maybe I am a little unpatient here but nothing works.

This is the last thing ivé tried as described above:

1. Formatted a USB-drive and named it "CONFIG"
2. Downloaded "hassos_rpi4-3.13.raucb" and named it to "hassos_rpi4-5.0.raucb"
3. Killed the raspberry, put memorystick in and waited for about 2 mins
4. Killed raspberry again, and rebooted it again with stick in and waited for 5 min
5. Killed raspberry again, and rebooted it and waited for 5 min

Nothing happens - its dead?!

Should i remove the stick and reboot? Should i wait for like 20 mins or so?

Thanks in advance for patient and help.

inverse, 2020-06-01:
@skitfull - I'm on a rpi3b+ with 3.13

I didn't do anything particular

1. hit the upgrade to 4.8 in hassio UI
2. it rebooted, never came back
3. rebooted few times waiting 3/5 mins between, nothing, couldn't shell in either
4. Plugged in via HDMI and saw all booting fine and could SSH in and it's back on 3.13

skitfull, 2020-06-01:
> @skitfull - I'm on a rpi3b+ with 3.13
>
> I didn't do anything particular
>
> 1. hit the upgrade to 4.8 in hassio UI
> 2. it rebooted, never came back
> 3. rebooted few times waiting 3/5 mins between, [...]

Ok, my raspberry seems dead - no response from USB-port (I got a fan attached for cooling and it dont start anymore).

I´ve restarted like 20 times now with different configurations on a USB-stick called "CONFIG" and even a few times without the USB-stick.

Perhaps all the different configs I´ve put to the machine through USB-stick broke something (becouse I have no idea what it acutally does).

Maybe its back to square 0.

Tomorrow when the worst panic over this is over from my side, maybe I can check the configs on the SD-card and see if I can find anything.

Thanks for your response @inverse

inverse, 2020-06-01:
@skitfull ouch :( that's no fun - so this all happened after the upgrade? 

I hope you get back to a working solution soon!

skitfull, 2020-06-01:
@inverse I´ve just bought a bag of new stuff for my "smart home" - a new ID Lock 150 (among other things) that took me forever to mount on my front door becouse I have kids that needs my attention. And when I was done with installation and was going to configure it to my z-wave stick I saw this update and thought "Why not".

Now my home is not so smart anymore, and thats probably becouse I got panic and tried everything with that USB-stick.

As I sed - I´ll sleep on it and pull that SD-card tomorrow and see what I can do to save all data.

JakobTewes, 2020-06-01:
oh dammn....two installations dead....one on site and one remote.....
ssh on port 22222 comes up but my ssh keys don't seem to be trusted :-(
will try the "powercycle some rounds" trick now :-)

dianlight, 2020-06-01:
@skitfull 

> 2\. Downloaded "hassos_rpi4-3.13.raucb" and named it to "hassos_rpi4-5.0.raucb"

The name must hassos-5.0.raucb to trigger the upgrade/rollback process.

L.

slim0287, 2020-06-02:
For what it's worth, I am running hass.io on a Raspberry Pi 2.  After upgrade to 4.8, I am no longer able to access the Pi.  I appreciate the instructions for a rollback; they worked well.

Bettinaglio, 2020-06-02:
Similar problem here: Error: 502: Bad Gateway when trying to update 3.13. to 4.8 on RPI3+. Now trying to ReBoot.

SVH-Powel, 2020-06-02:
I suppose if you are booting from a memory card the config.txt file could be edited on another computer with a text editor (notepad)

skitfull, 2020-06-02:
> @skitfull
> 
> > 2. Downloaded "hassos_rpi4-3.13.raucb" and named it to "hassos_rpi4-5.0.raucb"
> 
> The name must hassos-5.0.raucb to trigger the upgrade/rollback process.
> 
> L.

Thanks, I'll try this after work when i come home.

ajgnet, 2020-06-02:
Similarly I upgraded and my pi failed to boot. I was able to resolve this by commenting out:

#enable_uart=1

which I had enabled for my z-wave controller. Awaiting a fix before re-enabling...

eldeldani, 2020-06-02:
Same here, restarting several times brought my system to live in 3.13, I panicked a bit. Bought another PI4 to test full restore scenarios and avoid running into not having a smart home due to an failed upgrade.

talondnb, 2020-06-02:
Same issue here, used @dianlight approach here: https://github.com/home-assistant/operating-system/issues/703#issuecomment-636996734

Make sure USB key is named CONFIG. Thanks @dianlight, saved me a full reimage/restore.

hhromic, 2020-06-02:
Wanted to report that my RPI3B+ with a deCONZ Raspbee GPIO board *does boot correctly* but the Raspbee stops working. Similar to @ajgnet I also use `enable_uart=1` in my `config.txt`:
```
# Disable BT and enable UART
dtoverlay=pi3-disable-bt
enable_uart=1
```

So I suspect that there is a problem with UART in the new Kernel shipped with HassOS 4.8.

Yardco, 2020-06-02:
Same issue for me (RPI3B+ booting correctly) but with z-wave (Razberry) not working. In the process of rolling back to get my z-wave devices back.

Daxziz, 2020-06-02:
> Wanted to report that my RPI3B+ with a deCONZ Raspbee GPIO board does boot correctly but the Raspbee stops working. Similar to @ajgnet I also use enable_uart=1 in my config.txt:

Can confirm that Raspbee module isn't working on 4.8.

idoff, 2020-06-02:
Have the same problem but not sure how to downgrade again, can someone give me a short guide or link to one?

pedrohbr, 2020-06-02:
> > Wanted to report that my RPI3B+ with a deCONZ Raspbee GPIO board does boot correctly but the Raspbee stops working. Similar to @ajgnet I also use enable_uart=1 in my config.txt:
> 
> Can confirm that Raspbee module isn't working on 4.8.

If you’re lucky enough to get your Pi to boot in HassOS 4.8 (it takes me 2 or 3 power cycles), changing the DeConz config file to “device: /dev/ttyS0“ should get everything in working order. It was the case for me with a RaspBee 2, and there were similar reports in the Home Assistant Community. So far, my system has been running stable... I am just avoiding rebooting it.

Daxziz, 2020-06-02:
> If you’re lucky enough to get your Pi to boot in HassOS 4.8 (it takes me 2 or 3 power cycles), changing the DeConz config file to “device: /dev/ttyS0“ should get everything in working order. It was [...]

I've seen this as stated as a workaround as well, but on my RPI 3B+, Raspbee 2 and HassOS 4.8 this device isn't available for my setup.

skitfull, 2020-06-02:
Nothing seemed to work for me because some systemfiles must have changed when trying different USB-stick-configs. I did a fresh install of Home Assistant 3.13 and managed to get and old backup up and running.

Now its all good again - and I´ve learned the hard way to take regular backups, specially before updating.

nilsmau, 2020-06-02:
same issue here on RPI3B+

nilsmau, 2020-06-02:
downgrade to 3.13 fixed the issue.

cogneato, 2020-06-02:
@nilsmau do you have the default config.txt?

balloob, 2020-06-02:
If your system is not booting and you are not using uart2 in your config.txt overlay, please create a new issue. Always clearly state the hardware and if is is a new installation or an upgrade, and if so upgraded from the previous version.

Caine72, 2020-06-02:
Even though i never disabled enable_uart=1 my system booted after the upgrade on my Raspberry pi 3B+. But 4.8 killed the connection to the Razberry2 running on /dev/ttyAMA0.
Reverting back to 3.13 re-enabled it.

closed 2020-06-02 by pvizeli

stunney, 2020-06-03:
Might I ask WHY the this breaking release hasn't been pulled offline yet? I *just* his this this afternoon.

adrianrdzv, 2020-06-04:
I have an Elelabs Zigbee shield with the custom config line:
dtoverlay=pi3-disable-bt

It also stopped working with 4.8. I manually installed 4.9 but still not working there either.
Now I'm trying to revert to 3.13

balloob, 2020-06-04:
That is not related to this issue. Please open a new issue

PR #706 (2020-06-02): Fix RPi4 uart 2-5
Fix: #703