home-assistant/operating-system #707: Razberry stops working in 4.8

fencer, 2020-06-03:
After upgrading my os to 4.8 Razberry has stopped working on Rpi3b+

magin1, 2020-06-03:
First i updates Home Assistant 0.110.4 and everything was fine, and after the Operating System 4.8 update HA did not come back up, restarted raspberry 3b+ couple of times but no luck. Is there a rollback or fix possible?

kostas2911, 2020-06-03:
same here. Razberry never came up after upgrade to 4.8

Villa-Buenavista, 2020-06-03:
Same here as also reported in forum (as have several other users):
https://community.home-assistant.io/t/hassos-4-8-update-breaks-zwave-on-rpi/201455?u=buenavista

Any help?

drsthlm, 2020-06-03:
rebooted (hard) three times on my pi - then it came back again - and not with 4.8

chriselera, 2020-06-03:
Same issue with me. Since I have recent snapshots, I decided to do a fresh HassOS install.  Versions 4.8 and 4.7 will not work (RPi3 will not boot up).  Fresh installing 3.13 works.

balloob, 2020-06-03:
We've released 4.9 with fixes. Give it a try: https://github.com/home-assistant/operating-system/releases/tag/4.9

closed 2020-06-03 by balloob

MichaelBitard, 2020-06-03:
@balloob How can we help you to make the 4.9 image available for the rpi3?

MichaelBitard, 2020-06-03:
Oups, sorry for bothering you, I can see the azure pipeline running

alexandre-lg, 2020-06-03:
Hello,

My Razberry stop to work on my RPi3B+ after I upgrade to HassOS 4.8. For your information, I see the following lines in the Z-wave log :

```
2020-06-03 23:34:28.304 Always, OpenZwave Version 1.4.3469 Starting Up
2020-06-03 23:34:40.611 Info, Setting Up Provided Network Key for Secure Communications
2020-06-03 23:34:40.612 Info, mgr,     Added driver for controller /dev/ttyAMA0
2020-06-03 23:34:40.612 Info,   Opening controller /dev/ttyAMA0
2020-06-03 23:34:40.612 Info, Trying to open serial port /dev/ttyAMA0 (attempt 1)
2020-06-03 23:34:40.613 Info, Serial port /dev/ttyAMA0 opened (attempt 1)
2020-06-03 23:34:40.613 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2020-06-03 23:34:40.614 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2020-06-03 23:34:40.614 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2020-06-03 23:34:40.614 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2020-06-03 23:34:40.614 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2020-06-03 23:34:40.614 Detail,
2020-06-03 23:34:40.614 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2020-06-03 23:34:40.614 Detail, contrlr, Notification: DriverFailed
```

How can I test version 4.9 without having to reconfigure my entire Z-wave network?

Thanks for your help, your work and your time!

chriselera, 2020-06-04:
Unfortunately, this is still not working. Downloaded the 4.9 RPi3 image, flashed to spare SD card, tried twice and it won't install. Flashed 3.13 image to the same spare SD card, and it begin setting up after a few minutes.

balloob, 2020-06-04:
This issue has been resolved. If you're still experiencing this issue on 4.9, please open a new issue and include logs.

lastessa, 2020-06-05:
reopen please, because still no working with razberry board https://github.com/home-assistant/operating-system/issues/716

balloob, 2020-06-05:
Not going to reopen, as #716 is the new one to track.