home-assistant/operating-system #710: update to 4.8 failed

ohadlevy, 2020-06-03:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Hi,

I've just upgrade to 4.8 and my RPI stopped working, it seems none of the containers were able to start.

I had ssh debugging (not the ssh addon) enabled, and when trying to ssh i get
```
ssh root@haas -p 22222 
sh: /usr/bin/xauth: not found
Error: No such container: hassio_cli
Shared connection to haas closed.
```

any guidance on how to move forward, or must i revert to a snapshot?

alexisml, 2020-06-03:
+1 my raspy won't boot after the update

balloob, 2020-06-03:
We've released 4.9 with fixes. Give it a try: https://github.com/home-assistant/operating-system/releases/tag/4.9

closed 2020-06-03 by balloob

ohadlevy, 2020-06-03:
@balloob  thanks, can you suggest the procedure to upgrade from this stage? is there a way to put the new image on the sd card via a linux machine?

ghitchen, 2020-06-03:
> @balloob thanks, can you suggest the procedure to upgrade from this stage? is there a way to put the new image on the sd card via a linux machine?

https://www.raspberrypi.org/documentation/installation/installing-images/linux.md

ohadlevy, 2020-06-03:
thanks, but thats not what I had in mind... resolved anyway..