home-assistant/operating-system #711: Disk usage increases very rapidly after latest updates

tosion, 2020-06-03:
I have realized in my systems running Homeassistant in NUC (NUC image) that disk usage increases very rapidly. I think this happened within few days ago after updates with supervisor 226/HassOS 4.8/Grafana/Influx to latest versions.

For a long time my system monitor showed around 4 GiB used and 99 Gib free. Now it shows 7 GiB used and 96GiB free?

Below the status of my two NUCs with Disk Use:

I have realized in my systems running Homeassistant in NUC (NUC image) that disk usage increases very rapidly. I think this happened within few days ago after updates with supervisor/HassOS/Grafana/Influx to latest versions.

For a long time my system monitor showed around 4 GiB used and 99 Gib free. Now it shows 7 GiB used and 96GiB free?

Anyone else noticing same behaviour?

Below the status of one my two NUCs with Disk Use:

[image]

[image]

brinzlee, 2020-06-03:
There is a memory leak somewhere....I'm noticing the same issue this was HassOS 4.8

tomlut, 2020-06-03:
Yeah I've noticed an increase of ~5GB disk use per day since updating. My disk use was essentially flat for months before the update to 4.8. 
Using HassOS NUC image.

frenck, 2020-06-03:
Any indication on where the disk space grows? As in, a specific folder?

tomlut, 2020-06-03:
No not yet. It does not appear to be in /config. Or swapfile.

Apologies but I overstated my issue.

I thought I had a 500G SSD but it's only 125G. So the 1% increase today was only 1GB not 5GB. And it seems to have stopped increasing.

tosion, 2020-06-03:
I would like to help but I am not sure what to look for. Wit Win10/Samba I do not see too much. When connected thru RPI3B+/Rasbian (maybe first time in my life - I see this - But what to look, that's a question) 

[image]

mbo18, 2020-06-03:
I have the same issue, NUC, HA OS 4.8, HA 110.4. Growing by 220Mb per day (including 30Mb per day for my DB)
I did not find what's taking so much space. It does not seems to be an add-on as I have removed them all (except deconz) and re-installed them later

tomlut, 2020-06-03:
At the CLI try:
```
du -sh /*
```
To get a list of directory sizes.

tosion, 2020-06-03:
This?

[image]

mbo18, 2020-06-03:
Did it also in / using SSH Add-on from Franck. Made a diff a day later and only changed I saw is the size of the config folder due to my DB growing

boekhorstr, 2020-06-03:
I have noticed the same issue since the upgrade to HassOS 4.8. A little investigation led me to the systemd journals in /mnt/data/logs. In a few days time it grew to 4GB. 

[image]

So I wondered if this is normal/desired behaviour. But it seems that size limitations on the systemd journal config (/etc/systemd/journald.conf) have been removed in #649

balloob, 2020-06-03:
We made this change to align how other operating systems like Raspbian and Debian configure systemd. By limiting things we lose important logs that help with debugging.

Limiting used size would not help in reducing disk writes, as the `MaxUse` settings would remove old logs, while new logs are still written to disk.

For the default settings that apply now, see https://www.freedesktop.org/software/systemd/man/journald.conf.html#SystemMaxUse=

closed 2020-06-03 by balloob

tosion, 2020-06-04:
Thanks for this valuable piece of info!
 
If anyone knows is this disk use just going increase or stop somehow, I would appreciate that info also.

balloob, 2020-06-04:
It's in the docs I linked :)

tosion, 2020-06-04:
I tried my best to read it but with my background (you would be surprised how far I have gone with this HA as a passenger with no idea how to drive - many thanks for you guys!) understanding part was a bit difficult ;-)

mbo18, 2020-06-04:
If I understand correctly, 10 and 15% of the file system size with a maximum of 4Gb

`The first pair defaults to 10% and the second to 15% of the size of the respective file system, but each value is capped to 4G `

balloob, 2020-06-04:
This part is also important: "something else causes the file system to fill up, journald will stop using more space"

tosion, 2020-06-04:
Excellent - so I have plenty of room with my 120 G SSD ... and nothing to worry about

aechelon, 2020-06-18:
Is there an easy way to restore the original MaxUse settings for journald.conf?
Looks like the file system is mounted read-only.

aechelon, 2020-06-25:
This change is rather annoying for those of us who run HASSOS inside a VM and keep that image lean to accelerate snapshots.

Would it be possible to get an option to configure the MaxUse variables (or Storage=volatile) through the supervisor or even through ssh?

For those who are interested in a temporary solution, once you have ssh access to the host (see [here](https://developers.home-assistant.io/docs/operating-system/debugging/)) you can recreate the original journald.conf on a writable portion of the file system (e.g. under /root/), and then replace the default journald.conf with the modified version using mount -o bind.  Once that's done just reload journald:

```
mount -o bind /root/journald.conf /etc/systemd/journald.conf
systemctl force-reload systemd-journald
```

I've tried to get this running at boot through udev rules (see [here](https://community.home-assistant.io/t/solved-how-to-include-a-script-to-initialize-usb-device-on-hass-io-hassos-startup/93987)) but systemd-udev won't persist the mount outside the udev rule without mountflags=shared, so for now I'm stuck running the script manually after reboots (which luckily are infrequent).

CarlosGS, 2021-02-12:
@aechelon I'm stuck here as well, did you find a long term solution?

The journald.conf default parameters are incorrect.

aechelon, 2021-02-12:
Not yet.  Currently I have to remember to run a script that does the _mount -o bind_ on the host whenever I reboot the VM.  Luckily recently I haven't had to reboot very often.
I took the time to shrink the VM as much as possible and I keep an eye on the size of the backups I perform before any major upgrade.

hosoludi, 2021-05-25:
Same Problem here

soundslikeluke, 2021-08-12:
I am running on a Nuc On a VM and my storage has gone up to 300GB and I really dont know what is using it. The Home-assistant_v2.db is 29gb but that doesnt explain the 300GB disk ussage.

RobertusIT, 2021-08-17:
> I am running on a Nuc On a VM and my storage has gone up to 300GB and I really dont know what is using it. The Home-assistant_v2.db is 29gb but that doesnt explain the 300GB disk ussage.

same here and no one, know how to fix it. is very annoing

agners, 2021-08-18:
As state above, the journal grows to a 4GB max.

@RobertusIT  if your system fills up 300GB, there must be another reason and it probably warrants a separate issue/investigation. What add-ons are you using?

As for option to set size for the journal: TBH, what is 4GB these days? I think its well worth the debugging capabilities it brings.

RobertusIT, 2021-08-18:
> As state above, the journal grows to a 4GB max.
>
> @RobertusIT if your system fills up 300GB, there must be another reason and it probably warrants a separate issue/investigation. What add-ons are you [...]

[image]

As you can see, i have only some plugin actived.

Frigate get audio and video, but system, grow up since without frigate, so isn't frigate the problem.
As you can see, frigate, get only 3GB for now
[image]

My HA proxmox
[image]

And my local-lvm, as you can see, very annoing that grow up a lot
[image]

33.44GB of what ? :(

Every monday, you can see a decrease of space used, because i create a service with fstrim -a:
```
# systemctl 
fstrim.timer                                                                                           loaded active waiting   Discard unused blocks once a week 
```
I have also, for now, recorder in **configuration.yaml**:
`  purge_keep_days: 1`

How can i figure out ?

Thanks to reply

it4rz4n, 2021-11-09:
Did you ever figure this out? I have the exact same issue.

mofonics, 2022-09-15:
Did this get solved? I have the same problem

LookInIs, 2023-02-17:
Did this get solved? I have the same problem

jgoakley, 2025-02-18:
Still having this problem as well.

Running a bare-metal x86-64 install with a 128GB SSD and /mnt/data/logs grew to 53.4 GB leaving less than 5% of the disk available. Currently on HAOS 14.1.

Only just realized it when one of my backups failed due to lack of storage space.

[image]

[image]