home-assistant/operating-system #712: [  216.799302]   fe.dai-link-0:    ASoC:    No backend DAI enabled for fe.dai-link-0

jrhodk, 2020-06-03:
**HassOS release with the issue:**

Release 4.9 running on Odroid N2 on eMMC
It freezes after some time with a screen full of this message: 
[  216.799302]   fe.dai-link-0:    ASoC:    No backend DAI enabled for fe.dai-link-0
where the number in the start counts up..

**Supervisor logs:**

I cant run anything because the system is frozen

**Journal logs:**

**Kernel logs:**

**Description of problem:**
It freezes after some time with a screen full of this message: 
[  216.799302]   fe.dai-link-0:    ASoC:    No backend DAI enabled for fe.dai-link-0

I have had this problem since v. 4.5. This time i did start with fresh install with only the samba plugin and the deConz zigbee driver installed.

I have 2 Ordoid N2 with eMMC, they both crash with the fe.dai-link-0 message,

balloob, 2020-06-03:
1. How did you flash the eMMC?
2. Do you have any USB sticks inserted?
3. What kind of network do you have attached?

jrhodk, 2020-06-04:
1. I did the flashing with the Hardkernel eMMC flasher: https://www.hardkernel.com/shop/emmc-module-reader-board-for-os-upgrade/. 
2. There are only the deConz zigbee USB stick connected.
3. The Odroid N2 are connected directly to my networkwith ethernet cable.

jrhodk, 2020-06-04:
I get the ui, and it works fine until the system decides to freeze. Which eMMC drive do you use and which flasher?

balloob, 2020-06-04:
I use https://www.hardkernel.com/shop/usb3-0-emmc-module-writer/ with [Etcher](https://www.balena.io/etcher/)

jrhodk, 2020-06-04:
And your eMMC module?

balloob, 2020-06-04:
16GB from ODroid website.

balloob, 2020-06-04:
I checked with some people. 

```
 [ 216.799302] fe.dai-link-0: ASoC: No backend DAI enabled for fe.dai-link-0
```

This message is about the DAC, which is not supported. That should not be the cause for freezing.

jrhodk, 2020-06-04:
[image]

jrhodk, 2020-06-04:
I caught some kernelpanick on the screen after a crash! - Hope this gives a hint anout whats wrong!

jrhodk, 2020-06-04:
What could have corrupted the kernel stack? Is it a buffer overflow error, or a memory hardware error og something else?

balloob, 2020-06-05:
I don't know 🤷

pvizeli, 2020-06-05:
Can you complete reflash the system with 4.9

If you said, he freeze -> no blue light is blinking?

jrhodk, 2020-06-05:
I have reflashed it every time it did freeze, and yes freeze = no blue "heartbeat". I have downloaded a minimal ubuntu for ordroid n2 and plan to install some server on it, just to see if it crashes without homeassistant.

pvizeli, 2020-06-05:
I will create a private build with:
https://github.com/home-assistant/operating-system/pull/724

So you can try it out to check if that would work for you

Willybits, 2020-06-05:
If needed I can also flash the test build and see how it works on my end?
For me the latest stable version has been 4.6

If there is anything else I can do to test for you just let me know.

pvizeli, 2020-06-05:
https://drive.google.com/file/d/1_1pyDXcmGWAfBGUXEdPYquT56rfrVBbg/view?usp=sharing

jrhodk, 2020-06-05:
Its installing at the moment, interesting to see how it goes! - Im grateful for the time you use on this!

Willybits, 2020-06-05:
Thank you very much for your time and effort on everything HASS 👍 

Currently running the test version.
Here are some observations:
```
1. Installation had some issues like not wanting to automatically reboot, the heartbeat kept going very fast. Had to unplug it to reboot it.
2. After import of snapshot, the history did not work, deleted the database and turned off the device. Turning off did not work, had to manually unplug power
```
After this, the device ran fine. Let it go without doing anything with it.
After a while the following error came up:

```
[32m[ 1082.996662] [0m[33mrc rc0[0m[1m: two consecutive events of type space[0m
[32m[ 1742.683837] [0m[1m------------[ cut here ]------------[0m
[32m[ 1742.694136] [0m[1mURB 00000000383e0d56 submitted while active[0m
[32m[ 1742.704542] [0m[33mWARNING[0m[1m: CPU: 0 PID: 7976 at drivers/usb/core/urb.c:363 usb_submit_urb+0x534/0x540[0m
[32m[ 1742.717980] [0m[33mModules linked in[0m[1m: xfrm_user xfrm_algo cfg80211 rfkill snd_soc_hdmi_codec dw_hdmi_cec dw_hdmi_i2s_audio realtek rc_odroid meson_dw_hdmi dwmac_generic dw_hdmi meson_ir meson_drm rc_core crct10dif_ce drm_kms_helper ao_cec_g12a snd_soc_meson_axg_sound_card snd_soc_meson_card_utils rtc_meson_vrtc snd_soc_meson_g12a_tohdmitx cec meson_canvas drm snd_soc_meson_codec_glue dwmac_meson8b stmmac_platform snd_soc_meson_axg_tdmout stmmac snd_soc_meson_axg_frddr snd_soc_meson_axg_fifo mdio_xpcs snd_soc_meson_axg_tdm_interface snd_soc_meson_axg_tdm_formatter cdc_acm[0m
[32m[ 1742.797610] [0m[33mCPU[0m[1m: 0 PID: 7976 Comm: kworker/0:0 Not tainted 5.7.0 #1[0m
[32m[ 1742.810519] [0m[33mHardware name[0m[1m: Hardkernel ODROID-N2 (DT)[0m
[32m[ 1742.822945] [0m[33mWorkqueue[0m[1m: events acm_softint [cdc_acm][0m
[32m[ 1742.835461] [0m[33mpstate[0m[1m: 40000005 (nZcv daif -PAN -UAO)[0m
[32m[ 1742.848093] [0m[33mpc [0m[1m: usb_submit_urb+0x534/0x540[0m
[32m[ 1742.860794] [0m[33mlr [0m[1m: usb_submit_urb+0x534/0x540[0m
[32m[ 1742.873486] [0m[33msp [0m[1m: ffff800014fb3cc0[0m
[32m[ 1742.886217] [0m[33mx29[0m[1m: ffff800014fb3cc0 x28: ffff800011337000 [0m
[32m[ 1742.899085] [0m[33mx27[0m[1m: 0000000000000000 x26: ffff80001203bce8 [0m
[32m[ 1742.912017] [0m[33mx25[0m[1m: 0000000000000001 x24: ffff0000d5131798 [0m
[32m[ 1742.924979] [0m[33mx23[0m[1m: ffff0000d5131448 x22: 00000000fffffff0 [0m
[32m[ 1742.938061] [0m[33mx21[0m[1m: ffff0000f3d22940 x20: 0000000000000008 [0m
[32m[ 1742.951317] [0m[33mx19[0m[1m: ffff0000dd024900 x18: 0000000000000010 [0m
[32m[ 1742.964476] [0m[33mx17[0m[1m: 0000000000000000 x16: 0000000000000000 [0m
[32m[ 1742.977663] [0m[33mx15[0m[1m: 0000000000aaaaaa x14: 0000000000000020 [0m
[32m[ 1742.990868] [0m[33mx13[0m[1m: 00000000ffffffff x12: ffff800010d383f8 [0m
[32m[ 1743.004124] [0m[33mx11[0m[1m: 0000000000000001 x10: 0000000000000000 [0m
[32m[ 1743.017469] [0m[33mx9 [0m[1m: 0000000000000001 x8 : ffff8000129d9220 [0m
[32m[ 1743.030860] [0m[33mx7 [0m[1m: 0000000000aaaaaa x6 : 00000000000007e5 [0m
[32m[ 1743.044302] [0m[33mx5 [0m[1m: 0000000000000000 x4 : ffff0000f47761c8 [0m
[32m[ 1743.057810] [0m[33mx3 [0m[1m: ffff0000f477d078 x2 : ffff0000f47761c8 [0m
[32m[ 1743.071340] [0m[33mx1 [0m[1m: 17984e3b464b0200 x0 : 0000000000000000 [0m
[32m[ 1743.084975] [0m[1mCall trace:[0m
[32m[ 1743.098471] [0m[1m usb_submit_urb+0x534/0x540[0m
[32m[ 1743.111932] [0m[1m acm_submit_read_urb+0xa4/0x100 [cdc_acm][0m
[32m[ 1743.125530] [0m[1m acm_softint+0x170/0x1d0 [cdc_acm][0m
[32m[ 1743.139096] [0m[1m process_one_work+0x1b4/0x330[0m
[32m[ 1743.152560] [0m[1m worker_thread+0x48/0x420[0m
[32m[ 1743.166027] [0m[1m kthread+0x138/0x160[0m
[32m[ 1743.179407] [0m[1m ret_from_fork+0x10/0x18[0m
[32m[ 1743.192579] [0m[1m---[ end trace 9641a926b1151bd3 ]---[0m
[32m[ 1743.206211] [0m[7m[31mUnable to handle kernel NULL pointer dereference at virtual address 0000000000000001[0m
[32m[ 1743.223090] [0m[1m------------[ cut here ]------------[0m
[32m[ 1743.235285] [0m[33mWARNING[0m[1m: CPU: 0 PID: 9 at kernel/workqueue.c:1473 __queue_work+0x2fc/0x3b0[0m
[32m[ 1743.249282] [0m[33mModules linked in[0m[1m: xfrm_user xfrm_algo cfg80211 rfkill snd_soc_hdmi_codec dw_hdmi_cec dw_hdmi_i2s_audio realtek rc_odroid meson_dw_hdmi dwmac_generic dw_hdmi meson_ir meson_drm rc_core crct10dif_ce drm_kms_helper ao_cec_g12a snd_soc_meson_axg_sound_card snd_soc_meson_card_utils rtc_meson_vrtc snd_soc_meson_g12a_tohdmitx cec meson_canvas drm snd_soc_meson_codec_glue dwmac_meson8b stmmac_platform snd_soc_meson_axg_tdmout stmmac snd_soc_meson_axg_frddr snd_soc_meson_axg_fifo mdio_xpcs snd_soc_meson_axg_tdm_interface snd_soc_meson_axg_tdm_formatter cdc_acm[0m
[32m[ 1743.330728] [0m[33mCPU[0m[1m: 0 PID: 9 Comm: ksoftirqd/0 Tainted: G        W         5.7.0 #1[0m
[32m[ 1743.344575] [0m[33mHardware name[0m[1m: Hardkernel ODROID-N2 (DT)[0m
[32m[ 1743.356490] [0m[33mpstate[0m[1m: 20000085 (nzCv daIf -PAN -UAO)[0m
[32m[ 1743.368040] [0m[33mpc [0m[1m: __queue_work+0x2fc/0x3b0[0m
[32m[ 1743.379078] [0m[33mlr [0m[1m: __queue_work+0xec/0x3b0[0m
[32m[ 1743.389817] [0m[33msp [0m[1m: ffff80001007bba0[0m
[32m[ 1743.400411] [0m[33mx29[0m[1m: ffff80001007bba0 x28: ffff0000f3cb6e00 [0m
[32m[ 1743.410968] [0m[33mx27[0m[1m: ffff0000f477f880 x26: 00000000000000c0 [0m
[... cut ...]
[32m[ 3395.974311] [0m[33mcdc_acm 1-1.2:1.1[0m[31m: urb 3 failed submission with -16[0m
[32m[ 3410.973714] [0m[33mcdc_acm 1-1.2:1.1[0m[31m: urb 3 failed submission with -16[0m
[32m[ 3425.977691] [0m[33mcdc_acm 1-1.2:1.1[0m[31m: urb 3 failed submission with -16[0m
[32m[ 3441.010689] [0m[33mcdc_acm 1-1.2:1.1[0m[31m: urb 3 failed submission with -16[0m
[32m[ 3456.010645] [0m[33mcdc_acm 1-1.2:1.1[0m[31m: urb 3 failed submission with -16[0m
```
It is still running in this state at the moment

pvizeli, 2020-06-05:
Version 4.4 work?

Willybits, 2020-06-05:
Version 4.6 was the first I used and it is stable, everything else after this has these issues for me.
I am using emmc

pvizeli, 2020-06-05:
Could be something like: https://magazine.odroid.com/article/repairing-your-odroid-n2-how-to-recover-from-an-accidental-short-circuit/

If you run 4.4 unstable, I think it will be a hardware issue

Willybits, 2020-06-05:
Would be weird then that 4.6 is stable for me if it is a hardware issue?

I will install 4.4 now and see how that runs.

Willybits, 2020-06-05:
Ok, 4.4 is running now for about an hour.

Observations:
Installation went smooth, reboots worked, import of snapshot worked. Only issue was that the history did not work and had to delete the database. After that, rebooted again, worked fine. Started up and all is running fine.
No weird errors or issues yet, will keep it running and report back.

jrhodk, 2020-06-05:
The installation went fine. The screen on the odroid shows this:
[image]

Willybits, 2020-06-06:
Goodmorning, after a good nights sleep I woke up to a purring HassOS 4.4.

It is still running without any vissible issues for me.
Automations still run fine, Heartrate is normal.

If at any time you would like access to my system using Teamviewer or Anydesk or other means, let me know. You are all a lot better at the inner workings of the system.

Just let me know what is needed and I'll get that info for you :)

Thanks!

Here is the kernel log from yesterday evening till now:
```
[32m[  547.292634] [0m[33m fe.dai-link-0[0m[31m: ASoC: no backend DAIs enabled for fe.dai-link-0[0m
[32m[  547.299662] [0m[33m fe.dai-link-0[0m[31m: ASoC: no backend DAIs enabled for fe.dai-link-0[0m
[32m[ 7131.071214] [0m[33mhrtimer[0m[1m: interrupt took 733951 ns[0m
[32m[32865.481297] [0m[33mrc rc0[0m[1m: two consecutive events of type space[0m
[32m[44392.678373] [0m[33mhassio[0m: port 11(veth12c6d5b) entered blocking state
[32m[44392.680267] [0m[33mhassio[0m: port 11(veth12c6d5b) entered disabled state
[32m[44392.685788] [0mdevice veth12c6d5b entered promiscuous mode
[32m[44392.690331] [0m[33mkauditd_printk_skb[0m[1m: 5 callbacks suppressed[0m
[32m[44392.690339] [0m[33maudit[0m: type=1700 audit(1591430610.423:237): dev=veth12c6d5b prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[32m[44393.170057] [0m[33maudit[0m: type=1325 audit(1591430610.915:238): table=nat family=2 entries=0
[32m[44393.184344] [0m[33maudit[0m: type=1325 audit(1591430610.931:239): table=filter family=2 entries=0
[32m[44393.203911] [0m[33maudit[0m: type=1325 audit(1591430610.951:240): table=nat family=2 entries=5
[32m[44393.210784] [0m[33maudit[0m: type=1325 audit(1591430610.955:241): table=nat family=2 entries=7
[32m[44393.222741] [0m[33maudit[0m: type=1325 audit(1591430610.967:242): table=nat family=2 entries=8
[32m[44393.228763] [0m[33maudit[0m: type=1325 audit(1591430610.975:243): table=nat family=2 entries=10
[32m[44393.237357] [0m[33maudit[0m: type=1325 audit(1591430610.983:244): table=nat family=2 entries=11
[32m[44393.244676] [0m[33maudit[0m: type=1325 audit(1591430610.991:245): table=nat family=2 entries=12
[32m[44393.251499] [0m[33maudit[0m: type=1325 audit(1591430610.999:246): table=nat family=2 entries=13
[32m[44393.303124] [0m[33meth0[0m: renamed from veth9b0bc6f
[32m[44393.348658] [0m[33mIPv6[0m: ADDRCONF(NETDEV_CHANGE): veth12c6d5b: link becomes ready
[32m[44393.351958] [0m[33mhassio[0m: port 11(veth12c6d5b) entered blocking state
[32m[44393.357633] [0m[33mhassio[0m: port 11(veth12c6d5b) entered forwarding state
[32m[44396.568611] [0m[33mhassio[0m: port 11(veth12c6d5b) entered disabled state
[32m[44396.570452] [0m[33mveth9b0bc6f[0m: renamed from eth0
[32m[44396.652278] [0m[33mhassio[0m: port 11(veth12c6d5b) entered disabled state
[32m[44396.654600] [0mdevice veth12c6d5b left promiscuous mode
[32m[44396.658562] [0m[33mhassio[0m: port 11(veth12c6d5b) entered disabled state
# 
```

ckhyatt, 2020-06-06:
I am experiencing the same issue on a clean flash of 4.9 on a new n2. I will try a flash of 4.6.

Thanks!

Willybits, 2020-06-06:
I just installed HassOS 4.10

After installing all looks good, trying to reboot the system gives the following error:
Is it having a hard time with killing audio stuff???

[image]

Then unplugging power en booting again the system comes up fine.
Then within 15 minutes the system started to act up again like Deconz not being able to control the lights, and sensors not being renewed anymore.
Staring with the following in the kernel log:
```
[32m[  707.175603] [0m[1m------------[ cut here ]------------[0m
[32m[  707.191788] [0m[1mURB 00000000933a1cca submitted while active[0m
[32m[  707.196063] [0m[33mWARNING[0m[1m: CPU: 0 PID: 399 at drivers/usb/core/urb.c:363 usb_submit_urb+0x534/0x540[0m
[32m[  707.204095] [0m[33mModules linked in[0m[1m: xfrm_user xfrm_algo cfg80211 rfkill snd_soc_hdmi_codec dw_hdmi_cec dw_hdmi_i2s_audio realtek dwmac_generic meson_dw_hdmi rc_odroid dw_hdmi dwmac_meson8b meson_drm stmmac_platform snd_soc_meson_axg_frddr drm_kms_helper rtc_meson_vrtc meson_ir ao_cec_g12a rc_core snd_soc_meson_g12a_tohdmitx cec snd_soc_meson_g12a_toacodec snd_soc_meson_codec_glue drm snd_soc_meson_axg_tdmout snd_soc_meson_axg_fifo meson_canvas stmmac snd_soc_meson_t9015 crct10dif_ce mdio_xpcs snd_soc_meson_axg_sound_card snd_soc_meson_card_utils snd_soc_meson_axg_tdm_interface snd_soc_spdif_tx snd_soc_meson_axg_tdm_formatter snd_soc_simple_amplifier cdc_acm[0m
[32m[  707.271447] [0m[33mCPU[0m[1m: 0 PID: 399 Comm: kworker/0:3 Not tainted 5.7.0 #1[0m
[32m[  707.276915] [0m[33mHardware name[0m[1m: Hardkernel ODROID-N2 (DT)[0m
[32m[  707.281913] [0m[33mWorkqueue[0m[1m: events acm_softint [cdc_acm][0m
[32m[  707.286954] [0m[33mpstate[0m[1m: 40000005 (nZcv daif -PAN -UAO)[0m
[32m[  707.292031] [0m[33mpc [0m[1m: usb_submit_urb+0x534/0x540[0m
[32m[  707.297120] [0m[33mlr [0m[1m: usb_submit_urb+0x534/0x540[0m
[32m[  707.302222] [0m[33msp [0m[1m: ffff800012f53cc0[0m
[32m[  707.307314] [0m[33mx29[0m[1m: ffff800012f53cc0 x28: ffff800011337000 [0m
[32m[  707.312450] [0m[33mx27[0m[1m: 0000000000000000 x26: ffff80001005bce8 [0m
[32m[  707.317607] [0m[33mx25[0m[1m: 0000000000000001 x24: ffff0000d50b1798 [0m
[32m[  707.322783] [0m[33mx23[0m[1m: ffff0000d50b1448 x22: 00000000fffffff0 [0m
[32m[  707.328023] [0m[33mx21[0m[1m: ffff0000f3dca940 x20: 0000000000000008 [0m
[32m[  707.333362] [0m[33mx19[0m[1m: ffff0000db45f400 x18: 0000000000000010 [0m
[32m[  707.338627] [0m[33mx17[0m[1m: 0000000000000000 x16: 0000000000000000 [0m
[32m[  707.343889] [0m[33mx15[0m[1m: 0000000000aaaaaa x14: 0000000000000020 [0m
[32m[  707.349145] [0m[33mx13[0m[1m: 00000000ffffffff x12: ffff800010d383f8 [0m
[32m[  707.354415] [0m[33mx11[0m[1m: 0000000000000001 x10: 0000000000000000 [0m
[32m[  707.359710] [0m[33mx9 [0m[1m: 0000000000000001 x8 : ffff800012ad9220 [0m
[32m[  707.365022] [0m[33mx7 [0m[1m: 0000000000aaaaaa x6 : 00000000000007ed [0m
[32m[  707.370352] [0m[33mx5 [0m[1m: 0000000000000000 x4 : ffff0000f47751c8 [0m
[32m[  707.375706] [0m[33mx3 [0m[1m: ffff0000f477c078 x2 : ffff0000f47751c8 [0m
[32m[  707.381073] [0m[33mx1 [0m[1m: 86df69ad6c1f4000 x0 : 0000000000000000 [0m
[32m[  707.386403] [0m[1mCall trace:[0m
[32m[  707.391688] [0m[1m usb_submit_urb+0x534/0x540[0m
[32m[  707.397013] [0m[1m acm_submit_read_urb+0xa4/0x100 [cdc_acm][0m
[32m[  707.402333] [0m[1m acm_softint+0x170/0x1d0 [cdc_acm][0m
[32m[  707.407631] [0m[1m process_one_work+0x1b4/0x330[0m
[32m[  707.412933] [0m[1m worker_thread+0x48/0x420[0m
[32m[  707.418204] [0m[1m kthread+0x138/0x160[0m
[32m[  707.423395] [0m[1m ret_from_fork+0x10/0x18[0m
[32m[  707.428451] [0m[1m---[ end trace 7bd457ac87291602 ]---[0m
[32m[  707.433642] [0m[7m[31mUnable to handle kernel NULL pointer dereference at virtual address 0000000000000001[0m
[32m[  707.441703] [0m[7m[31mMem abort info:[0m
[32m[  707.446368] [0m[7m[31m  ESR = 0x96000006[0m
[32m[  707.450998] [0m[33m  EC = 0x25[0m[7m[31m: DABT (current EL), IL = 32 bits[0m
[32m[  707.455507] [0m[7m[31m  SET = 0, FnV = 0[0m
[32m[  707.459792] [0m[7m[31m  EA = 0, S1PTW = 0[0m
[32m[  707.463878] [0m[7m[31mData abort info:[0m
[32m[  707.467828] [0m[7m[31m  ISV = 0, ISS = 0x00000006[0m
[32m[  707.471733] [0m[7m[31m  CM = 0, WnR = 0[0m
[32m[  707.475459] [0m[33muser pgtable[0m[7m[31m: 4k pages, 48-bit VAs, pgdp=0000000093d14000[0m
[32m[  707.479909] [0m[7m[31m[0000000000000001] pgd=000000007a67f003, pud=000000007a680003, pmd=0000000000000000[0m
[32m[  707.488466] [0m[1m------------[ cut here ]------------[0m
[32m[  707.488482] [0m[33mWARNING[0m[1m: CPU: 0 PID: 399 at kernel/workqueue.c:1473 __queue_work+0x2fc/0x3b0[0m
[32m[  707.488484] [0m[33mModules linked in[0m[1m: xfrm_user xfrm_algo cfg80211 rfkill snd_soc_hdmi_codec dw_hdmi_cec dw_hdmi_i2s_audio realtek dwmac_generic meson_dw_hdmi rc_odroid dw_hdmi dwmac_meson8b meson_drm stmmac_platform snd_soc_meson_axg_frddr drm_kms_helper rtc_meson_vrtc meson_ir ao_cec_g12a rc_core snd_soc_meson_g12a_tohdmitx cec snd_soc_meson_g12a_toacodec snd_soc_meson_codec_glue drm snd_soc_meson_axg_tdmout snd_soc_meson_axg_fifo meson_canvas stmmac snd_soc_meson_t9015 crct10dif_ce mdio_xpcs snd_soc_meson_axg_sound_card snd_soc_meson_card_utils snd_soc_meson_axg_tdm_interface snd_soc_spdif_tx snd_soc_meson_axg_tdm_formatter snd_soc_simple_amplifier cdc_acm[0m
[... cut ...]
[32m[  708.063342] [0m[33mcdc_acm 1-1.2:1.1[0m[31m: urb 3 failed submission with -16
```
Then it repeats this every x minutes or so.

```
[  940.026147] cdc_acm 1-1.2:1.1: urb 3 failed submission with -16
[  955.024655] cdc_acm 1-1.2:1.1: urb 3 failed submission with -16
[  970.111112] cdc_acm 1-1.2:1.1: urb 3 failed submission with -16
[  985.113731] cdc_acm 1-1.2:1.1: urb 3 failed submission with -16
[ 1000.154804] cdc_acm 1-1.2:1.1: urb 3 failed submission with -16
```
Could it have something to do with sound or does 4.6 also have sound enabled?

jrhodk, 2020-06-07:
Since somebody have told that they run Hassos on Odroid N2 without crashes, i have tried to update the firmware from 20190416 to the latest 2020 version. Hope that does some difference. The v.5 hassos pvizeli did create did also crash.

Willybits, 2020-06-07:
jrhodk, does <=4.6 also crash for you?

jrhodk, 2020-06-13:
it isnt usb crash, but emmc crash!

BasR-NL, 2020-06-13:
Experienced same problem on my N2. 

On hassos 4.9 and 4.10 I see kernel dumps / traces in the log (but HA keeps working), and USB starting to fail at some point after reboot, causing deconz to no longer see my conbee stick. 

Seeing the "urb 3 failed submission with.." error in the logs as well. 

Reboot fixes it (usually power cycle only way, supervisor or CLI commands to reboot don't work) 

Problems reappear fairly quickly after reboot. 

Back on Hassos 4.6, issues seem gone, all good for last 20 hours.

Willybits, 2020-06-25:
> We need the output of:
> `lsusb -vv`
> 
> If the USB device is attached in.

Do we need to do this in HassOS 4.10 or is HassOS 4.6 also ok?
If we need to do this in HassOS 4.10, do you want this before the crash or after the usb-crash?

Thanks!

BasR-NL, 2020-06-25:
Here's the output of lsusb -vv on my HassOS 4.6:

[image]

I have 2 sticks installed:

- Conbee (firmware 264A0700 , Deconz = 2.05.75)
- Aeotec z-wave stick

on HassOS 4.9/4.10 when I experienced the issues, the USB port seemed to fail completely (until reboot), but only for Conbee stick. Z-wave kept working.

Today after examining logs, I am noticing something in my current setup with HassOS 4.6 as well. The setup is working for both Z-Wave and Zigbee sticks, I don't have USB ports failing permanently (until reboot). BUT , I do see USB errors in the system log:

[image]

These errors first appeared after 41 hours of running, and then they keep repeating but it's not constant. Now after uptime of 9 days, I have 32 occurences. Seems something is having issues with USB, and it's temporarily disabling the port, but only with the Conbee stick.

My idea about this:

- On HassOS 4.6 the re-enabling of the port leads to reconnect of the USB stick and it keeps working (see screenshot),

- On HassOS4.9 / 4.10 I think the event causes more issues and takes down USB functionality until reboot

Willybits, 2020-06-25:
Output on my HassOs 4.6:

# lsusb -vv
Bus 002 Device 002: ID 05e3:0620
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 05e3:0610
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 004: ID 1cf1:0030
#

pvizeli, 2020-07-02:
Can someone try 4.11?

BasR-NL, 2020-07-02:
I've updated to 4.11

[image]

For me it runs well so far , I don't experience any USB issues to this point. Both USB sticks are working (Aeotec and Conbee 2). 

I did see 2 kernel warnings. See log at https://pastebin.com/S2RXLX1C

Willybits, 2020-07-02:
Runs fine for me as well for the past 2 hours.

BasR-NL, 2020-07-02:
Usb disconnected after approx 2.5 hours. 
See log below. 
I verified, Conbee stick is gone, not visible in deconz. 

[ 1645.099065] hrtimer: interrupt took 949880 ns
[ 2396.793370] rc rc0: two consecutive events of type space
[ 7689.137128] audit: type=1334 audit(1593722459.971:208): prog-id=13 op=LOAD
[ 7689.141152] audit: type=1334 audit(1593722459.975:209): prog-id=14 op=LOAD
[ 7719.444377] audit: type=1334 audit(1593722490.280:210): prog-id=14 op=UNLOAD
[ 7719.453100] audit: type=1334 audit(1593722490.280:211): prog-id=13 op=UNLOAD
[ 9368.266652] usb 1-1.2: USB disconnect, device number 5
[ 9368.275296] cdc_acm 1-1.2:1.0: failed to set dtr/rts

Willybits, 2020-07-03:
For me it keeps on working.
Did an update to Hass 0.112.1 this morning, first reboot did not start, had to cut power. After that, boot ok, update ok, reboot ok and it is still running since. That's 3 hours now.
Will report back this evening and see if it keeps on going.

Willybits, 2020-07-03:
Well, for me it crashed USB now as well. Same as before, but it took longer.

BasR-NL, 2020-07-03:
I went back to 4.6, did that yesterday right after I had the issues on 4.11. No issues on 4.6 so far, USB keeps working and also no USB problems reported in the logs (also not the disconnect / reconnect I experienced earlier)

BasR-NL, 2020-07-04:
Now I do have the same situation as before, still on 4.6, usb keeps working, but I see disconnect / reconnect again.

88369.323306] usb 1-1.2: USB disconnect, device number 4
[88369.335990] cdc_acm 1-1.2:1.0: failed to set dtr/rts
[88369.601887] usb 1-1.2: new full-speed USB device number 6 using xhci-hcd
[88369.803120] cdc_acm 1-1.2:1.0: ttyACM2: USB ACM device
[89421.738638] usb 1-1.2: USB disconnect, device number 6
[89422.048865] usb 1-1.2: new full-speed USB device number 7 using xhci-hcd
[89422.252232] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[89425.834749] usb 1-1.2: USB disconnect, device number 7
[89425.851306] cdc_acm 1-1.2:1.0: failed to set dtr/rts
[89426.144581] usb 1-1.2: new full-speed USB device number 8 using xhci-hcd
[89426.347146] cdc_acm 1-1.2:1.0: ttyACM2: USB ACM device
[93027.754320] usb 1-1.2: USB disconnect, device number 8
[93028.036704] usb 1-1.2: new full-speed USB device number 9 using xhci-hcd
[93028.748856] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[93031.595120] usb 1-1.2: USB disconnect, device number 9
[93031.875595] usb 1-1.2: new full-speed USB device number 10 using xhci-hcd
[93032.074686] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device

Willybits, 2020-08-09:
Anyone tried 4.12 yet to see if this issue is resolved?

dziobson, 2020-09-07:
Has anyone managed to fix this problem.
It still occurs for me.
Latest stable release of HassOS 4.12, and pre-release 5.2

agners, 2020-10-21:
@jrhodk @Willybits @BasR-NL @dziobson the latest 4.15 release should fix the issue by making sure the USB ACM error recovery is working again. Would be good if you can test 4.15 to see if this did the trick. It is not yet on the stable channel, but you can manually installing it using `ha os update --version 4.15`.

Willybits, 2020-10-22:
I installed it, updated from 4.6 to 4.15.
Update went smooth, will report back tomorrow with a progress report.

pvizeli, 2020-10-22:
Next 5.4 will include the official bugfix which going into next 4.16 release

Willybits, 2020-10-23:
So far so good!
No issues at all, any version after 4.6 has never worked for so many hours as this one.
Will keep you posted!

agners, 2020-11-03:
Consider this closed as we have the revert in 4.15 and the official fix in 4.16.

closed 2020-11-03 by agners