home-assistant/operating-system #713: Upgrade RPI3B+ to 4.8 kills my install

stunney, 2020-06-03:
I have a feeling that this is closely related to the recently closed issue:

https://github.com/home-assistant/operating-system/issues/703

I am *not* using UART.

My config.txt file:
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# HassOS - don't change it!
disable_splash=1
kernel=u-boot.bin

# uncomment for aarch64 bit support
#arm_64bit=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[all]
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
```

Device does *not* boot.  Doesn't even POST on the display.

I upgraded from 3.13 from the Web UI button.

AdighaLogic, 2020-06-04:
Same here just tried to upgrade from 3.13 to 4.8 left my Rpi for 30min to boot but still wouldn't.
Had to hard reset it 3-4 times to get it come back. it has 3.13 atm.
I'm sorry i couldn't find any logs to share.
Everything else is up to date.

MichaelBitard, 2020-06-04:
Is there an easy way to revert to 3.13?

MichaelBitard, 2020-06-04:
I was able to do that (awesome work with the rauc images by the way) by booting on the second slot (the one with the 3.13)

ssh on your machine

```bash
$rauc status
Compatible:  hassos-rpi3                                                                                               
Variant:                                                                                                                                                                                                                                       
Booted from: kernel.0 (A)                                                                                                                                                                                                                      
Activated:   kernel.0 (A)                                                                                              
slot states:                                                                                                           
  boot.0: class=boot, device=/dev/disk/by-partlabel/hassos-boot, type=vfat, bootname=(NULL)                            
      state=inactive, description=, parent=(none), mountpoint=(none)                                                   

  kernel.0: class=kernel, device=/dev/disk/by-partlabel/hassos-kernel0, type=ext4, bootname=A
      state=booted, description=, parent=(none), mountpoint=(none)
      boot status=good
  rootfs.0: class=rootfs, device=/dev/disk/by-partlabel/hassos-system0, type=raw, bootname=(NULL)
      state=active, description=, parent=kernel.0, mountpoint=(none)

  kernel.1: class=kernel, device=/dev/disk/by-partlabel/hassos-kernel1, type=ext4, bootname=B
      state=inactive, description=, parent=(none), mountpoint=(none)
      boot status=good
  rootfs.1: class=rootfs, device=/dev/disk/by-partlabel/hassos-system1, type=raw, bootname=(NULL)
      state=inactive, description=, parent=kernel.1, mountpoint=(none)
```

`rauc status mark-active other` then reboot

martymarty004, 2020-06-04:
We had a long discussion about this problem with 4.8, 4.9 should have fixed it. (you can see the discussion and the multiple solutions on this thread  #703 and on the [HA comunity](https://community.home-assistant.io/t/hassos-4-8-upgrade-breaks-my-install/201022/88))

pvizeli, 2020-06-08:
Fix with 4.10

closed 2020-06-08 by pvizeli