home-assistant/operating-system #714: HassOS 4.9 won't install on Raspberry Pi 3b+

chriselera, 2020-06-04:
**HassOS release with the issue:**
4.9

**Supervisor logs:**
N/A

**Journal logs:**
N/A

**Kernel logs:**
N/A

**Description of problem:**
HassOS 4.9 won't install on Raspberry Pi 3b+.  Downloaded 4.9 image, flashed to SD card using Balena Etcher, plugged in to Pi3.  Installation process does notstart and http://<Pi's Ip address>:8123 is unreachable even after 10 minutes.  Checked on router and Pi 3 is not online.  Note: this is the same issue with 4.7 and 4.8.

Flashed 3.13 to the same SD card and HassOS installs fine on the same Pi 3.

Visje31, 2020-06-04:
I'm having the same issue.
Update of HassOS doesn't work. Not from the interface nor from the shell.
There is enough disk space, only way back is to cut the power, disconnect the pi and remove the SD-card. After that when trying to reboot the OS will revert to version 3.13.
(I know, sounds silly, but only that way I get the OS to revert without flashing the card again.)

Br4v3St4rr, 2020-06-04:
Also having the same issue.  Attempt to log into CLI reports

`Error: No such container: hassio_cli`

Same fix as @Visje31, not sure about removing the SD Card but a couple of powercycles reverted it.

brusarp, 2020-06-04:
@Visje31 and @Br4v3St4rr are you talking about upgrading to 4.** 8 ** or 4.** 9 ** version?

pverberckmoes, 2020-06-04:
I've got the same.
Manual update won't work because the 4.9 version is not visible.
(
[image]

)

Visje31, 2020-06-04:
I was upgrtading from 3.13 to the latest 4.9 version

Tom-ahawk, 2020-06-04:
I have tried 4 times now,  upgrading from 3.13 to the latest 4.8 version. Every time I have to restart 3 times to go back  to 3.13.

brusarp, 2020-06-04:
This issue was opened for 4.**9**
Don't go OT...

Sagitr, 2020-06-04:
After upgrade to 4.8 RPI4 and 4.9 the system will no longer boot. Not visible on the network. Just the end. Many users have this problem. reinstallation from an older backup will help. :(

0plus6, 2020-06-04:
> 
> 
> I have tried 4 times now, upgrading from 3.13 to the latest 4.8 version. Every time I have to restart 3 times to go back to 3.13.

Same. 4.8 or 4.9, doesn't matter. Claims installed, reboots and still 3.13.

schusz, 2020-06-04:
Powercycling 5 times worked for me to revert HassOS v3.13.

phcella, 2020-06-04:
I'm on 4.8 and now it's saying to "upgrate" to 3.13.  Is it right?

pverberckmoes, 2020-06-04:
> I'm on 4.8 and now it's saying to "upgrate" to 3.13. Is it right?
I've got the same. Expected to see 4.9 but did the downgrade. System is stable now and I'm waiting for 3.9 and some positive reviews before upgrading.

TopoDiFogna, 2020-06-05:
> I'm on 4.8 and now it's saying to "upgrate" to 3.13. Is it right?

Same here, there's the 4.9 for rpi3 but the system doesn't detect it and instead suggests 3.13.

Adorem, 2020-06-05:
IDK why but yes for now for rpi2 and rpi3 the version is now back to "3.13".
https://version.home-assistant.io/beta.json

TopoDiFogna, 2020-06-05:
Also for stable version, noticed that rpi3 64bit is 4.9

https://version.home-assistant.io/stable.json

balloob, 2020-06-05:
We've set it back to 3.13 while we address some of the issues that have come up.

pvizeli, 2020-06-08:
Fix with 4.10

closed 2020-06-08 by pvizeli

Adorem, 2020-06-08:
Tested just now 4.10 on my rpi and it's working fine this time. Thanks :+1:

phcella, 2020-06-08:
Working like a charm for me too. Thanks

Visje31, 2020-06-08:
Works great! thanks!