home-assistant/operating-system #715: No Raspberry Pi 2 upgrade to 4.9?

slim0287, 2020-06-04:
There doesn't appear to be a Pi2 version available for 4.9 despite one of the "Changed" items being "Fix issue with RPi2 armhf / armv7."  I see _rpi, _rpi0, _rpi3, _rpi4 however _rpi2 is missing.

martymarty004, 2020-06-04:
I have the same question

martymarty004, 2020-06-04:
@slim0287  It's out now, probably they had to finish compiling it

balloob, 2020-06-04:
Each board needs to be build separately, it takes time. You can follow releases here https://dev.azure.com/home-assistant/Operating-System/_build?definitionId=58

closed 2020-06-04 by balloob

bcmitri, 2020-06-05:
> Each board needs to be build separately, it takes time. You can follow releases here https://dev.azure.com/home-assistant/Operating-System/_build?definitionId=58

Not to be impatient, but how long does this normally take?  I'm still only seeing 3.13 for the rpi 2 and rpi 3.