home-assistant/operating-system #716: Zwave not working on RPI under OS 4.8 OR 4.9

Villa-Buenavista, 2020-06-04:
Hi there still no working ZWave driver under OS 4.9. Any help is greatly appreciated!

ZWave log:
2020-06-04 13:51:46.971 Always, OpenZwave Version 1.4.3469 Starting Up
2020-06-04 13:52:40.976 Info, Setting Up Provided Network Key for Secure Communications
2020-06-04 13:52:40.977 Info, mgr,     Added driver for controller /dev/ttyAMA0
2020-06-04 13:52:40.977 Info,   Opening controller /dev/ttyAMA0
2020-06-04 13:52:40.977 Info, Trying to open serial port /dev/ttyAMA0 (attempt 1)
2020-06-04 13:52:40.978 Info, Serial port /dev/ttyAMA0 opened (attempt 1)
2020-06-04 13:52:40.978 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2020-06-04 13:52:40.978 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2020-06-04 13:52:40.978 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2020-06-04 13:52:40.978 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2020-06-04 13:52:40.978 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2020-06-04 13:52:40.979 Detail,
2020-06-04 13:52:40.979 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2020-06-04 13:52:40.979 Detail, contrlr, Notification: DriverFailed

adrianrdzv, 2020-06-04:
Same with Elelabs Zigbee Shield. Does not work with 4.8 or 4.9. Reverted to 3.13 and works fine.

Using default 3.13 config file with extra line:
```
dtoverlay=pi3-miniuart-bt
```

Sagitr, 2020-06-04:
After upgrade to 4.8 RPI4 and 4.9 the system will no longer boot. Not visible on the network. Just the end. Many users have this problem. reinstallation from an older backup will help. :(

olapettersson, 2020-06-04:
Same problem here. RPi4B and Aeotech Z-stick Gen 5 via a USB2-hub. Tried downgrading with `ha os upgrade --version 3.13` and the system came backup, but several addon's didn't, like Grafana and Node-RED. After another reboot attempt, the system doesn't start, although something was doing something, because I got my TTS-startup message on my Google Home, and my Philips Hue flashed (as per my startup-automation). I guess my only choice now is to re-install, downgrade to 3.13 and restore my latest snapshot, which is a bit of a bummer. :-/

lastessa, 2020-06-04:
same problem on 4.8 rpi4 32 bit

Villa-Buenavista, 2020-06-05:
https://github.com/home-assistant/operating-system/issues/716

nickbarau, 2020-06-07:
Having issues getting Elelabs USB Zigbee stick running on 4.9 0.110.5

2020-06-07 18:48:17 ERROR (MainThread) [zigpy.application] Couldn't start application
2020-06-07 18:48:17 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 217, in async_setup
    hass, self
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 141, in async_initialize
    app_config, auto_form=True, start_radio=True
  File "/usr/local/lib/python3.7/site-packages/zigpy/application.py", line 65, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 138, in startup
    await self.initialize()
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 88, in initialize
    await self._cfg(t.EzspConfigId[config], value)
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 230, in _cfg
    assert v[0] == t.EmberStatus.SUCCESS  # TODO: Better check
AssertionError
2020-06-07 18:48:17 DEBUG (bellows.thread_0) [bellows.uart] Closed serial connection

czesiu20, 2020-06-07:
Same problem. Z-wave is not working on RPi4B (32-bit) with Aeotech Z-stick Gen 5. I tried to use Razbery2 but after adding enable_uart=1 to the config file the system is not starting.

pvizeli, 2020-06-08:
Fix with 4.10

closed 2020-06-08 by pvizeli