home-assistant/operating-system #718: Q: How to access /share from a docker container added w/ Portainer addon?

PieBru, 2020-06-04:
**HassOS release with the issue:** 4.8

arch | aarch64
-- | --
chassis | embedded
dev | false
docker | true
hassio | true
host_os | HassOS 4.8
installation_type | Home Assistant
os_name | Linux
os_version | 5.4.42-v8
python_version | 3.7.7
supervisor | 227
timezone | Europe/Rome
version | 0.110.4
virtualenv | false

**Supervisor logs:** N/A

**Journal logs:** N/A

**Kernel logs:** N/A

**Description of problem:**
On HassOS with the Portainer addon I can successfully install JellyFin. However I didn't find a way to give to the JellyFin container the RW access to the `/share` folder, pretty much as the MotionEye addon already does.
I added to JellyFin container SYS_ADMIN and tryed with PRIVILEGED both YES and NO, nothing changed.

Via ssh, executing `mount | grep share` the mount is RW, but the JellyFin container can't start because of the error below, it seems `/share` isn't really `shared`:
[image]

In my home and my camper van I'm trying to embrace the HassOS way instead of the deprecated "supervised". Especially on the camper van (offgrid) it isn't feasible more than a RasPI always powered on.

NOTE: before you suggest the Plex addon, I strongly prefer JellyFin because it is stable open source, it gives more than I need, and it isn't a privacy nightmare ;-) so a JellyFin addon may be a nice touch, but I'm not able to write it.

IMO the difference between the HassOS "appliance" and the "supervised" paradigms can be mitigated if our beloved additional functions can be added as a container having access only to the base resources. Just my 2 cents.

Thank you all for the shiny HASS ecosystem.
Piero

PieBru, 2020-06-04:
Forgot to mention I [already asked on the community forum](https://community.home-assistant.io/t/hassos-help-please-how-to-access-share-from-a-docker-container-added-w-portainer/200585) and on the homeassistant discord channel, both without success.

pvizeli, 2020-06-05:
Portainer is not supported and also not run container outside of add-ons

closed 2020-06-05 by pvizeli