home-assistant/operating-system #719: Wrong Update

PierreScerri, 2020-06-04:
**HassOS release with the issue:**

0.110.4

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
It is showing the wrong update:
[image]

manuelbarbosa, 2020-06-04:
This update will actually make the HomeassistantOS unresponsive :P

PierreScerri, 2020-06-04:
I don't get how changing from 4.8 to 3.13 is an update.

manuelbarbosa, 2020-06-04:
Ok it took a long time but reverted back to 4.8

ludeeus, 2020-06-04:
It is not wrong, <https://version.home-assistant.io/stable.json> 3.13 is the latest stable version for some boards.
If that "update" is to a lower version, it is still considered an update since you currently are not running the latest stable version.

RIKWEBB, 2020-06-04:
Well, I'm puzzled !
I'm running 4.8, but I believe there is also a 4.9 out there, yet the 'Update' is for 3.13
I think I will wait and see how the dust settles on this one!

ludeeus, 2020-06-04:
You are running rpi2 or rpi3
```
    "rpi2": "3.13",
    "rpi3": "3.13",
```

PierreScerri, 2020-06-04:
That is confusing. My board did not change. It's the same Rpi3 I had for a while.  How come it was progressing along the 4.x line and suddenly it's 3.13?

WeaselBreath, 2020-06-04:
release notes link point to v 3.13 not 4.9. I will hold off on this update til they get it fixed

ludeeus, 2020-06-04:
@PierreScerri can you close this?
This should be a feature request in the frontend repo (to show "downgrade" instead of "upgrade") on that button.
This is not an issue with the OS, but a missing feature in the frontend.

PierreScerri, 2020-06-04:
Sorry but I don't get it

closed 2020-06-04 by PierreScerri

WeaselBreath, 2020-06-04:
why confuse ppl????

RIKWEBB, 2020-06-04:
RPi3 - but its not changed.... where did 4.8 come from?

WeaselBreath, 2020-06-04:
why would anyone push a downgrade button???

ludeeus, 2020-06-04:
One reason: You are running the beta, no longer want beta, you want back to the stable.
(the same applies to home assistant core, supervisor and the OS)

RIKWEBB, 2020-06-04:
HassOS Release-4 build 8 (Stable) was not beta - or I certainly did not select it as such

RIKWEBB, 2020-06-04:
Is this the reason? @ludeeus 
https://community.home-assistant.io/t/downgrading-hassos-4-8/201695

ludeeus, 2020-06-04:
https://github.com/home-assistant/frontend/blob/dev/hassio/src/dashboard/hassio-update.ts#L45-L47 is the reason if version A is not the same as version B, it presents an update to version B

RIKWEBB, 2020-06-04:
> https://github.com/home-assistant/frontend/blob/dev/hassio/src/dashboard/hassio-update.ts#L45-L47 is the reason if version A is not the same as version B, it presents an update to version B

Not helpful! (biting my lip a little, as Im not quite sure your intentions)
I was just trying to understand why I had 4.8 and it seems to being downgraded. 
If you cant answer that's fine. I will assume the reason for downgrade is the one I found searching.

chris669, 2020-06-05:
Not clear at all, I was never running beta and sometimes ago I was asked to update to 4.8. And now I would have to rollback to 3.13??? Wait and see...

PierreScerri, 2020-06-05:
After backing up, I took the plunge and went to 3.13.  Home assistant started without a hitch. It even reduce the number of log errors. 
And it's not prompting for any updates. I'm happy with that.

CaptTom9, 2020-06-05:
4.8 caused massive problems when it was rolled out as allegedly "stable." It really messed up my RPi 3B+. It was completely unreachable via HTTP or SSH. Fortunately, it took me a few days before I could really devote any time to diagnosing it, and by then they'd retrieved it and set 3.13 (the release I had been running before) back to the latest "stable." Magically on the third reboot, I was safely rolled back and everything started working again.

So, if you want to remain stable, revert back to 3.13. But if 4.8 (or 4.9, which was also floating around for a short while) works for you, then leave well enough alone. Personally, I do not plan to update again until whatever release comes next has had time to prove itself.

RIKWEBB, 2020-06-05:
@CaptTom9  - Thank you :)
That was the sort of information I was looking for and very helpful.
I'm on 4.8 and have no major issues, and plan to do as you suggest 'leave well alone' 
I think, like you, I will be considering carefully any future updates until I see feedback from a number of days.

Rik 👍

PierreScerri, 2020-06-05:
@CaptTom9 - That seems to make sense. Thank you.
As I said in my previous comment.  I reverted to 3.13 without any problems first time.  It actually improved the system as 4.8 was throwing all sorts of errors.

chris669, 2020-06-05:
back to 3.13, everything seems to be ok.

schergr, 2020-06-05:
So, should we be going to 3.13 from 4.8?  Is this being "forced"...will there be a 4.9 that moves us forward, rather then backward?

PierreScerri, 2020-06-05:
I don't believe anyone is forcing anyone to do anything.
@CaptTom9 offers a sensible explanation and some good advice.
If it ain't broke don't fix it, otherwise make a backup and try to fix it.  If that involves 'updating' to 3.13 then so be it.

In my case, reverting to 3.13 fixed it.

BillionReyz, 2020-06-06:
> This "update" will actually make the HomeassistantOS unresponsive :P

I got the same issue. I tried to ssh 22222 into the system but received 
jq: error: Could not open file /mnt/data/hassos.json: No such file or directory

CaptTom9, 2020-06-06:
> This "update" will actually make the HomeassistantOS unresponsive :P

Just to be clear, it's the 4.8 update which made HA unresponsive for me, and from what I've read, for others as well.

So, if you saw a prompt to "update" from 4.8 to 3.13, that was intentional. That would get you back to a stable release.

chris669, 2020-06-08:
Yes, now will try the 4.10! So exciting.

CaptTom9, 2020-06-08:
> Yes, now will try the 4.10! So exciting.

As an FYI, I was reading a post over on the https://community.home-assistant.io/ forum from one user who reported problems after updating to 4.10. He/She had updated from 4.8, without going back to 3.13, so it's possible whatever problem 4.8 had introduced was simply carried forward. Others have updated to 4.10 (apparently) without issue.

Personally, I'm waiting this one out until I can dedicate a day or two to re-installing. I've lost all confidence in the beta testing for this product. Someone on that other forum posted that they simply don't have enough volunteers to test properly. But I can't help but notice they have plenty of volunteers willing to make coding changes. I get it. I've been there. Coding is fun. Testing isn't.

PierreScerri, 2020-06-08:
Updated to 4.10 without any problems (for now).  Log reports only warnings about using custom integrations.  Haven't seen such a clean log in a while. Yay!

chris669, 2020-06-09:
Following the "treasure hunt" and desperately waiting the rollback in 1.29, this is just a psychological test, be patient!