home-assistant/operating-system #720: Pi 4B GPIO DHT sensor returns "RuntimeError: Unknown Platform" error

DoctorOctagonapus, 2020-06-04:
**HassOS release with the issue:**

4.8 and 4.9
**Supervisor logs:**

Nothing of relevance
**Journal logs:**

```
Jun 04 21:15:53 homeassistant 871b551da784[403]: 2020-06-04 22:15:53 ERROR (MainThread) [homeassistant.components.sensor] dht: Error on device update!
Jun 04 21:15:53 homeassistant 871b551da784[403]: Traceback (most recent call last):
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 322, in _async_add_entity
Jun 04 21:15:53 homeassistant 871b551da784[403]:     await entity.async_device_update(warning=False)
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 472, in async_device_update
Jun 04 21:15:53 homeassistant 871b551da784[403]:     await self.hass.async_add_executor_job(self.update)
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
Jun 04 21:15:53 homeassistant 871b551da784[403]:     result = self.fn(*self.args, **self.kwargs)
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/src/homeassistant/homeassistant/components/dht/sensor.py", line 137, in update
Jun 04 21:15:53 homeassistant 871b551da784[403]:     self.dht_client.update()
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/src/homeassistant/homeassistant/util/__init__.py", line 239, in wrapper
Jun 04 21:15:53 homeassistant 871b551da784[403]:     result = method(*args, **kwargs)
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/src/homeassistant/homeassistant/components/dht/sensor.py", line 173, in update
Jun 04 21:15:53 homeassistant 871b551da784[403]:     humidity, temperature = self.adafruit_dht.read_retry(self.sensor, self.pin)
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 94, in read_retry
Jun 04 21:15:53 homeassistant 871b551da784[403]:     humidity, temperature = read(sensor, pin, platform)
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 80, in read
Jun 04 21:15:53 homeassistant 871b551da784[403]:     platform = get_platform()
Jun 04 21:15:53 homeassistant 871b551da784[403]:   File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 63, in get_platform
Jun 04 21:15:53 homeassistant 871b551da784[403]:     raise RuntimeError('Unknown platform.')
Jun 04 21:15:53 homeassistant 871b551da784[403]: RuntimeError: Unknown platform.
Jun 04 21:15:53 homeassistant 871b551da784[403]: 2020-06-04 22:15:53 INFO (MainThread) [homeassistant.setup] Setting up device_tracker
```
**Kernel logs:**

Nothing of relevance
**Description of problem:**
I have an open issue on Homeassistant Core here: https://github.com/home-assistant/core/issues/36374, but raising an issue here as it seems like a HassOS issue.

Since upgrading to 4.8, the GPIO DHT sensor component will not start. Instead, the following stacktrace appears in the Core logs:

```
Logger: homeassistant.components.sensor
Source: components/dht/sensor.py:173
Integration: Sensor (documentation, issues)
First occurred: 22:15:53 (2 occurrences)
Last logged: 22:15:53

dht: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 322, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 472, in async_device_update
    await self.hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/dht/sensor.py", line 137, in update
    self.dht_client.update()
  File "/usr/src/homeassistant/homeassistant/util/__init__.py", line 239, in wrapper
    result = method(*args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/dht/sensor.py", line 173, in update
    humidity, temperature = self.adafruit_dht.read_retry(self.sensor, self.pin)
  File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 94, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 80, in read
    platform = get_platform()
  File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 63, in get_platform
    raise RuntimeError('Unknown platform.')
RuntimeError: Unknown platform.
```
This is not an issue with core as I've been running the latest version of Core for some days before upgrading HassOS. It was only the act of upgrading HassOS that broke the component.

pvizeli, 2020-06-08:
Fix with 4.10

closed 2020-06-08 by pvizeli

mehdi7129, 2020-06-08:
Same issue in 4.10

DoctorOctagonapus, 2020-06-08:
You must have a separate issue. Mine started working again after updating to 4.10.

mehdi7129, 2020-06-08:
Maybe...
What's in your configuration.yalm file? I just added this at the end of the file :

sensor:
  platform: dht
  sensor: DHT22
  pin: 4
  monitored_conditions:
    - temperature
    - humidity

In reference of this : https://www.home-assistant.io/integrations/dht/

Do i need to add or configure something? On raspian, when I run a .py, the DHT22 sensor works fine.
I'm not an expert ^^

DoctorOctagonapus, 2020-06-08:
Hm that looks OK, I use a different pin so mine's slighty different:

``` yaml
  - platform: dht
    name: "Living Room"
    sensor: DHT22
    pin: 17
    monitored_conditions:
      - 'temperature'
      - 'humidity'
```
Only obvious difference I can see is I have my monitored conditions in single quote marks, though there's nothing about that in the official docs so I might not need them.

mehdi7129, 2020-06-08:
Thanks for your help.
I have a pi zero-w and it's listed as not recommended hardware, so I'll try with a pi 3B+ 32bit tonight. But i saw some guys doing good stuff with HA and a pi zero ^^

https://www.home-assistant.io/hassio/installation/