home-assistant/operating-system #729: On 4.8 offered to install 3.13

fechy, 2020-06-06:
My Raspberry Pi running on latest HA (0.110.5), and HassOS 4.8, is offering to "update" to 3.13.

[image]

Here's the system tab info:
[image]

frenck, 2020-06-06:
4.8 is not marked stable yet for all boards. It is offering you an "upgrade" to the latest stable.

closed 2020-06-06 by frenck

fechy, 2020-06-06:
Ok, thanks.

Any reason it offered me to upgrade to 4.8 in the first place if it wasn't stable? I'm not on enrolled in beta testing AFAIK and I don't normally upgrade if it's not stable.

isamuk, 2020-06-06:
I have the same issue - I am on HaaOS 4.8  the thing is I see no issues and read some concerns about zwave but am not experiencing them (yet) - RPi3+ 0.110.4 - @fechy were you able to go back to 3.13 as is suggested? It was not obvious to me the supervisor recommended update to 4.8 would move the environment to an unstable release.  Good to know and I guess more research before moving is the approach.

fechy, 2020-06-06:
I haven't tried. I'm afraid going back in versions might leave my system unusable. 
I don't think this issue should be closed though, the version was offered as stable otherwise I would not install it.

I'll try to do a full back up first and see if the regression works.

isamuk, 2020-06-07:
Yeah I will need to figure out which backup has my n-1 release and go back there. From what I am seeing going back from 4.8 to 3.13 may be a problem. Painful. 4.8 is simply being described as a dev release.

tofuSCHNITZEL, 2020-06-07:
going back to 3.13 from 4.8 broke everything for me and I had to do a complete reinstall...

1 Jun 2020 4.8 was set to stable for rpi and then reverted to 3.13 on the 2nd (but only for rpi2) 
on the 4th were 4.9 was set to stable for all other rpis 3.13 was set for the rpi3

and since its stable for all other rpis it seems more of an oversight to me...

fechy, 2020-06-07:
I guess I'll leave my Rpi as it is now and wait until they catch up or finally acknowledge this as a problem and offer a solution that doesn't include reinstall all. 

@frenck since you closed this issue, would you mind take a look at possible solutions we can try?

JackPoint, 2020-06-08:
I also use the stable channel but got an upgrade to 4.8. A few days later I got a message to update it to 3.13. I downgraded to 3.13 and it works fine.

Now I get a message to update to 4.10.
[image]

This https://version.home-assistant.io/stable.json says thats te latest stable version. How do I know that's real this time? I don't see any messages about 4.10 on the twitter channels.

allanpersson, 2020-06-08:
@JackPoint of course you know it real this time. Roll backs and/or human mistakes will always be possible but we all have to acknowledge and appreciate the hard work developers are doing both when successful and when mistakes happens 🙏

Give it a go and else you can get back to another version 😉