home-assistant/operating-system #731: Cannot Deploy OVA v4.9 onto ESXi 6.7

tiempososcuros, 2020-06-08:
**HassOS release with the issue:**
hassos_ova-4.9.ova

**Supervisor logs:**
N/A

**Journal logs:**
N/A

**Kernel logs:**
N/A

**Description of problem:**
I am attempting to import the OVA image onto a VMWare ESXi 6.7 host I have in my lab. I am receiving the following error when attempting to deploy:

`Issues detected with selected template. Details: - 23:5:PARSE_ERROR: Parse error: Undeclared namespace prefix "vmw" (for attribute "osType") at [row,col,system-id]: [23,74,"descriptor.ovf"].`

If there are any further details needed just let me know. Thanks!

frenck, 2020-06-08:
This should be solved with the 4.10 version, could you please try that one?

tiempososcuros, 2020-06-08:
With hassos_ova-4.10.ova, same error message:
`Issues detected with selected template. Details: - 23:5:PARSE_ERROR: Parse error: Undeclared namespace prefix "vmw" (for attribute "osType") at [row,col,system-id]: [23,74,"descriptor.ovf"].`

frenck, 2020-06-08:
Yep, I see the issue, opening a PR to address this. Thanks for reporting! :+1:

PR #732 (2020-06-08): Fix missing vmw namespace in ovf definition
Add in the missing `vmw` namespace declaration, which prevents the OVA from importing on VMWare platforms.

fixes #731
closed 2020-06-08 by pvizeli