home-assistant/operating-system #733: Ver 4.10 RPI4 WiFi don't work anymore.

dianlight, 2020-06-08:
**HassOS release with the issue:**
- Operating System: 4.10

**Kernel logs:**
.. alot of log like:
[  556.305711] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  558.917741] brcmfmac: brcmf_set_key_mgmt: invalid cipher group (1027074)
[  558.917766] brcmfmac: brcmf_cfg80211_connect: wl_set_key_mgmt failed (-22)
....

**Description of problem:**
After uprgading my RPi4 installation to ver 4.10 the wifi don't work anymore.
I also try with a new installation and the problem is the same. No Wifi connection. If I revert to a 4.9 backup with the same config all work.

This is my-network config:

```
[connection]
id=my-network
uuid=72111c67-4a5d-4d5c-925e-f8ee26efb3c3
type=802-11-wireless

[802-11-wireless]
mode=infrastructure
ssid=LTOfficenet
# Uncomment below if your SSID is not broadcasted
#hidden=true

[802-11-wireless-security]
auth-alg=open
key-mgmt=wpa-psk
psk=<private>

[ipv4]
method=auto

[ipv6]
addr-gen-mode=stable-privacy
method=auto
```

cogneato, 2020-06-09:
What is the hardware for this install?

dianlight, 2020-06-09:
RasperryPi 4 Model B 4GB.

da-anda, 2020-06-09:
missing kernel driver due to the kernel downgrade to 4.19 in #723 ?

LoermansA, 2020-06-11:
Can I revert back to 4.9 without a reinstall? I have this same issue. Plugging in an ethernet cable doesn't seem to work. I have no monitor connected to the pi so I cannot see the bootlog. So; no network, no display but I do hava access to the sdcard ofcourse. Can I downgrade?

LoermansA, 2020-06-11:
I managed to get the pi running on eth0 instead of wifi by altering the NetworkManager config in one of the overlay partitions (i think). So it is running again. Now to get wifi back online...

cogneato, 2020-06-11:
The network config file @dianlight is using as formatted does not work for me. Creating a file with the following format does work on my pi4 when imported from usb CONFIG stick (the name of the file does not matter, as long as it is in CONFIG/network/ and has no extension).

I had no issue connecting to wifi using nmcli commands.

```
[connection]
id=wovenomena
uuid=9344b185-51e9-4483-921d-d64572f9bf21
type=wifi
interface-name=wlan0
permissions=

[wifi]
mac-address-blacklist=
mode=infrastructure
ssid=wovenomena

[wifi-security]
auth-alg=open
key-mgmt=wpa-psk
psk=test password

[ipv4]
dns-search=
method=auto

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=auto

[proxy]
```

jetatomic, 2020-06-12:
I'm getting the same errors on RPi 3B+.

```
...brcmf_set_key_mgmt: invalid cipher group
...wl_set_key_mgmt failed
```

First appeared on upgrade to 4.10 via Supervisor GUI. Still happening on a clean install with new 4.10 image.

dianlight, 2020-06-12:
@cogneato don't understand the difference format of your config and why my format not work for you. ( my come from official documentation )

I also made other tests by changing the wifi network and I discovered that the problem is the connection to the Wifi Mesh. If I connect to a Wifi without Mesh ( or disabled) it works.

 @jetatomic , @LoermansA your wifi is mesh or normal? 

This are my tests, Work on:
- [x] `Iphone hotspot IOS 13.5.1`
- [x] `Deco M5 Firmware 1.4.4` --> ***Mash OFF***
- [x] `Sercom VSDL router (TIM Firmware)`

Don't work on:
- [ ] `Deco M5 Firmware 1.4.4` --> ***Mash ON***
- [ ] `Google Wifi Firmware 12371.52.22` --> ***Mash ON*** ( I don't know how disable mash )

Is not a hw issue because version 3.13 and 4.9 works and also raspbian works.

L.

LoermansA, 2020-06-12:
@dianlight yes I'm running a Deco M9 mesh.

I'm oke with running with a wired connection for the time being but that seems to have broken now also. There were some updates last night (supervisor and home assistant) that I ran thinking they might have solved the wifi problem, but as of now I cannot get hassio up, even with a wired connection...

cogneato, 2020-06-12:
@dianlight

The one I posted comes from creating a working connection using nmcli and then looking at the resulting network file as generated by the system itself. I then used that to create another working connection. I am aware of the official docs, and that is why I shared.

LoermansA, 2020-06-12:
With a wired connection I now get this:

`jun 12 18:42:08 homeassistant systemd[1]: Starting HassOS supervisor...
jun 12 18:42:08 homeassistant docker[2087]: Error response from daemon: No such container: hassio_supervisor
jun 12 18:42:08 homeassistant systemd[1]: Started HassOS supervisor.
jun 12 18:42:08 homeassistant hassos-supervisor[2099]: Error: No such object: homeassistant/aarch64-hassio-supervisor
jun 12 18:42:08 homeassistant hassos-supervisor[2117]: Error: No such object: hassio_supervisor
jun 12 18:42:08 homeassistant hassos-supervisor[2097]: [WARNING] Supervisor image missing, downloading a fresh one: latest
jun 12 18:42:23 homeassistant dockerd[372]: time="2020-06-12T16:42:23.549237299Z" level=warning msg="Error getting v2 registry: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeou>
jun 12 18:42:23 homeassistant dockerd[372]: time="2020-06-12T16:42:23.549426076Z" level=info msg="Attempting next endpoint for pull after error: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connecti>
jun 12 18:42:23 homeassistant dockerd[372]: time="2020-06-12T16:42:23.549635204Z" level=error msg="Handler for POST /v1.40/images/create returned error: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for >
jun 12 18:42:23 homeassistant hassos-supervisor[2136]: Error response from daemon: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
jun 12 18:42:23 homeassistant hassos-supervisor[2097]: [WARNING] Supervisor downloading failed trying: latest
jun 12 18:42:38 homeassistant dockerd[372]: time="2020-06-12T16:42:38.644248273Z" level=warning msg="Error getting v2 registry: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeou>
jun 12 18:42:38 homeassistant dockerd[372]: time="2020-06-12T16:42:38.644396680Z" level=info msg="Attempting next endpoint for pull after error: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connecti>
jun 12 18:42:38 homeassistant dockerd[372]: time="2020-06-12T16:42:38.644586939Z" level=error msg="Handler for POST /v1.40/images/create returned error: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for >
jun 12 18:42:38 homeassistant hassos-supervisor[2153]: Error response from daemon: Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
jun 12 18:42:38 homeassistant systemd[1]: hassos-supervisor.service: Main process exited, code=exited, status=1/FAILURE
jun 12 18:42:38 homeassistant systemd[1]: hassos-supervisor.service: Failed with result 'exit-code'.
jun 12 18:42:43 homeassistant systemd[1]: hassos-supervisor.service: Scheduled restart job, restart counter is at 19.
jun 12 18:42:43 homeassistant systemd[1]: Stopped HassOS supervisor.
`
Perhaps this does not belong in this issue anymore... 

It appears that hassi is missing the superviser and is trying to download the latest but failing...

jetatomic, 2020-06-13:
> @jetatomic , @LoermansA your wifi is mesh or normal?

I don't have a mesh. 802.11r fast roaming might be enabled, but there aren't any other APs in range.

I fiddled with this for a while, including swapping in @cogneato 's config (adjusting for my network), but kept getting the same errors on wifi. It worked fine on eth in my tests. At the moment, I just need it to work, so I went back to a Supervised install, and wifi is working fine.

LoermansA, 2020-06-15:
I've reverted to 4.9 using a clean install and restoring a snapshot. Everything is working again. I'll skip 4.10 for now.

corvy, 2020-06-16:
Any way to downgrade HassOS? I have the same issue. Wi-Fi not working, but wired is fine.

kirava, 2020-06-16:
I had a similar issue when I upgraded to 4.10 with raspberry connected to a Deco M5 MESH wifi.
I connected screen & keyboard, boot the raspberry up and ran:
`ha os update --version 4.9`
at first it gave me an error but after a while downgraded to 4.9 and wifi was functional again

corvy, 2020-06-16:
Thank you! I could not get it to work so I have started a new install with snapshot restore to get back on 4.9. Should be working within a few hours (hopefully before my wife gets home!).

lishan89uc, 2020-06-17:
Not sure if my issue is related to 4.10 but it only started after i upgraded from 3.13 to 4.10. And i read something similar on a very old non HASSOS related thread. 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=160021&sid=a21a78934349f5cb1e04057f8a13d52d&start=50
[image]

edwardreed81, 2020-06-17:
I had these wifi issues on my raspberry pi 3 with 4.10 and fixed them by disabling 802.11r (Fast Roaming) on my wifi network (Unifi).

corvy, 2020-06-17:
Good tip. FYI I also have Unifi with fast roaming enabled. Have now turned it off pending next OS release.

azan3l4t, 2020-06-18:
Similar issue here, using Rpi3+ and wifi mesh from Intelbras with Fast Roaming. 
WIFI Stopped working. Tried disabling Fast Roaming, but still not working. 

Have changed to wired connection and it's working fine on wired ethernet.

guilhermelirio, 2020-06-22:
Same here. I using Rasp Zero W with Home Assistant, and nota connect in Mesh wi-fi (Tenda MW6). I'm testing in my Android hotspot qnd working, but i need to connect in my wi-fi :/

corvy, 2020-07-05:
Anyone tested 4.11? I still havent mustered enough motivation.... 👍

LoermansA, 2020-07-05:
> Anyone tested 4.11? I still havent mustered enough motivation.... +1

The new release contains a new(er) kernel; 4.19.127 instead of 4.19.126. Can't find anything about it containing any fixes for the Wifi issue, so I'll pass on this one until an official update is given regarding this issue.

kirava, 2020-07-05:
> > Anyone tested 4.11? I still havent mustered enough motivation.... +1
> 
> The new release contains a new(er) kernel; 4.19.127 instead of 4.19.126. Can't find anything about it containing any fixes for the Wifi issue, so I'll pass on this one until an official update is given regarding this issue.

Same here :(

guilhermelirio, 2020-07-05:
I disabled the Fast Roaming function on my router and it worked first, in version 4.10.

corvy, 2020-07-05:
I did aswell but it still does not work.

corvy, 2020-08-10:
The new 4.12 HassOS solves this problem. :) I think this issue can be closed. This is my Ubiquity Mesh settings and a working HassIO. 

[image]

agners, 2020-08-10:
@corvy hm, that is a bit curious, the kernel hash has been updated with 4.12, but there were only 3 commits made to it. And they do not seem to fix a WiFi related issue: https://github.com/raspberrypi/linux/commits/rpi-4.19.y. The bump to the Buildroot release 2020.02.3 also brought some updates to some user-space packages, but it also does not seem like something relevant got updated... Are you sure this bug was still present in 4.11? Otherwise I guess good thing if its resolved... :-)

danilogd, 2020-08-10:
I tried the 4.12 HassOS on a Raspberry Pi 3 B+ before I update my working Homeassistant RPi and the issue isn't resolved yet. I have a TP-Link Deco M5 mesh wireless network, working on both 2,4 and 5 Ghz. Every time I try to connect to WiFi it says "invalid cipher group", but this issue is not present on release 4.9.

glenn20, 2020-08-11:
I can also confirm that my Raspberry Pi 4 also fails to connect to wifi on Hassos 4.12 with Fast Roaming (802.11r) enabled.

It connects fine when I disable 802.11r - reproducibly (3 repeat enable/disable cycles).

My router is a TP-Link Archer C2600 running OpenWrt 19.07.3.

corvy, 2020-08-11:
At least it works fine on Ubiquity Unifi Mesh network now, sad to hear that this is not the case for you gents.

danilogd, 2020-08-13:
I have found the cause of this problem. The driver for Broadcom WiFi on Raspberry Pi kernel, was updated to support 802.11r on release 4.20, this was included on Kernel 5.0+ that HassOs was using until release 4.9. When HassOs moved back to kernel 4.19 it started using a kernel branch that never included 802.11r support (even though this patch was made on 2018). I found the commit on Github that added 802.11r suppport  [https://github.com/raspberrypi/linux/commit/4ad298da9392975f0dc1355dd302a7e95d3c9801#diff-66ea469ce534d8c3ba7147099b87fe78](url) and I guess that the solution would be to ask for this patch to be added to the kernel branch used today by HassOs (4.19.127), however I have no idea how to do this.
Can someone help?

agners, 2020-08-17:
@danilogd interesting find! Not sure how the process is to send the patch back to the RPi community, I guess just a pull-request on Github. But maybe it would be good to give it a try first: I created a rel-4 build with the patch applied for RPi4 32-bit, can you give [this](https://www.temporary-url.com/0496) image a try?

Sources: https://github.com/agners/operating-system/tree/rel-4-fix-wifi

agners, 2020-09-09:
This should be resolved with the 4.13 release (or 5.2).

closed 2020-09-09 by agners