home-assistant/operating-system #737: Crashing

jjanderson, 2020-06-09:
**HassOS release with the issue:**
4.10
Frontend version: 20200519.5 - latest

**Supervisor logs:**

Failed to get supervisor logs, [object Object]

--> This was as far as I could get before the machine became unresponsive.. .it does not even appear in my Unifi Controller client list....

**Journal logs:**

**Kernel logs:**
Failed to get supervisor logs, [object Object]

**Description of problem:**

With this latest build, my HA instance simply keeps crashing. It does not stay up long enough for me to pull proper logs... I have attached what I can... 

I am running HassOS directly on NUC (no VM ware)

I am really sorry that I am not able to provide more info, but with limited access due to craching, this is all I can get... I will however endevour to get more information for you.....

jjanderson, 2020-06-09:
arch | x86_64
-- | --
chassis | embedded
dev | false
docker | true
hassio | true
host_os | HassOS 4.10
installation_type | Home Assistant
os_name | Linux
os_version | 5.4.44
python_version | 3.7.7
supervisor | 227
timezone | Europe/Jersey
version | 0.110.6
virtualenv | false

[multicast.txt](https://github.com/home-assistant/operating-system/files/4752636/multicast.txt)
[audio.txt](https://github.com/home-assistant/operating-system/files/4752637/audio.txt)
[dns.txt](https://github.com/home-assistant/operating-system/files/4752638/dns.txt)
[core.txt](https://github.com/home-assistant/operating-system/files/4752639/core.txt)
[supervisor.txt](https://github.com/home-assistant/operating-system/files/4752640/supervisor.txt)
[host.txt](https://github.com/home-assistant/operating-system/files/4752641/host.txt)
[OZW_Log.txt](https://github.com/home-assistant/operating-system/files/4752642/OZW_Log.txt)
[home-assistant.log](https://github.com/home-assistant/operating-system/files/4752644/home-assistant.log)

jjanderson, 2020-06-09:
I managed to get in to get the above.

It is up and running for now

jjanderson, 2020-06-10:
Crashed again early evening... updated to 0.111 this evening... will see if that holds up

pvizeli, 2020-06-11:
Can you post the journal from the crash? `journal -b -1` should show the logs from boot before

jjanderson, 2020-06-11:
Hi, thanks for getting back to me.

When I type the command into termial (I use the ssh addon - https://github.com/home-assistant/hassio-addons/tree/master/ssh) I get `bash: journal: command not found`

Sorry for the handholding....

jjanderson, 2020-06-20:
Just an update from my side... system seems to now be stable with no crashes for the past week. Not sure if this was due to updates, but all seems ok for now. Thanks

closed 2020-06-22 by balloob

cjramseyer, 2021-01-12:
the command is journalctl, I am having a similar issue.  Seems my NUC (NUC6CAYH) will stay online (HA, and my addons work great) for a little while 6-8 hours?  then poof it falls off the network and nothing on the disply it is attached too
Running the latest version of HASSOS and HA

I would be thrilled for any suggestions to check beyond network, obvious things in the logs, those have been checked and nothing found so far