home-assistant/operating-system #740: HassOS 4.10 fails to boot on Raspberry Pi 4 Model B – 8Gb

alexgoldstone, 2020-06-11:
HassOS release with the issue: 4.10

Using a new Raspberry Pi 4 Model B – 8Gb

Connected to router via ethernet cable and the router supports DHCP

I followed the instructions to write the latest **hassos_rpi4-4.10.img** to a 64GB A2 SD card

I can’t get past the initial boot which continuously cycles through 5 steps:

* A start job is running for Wait Until Kernal Time Synchronised
* A start job is running for HassOS ZRAM tmp
* A start job is running for HassOS ZRAM swap
* A start job is running for HassOS data resizing
* A start job is running for HassOS ZRAM var
* I’ve attached relevant screenshots.

I don’t need a network configuration according to the instructions (happy to use DHCP via ethernet) but also tried with a network config on USB but made no difference.

[image]

[image]

[image]

[image]

[image]

cogneato, 2020-06-12:
I would try a reflash of the image and no usb CONFIG networking. I've flashed a pi4 with 4.10 just now to a 32GB SD without issues.

alexgoldstone, 2020-06-12:
Re-flashed the SD card with 4.10 and didn't insert the USB stick... exactly the same behaviour.

Did you run on an 8GB version of the pi4?

It seems there are differences between the pi4 with 8GB and others:

https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=276186&p=1677662&hilit=8gb#p1677662

> There is a difference in EEPROM configuration (and a revised PCB) that needs up-to-date files for the new model.

pvizeli, 2020-06-12:
> 8GB version of the pi4

Yes, currently not supported until our boot loader (u-boot) support it. Mainly 2-3 month

alexgoldstone, 2020-06-12:
@pvizeli Thanks... at least I know this is not caused by something I have done wrong.

I purchased the pi4 8GB specifically for use with this project (which is frustrating). Can I suggest something is added to the [installation](https://www.home-assistant.io/hassio/installation/) page that states that the 8GB model is currently not supported as it currently looks like any Raspberry Pi 4 Model B will be supported?

cniweb, 2020-06-15:
I have the same issue. When will the new bootloader support the new 8GB pi4?

frenck, 2020-06-15:
@cniweb <https://github.com/home-assistant/operating-system/issues/740#issuecomment-643180443>

mlashley, 2020-07-08:
Bumping to uboot 2020.07 seems to get my build of the -dev branch to boot on the rpi4-8gb

[image]

Changes https://github.com/mlashley/operating-system/commit/f9e7161f0b8b378472cbf3377c0c5e342a9b372c , I can send a PR, but I'm really not sure what else to test first, @pvizeli ? 
(I will build/boot-test 32bit on the rpi4-8gb as well, but I really have none of the other hardware.)

xthedakmanx, 2020-07-09:
> Bumping to uboot 2020.07 seems to get my build of the -dev branch to boot on the rpi4-8gb
> 
> [image]
> 
> Changes [mlashley@f9e7161](https://github.com/mlashley/operating-system/commit/f9e7161f0b8b378472cbf3377c0c5e342a9b372c) , I can send a PR, but I'm really not sure what else to test first, @pvizeli ?
> (I will build/boot-test 32bit on the rpi4-8gb as well, but I really have none of the other hardware.)

Awesome! Here's hoping it gets pushed to a new build soon

mlashley, 2020-07-09:
So the good news is - I was able to take the aarch64 build all the way to running home-assistant, connect to the web-ui and run thru initial setup etc. (I'm not quite ready to move all my devices from my existing pi3 install to test further yet...)
The bad news is - I'm unable to get the 32bit build past a black screen. I guess I need to find a 3v3 tolerant usb-serial adaptor to make any progress there...

erik73, 2020-07-10:
> So the good news is - I was able to take the aarch64 build all the way to running home-assistant, connect to the web-ui and run thru initial setup etc. (I'm not quite ready to move all my devices [...]

Hi mlashley. 

I implemented your changes in my fork. 
I also updated to Buildroot 2020.05, updated to 5.4.50 kernel and updated to a new RPI firmware. 
When building for RPI64 I can even boot from my SSD, using the new RPI4 EEPROM from 2020-06-15. Cool!
32 bit build dos not work, at least not for USB boot. Have not tried it on MC-boot.

Zwordi, 2020-07-15:
Hello @erik73 ,

Regarding your last input you said that you were able to boot on your ssd. 
Regarding #746 , could you say that your fork could resolve this need ?

Where could i test your build on my RPI4-4GB?

Thanks,

TNTLarsn, 2020-07-15:
> Hi mlashley.
>
>
>
> I implemented your changes in my fork.
>
> I also updated to Buildroot 2020.05, updated to 5.4.50 kernel and updated to a new RPI firmware.
>
> When building for RPI64 I can even boot [...]

Could you please describe how to get there! I'm not a programmer but own a RPi 4b 8GB and an SSD and would love to run hass.os on that. Thank you

erik73, 2020-07-15:
@TNTLarsn and @Zwordi. Look at my GitHub page. There is a release there. It has lots of beta/dev stuff in it, and probably a lot of bugs. It is not supported in any way, and the Home Assistant team can NOT be contacted for support.

Zwordi, 2020-07-16:
@erik73 

Thanks for your answer, of course, no warranty as it’s your WIP repository.
Do you think you could help in order to have a PR about the USB/SSD boot on hass #746 ?

Thanks

erik73, 2020-07-16:
> @erik73
> 
> Thanks for your answer, of course, no warranty as it’s your WIP repository.
> Do you think you could help in order to have a PR about the USB/SSD boot on hass #746 ?
> 
> Thanks

Hi.

The problem is that u-boot needs to be updated. Current HassOS has version 2020.01. Some progress was made in the 2020.07 release regarding RPI4 8 GB, but my guess is that we need to wait for u-boot 2020.10 until we have a working USB boot for RPI4. My fork uses the 2020.10 development version.

scrthq, 2020-07-18:
Just got a Pi4 8GB in the mail for HassOS and was bummed to learn about lack of support after the fact, but I'm excited that it appears the support is nearing! Following here for updates 😃

agners, 2020-07-18:
Got my RPi4 8GB today too, and hit the same issue. The serial console capture looks as follows:
[haos-dev-rel-5-crash-rpi4-8gb.log](https://github.com/home-assistant/operating-system/files/4941954/haos-dev-rel-5-crash-rpi4-8gb.log).

I took @mlashley changes as base to migrate to U-Boot 2020.07, and indeed 64-bit build looks good! :tada: 

However, 32-bit failed for me too. I enabled serial console, and it seems that U-Boot freezes really early:

```
U-Boot 2020.07 (Jul 18 2020 - 15:44:29 +0000)

DRAM:  7.9 GiB
<freeze>
```

Not sure what the issue is, I will investigate further.

The 64-bit build with the USB patches applied seems to freeze for me too, but I gues I am missing firmware here?

```
U-Boot 2020.07 (Jul 18 2020 - 16:45:48 +0000)

DRAM:  7.9 GiB
RPI 4 Model B (0xd03114)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: genet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (!SSC)
startin USB...
Bus xhci_pci: probe failed, error -110
No working controllers found
switch to partitions #0, OK
mmc0 is current device
                      "Synchronous Abort" handler, esr 0x96000005
elr: 000000000009049c lr : 00000000000904a8 (reloc)
elr: 000000003b6e49c lr : 000000003b36e4a8
x0 : 00000000ffffffb8 x1 : 000000003b3bb459
x2 : 00000000ffffffb8 x3 : 000000000000000
x4 : 0000000000000000 x5 : 0000000000000000
x6 : 000000003afe5f40 x7 : 000000003afe6290
x8 : 000000000000000 x9 : 0000000000000008
x10: 00000000ffffffd0 x11: 000000000000000a
x12: 0000000000000001 x13: 00000000000189f
x14: 000000003af51810 x15: 0000000000000002
x16: 000000003b3794f4 x17: d39acf9a695765fe
x18: 000000003af5990 x19: 000000003b3bb459
x20: 0000000b00000000 x21: 0000000000000000
x22: 000000003af60bb0 x23: 000000000000000
x24: 0000000000000000 x25: 0000000000000000
x26: 0000000000000000 x27: 000000003afe5f80
x28: 000000003afe62d x29: 000000003af50890

Code: d2800000 a94153f3 a8c27bfd d65f03c0 (f9400280) 
Resetting CPU ...

resetting ...

```

Nowaker, 2020-07-26:
It looks like HassOS 5.0-dev was released 2 days ago: https://github.com/home-assistant/operating-system/releases/tag/5.0

pvizeli, 2020-07-27:
Yes, at least version >= 5.0 RPi4 64bit support it

closed 2020-07-27 by pvizeli

Nowaker, 2020-07-30:
I can confirm - the image not only works, but works with my existing data! Here's what I did. **Backup your SD card first!**

```
losetup -Pf hassos_rpi4-64-5.0.img
for NUM in 1 2 3 4 5 6 7; do pv < /dev/loop0p${NUM} > /dev/sdh${NUM}
```

Partitions 1-7 are HassOS. Partition 8 is data. Flash first 7 partitions to your existing SD and boom, your existing SD is now RPI 4 8GB enabled.

olegmini, 2020-09-20:
Удалось ли решить проблему?