home-assistant/operating-system #745: curl missing support for ldap

Megachip, 2020-06-17:
**HassOS release with the issue:**

all newer ones

**Description of problem:**
As homeassistant started to supporting external auth providers, there was support for authentication against ldap. The default configuration using (lib)curl for that, cause it was included in hassos. But seems since the switch to alpine, it isn't anymore (seems docs are not up2date)

**see also:**
https://www.home-assistant.io/docs/authentication/providers/
https://github.com/home-assistant/supervisor/issues/940
https://github.com/efficiosoft/ldap-auth-sh/issues/4
https://community.home-assistant.io/t/ldap-active-directory/150033

pvizeli, 2020-06-19:
Double post with #940

closed 2020-06-19 by pvizeli

Megachip, 2020-06-19:
issue#940

samsam-rolon, 2021-04-20:
Hello,
I still confirm this issue ! How #940 is linked to this  @pvizeli ?

My HA docker : 2021.4.1
## System Health

version | core-2021.4.1
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 4.4.59+
arch | x86_64
timezone | Europe/Paris

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 0
views | 8
mode | storage

</details>
``` bash
bash-5.0# curl --version
curl 7.69.1 (x86_64-alpine-linux-musl) libcurl/7.69.1 OpenSSL/1.1.1i zlib/1.2.11 nghttp2/1.41.0
Release-Date: 2020-03-11
Protocols: dict file ftp ftps gopher http https imap imaps pop3 pop3s rtsp smb smbs smtp smtps telnet tftp
Features: AsynchDNS HTTP2 HTTPS-proxy IPv6 Largefile libz NTLM NTLM_WB SSL TLS-SRP UnixSockets

```

We can see that LDAP supported by curl is not implemented in  the docker image