home-assistant/operating-system #746: Now RPI4 have a directly Boot to SSD

alesoft73, 2020-06-19:
**HassOS release with the issue:**

HassOS 4.10
**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Hi all,  now is  on line new firmware for RPI4...
When you add this Boot feature on HassOS :-)
Mant THANKS!

condio, 2020-06-21:
I'll chime in and say this is extremely important to me too. USB boot on the PI4 is no longer in beta. It is "stable" and ready to be used. Can we please get USB boot support from HassOS?

Thanks!

mrIngrid, 2020-06-22:
I join for the support of the boot on ssd, it would be an important turning point for HA on RPI4 considering that usb 3.0 can reach speeds up to 620mbyte/s. @pvizeli @balloob @frenck  hurry please 😍. Cheers Guys 💪

frenck, 2020-06-22:
@mrIngrid Please don't mention or tag random people. That is generally not appreciated. Thanks 👍

bogmaerke, 2020-06-22:
Yes, this would be greatly appreciated!

Toni981, 2020-06-22:
Hi...I need a support... Can we please get USB boot support from HassOS?

frenck, 2020-06-22:
Instead of everybody leaving a response with "Yeah I want this too!", please consider not responding and leaven an 👍 on the first post in the issue.

Thanks!

Peter750, 2020-06-25:
Liked, Would give me a zen state of mind knowing that the SD is not used anymore ;-)

LeChuck2909, 2020-07-12:
any comment on this, if it is almost implemented, scheduled, eta, impossible to implement ?

thank you so much.

liamstears, 2020-07-15:
I would also really like to see this feature added, have a new Pi4 and having USB boot would mean having a fast stable system...

agners, 2020-07-19:
Created a merge request which implements USB boot for RPi 4, see #775.

I had to backport quite some patches to make PCIe/USB XHCI support work ontop of U-Boot 2020.07. Also, some changes to the device tree are necessary to make U-Boot properly reset the VL805 USB3 controller.

It does boot fine here, however, the merge request adds quite some patches :-( This is mainly due to backporting needed for U-Boot PCIe/USB XHCI support.

RenzoBeux, 2020-07-23:
Hey agners, sorry i didnt understood, it seems you have added this feature, but is this working on 32bit builds too (asking this cause it say on #775 something like that)? Also, when or how it will be availiable to update? Thank you, noobie here!

jodur, 2020-07-25:
Hey agnes, i posted a issue with the new USB boot.
[https://github.com/home-assistant/operating-system/issues/788](https://github.com/home-assistant/operating-system/issues/788)

Could you also have a look at it?

McGiverGim, 2020-07-28:
@RenzoBeux I can confirm only works for 64bit builds... I've tested both and only 64bit starts.

The documentation provided in the PR that added the usb boot says that only 64bit is compatible.

McGiverGim, 2020-07-29:
If someone is interested in, I needed to return back to the SD version... 

The system has become totally inestable in less than 24h. It seems it is almost totally unresponsive and some things (like zigbee) are unavailable. I can connect with ssh, but is very, very slow, impossible to execute some commands. A power cycle did not fix it. 

I don’t know if it has been the SSD, the 64bit version of Hassio or simply a coincidence, but at this moment I will return back to the SD some time more… I will wait for others experience to see if it was only my system or is a general problem.

tuffelh, 2020-07-30:
I installed yesterday and no issues so far. Will report if something appears.

McGiverGim, 2020-08-03:
It seems that some SSD have problems with the way the raspberry is using it. There is a complete description of the problem and the workaround here: https://www.raspberrypi.org/forums/viewtopic.php?t=245931

My SSD adapter seems to be one of the affected. So now I'm testing the workaround, I will report later if it goes unstable again or this fixes the issue.

scorfman, 2020-08-04:
Installed v5 a couple days ago and am booting from a 250 SSD. Completely normal and lightning fast on my 4gig Pi 4. Welcome to the brave new world. Thanks folks!!

RenzoBeux, 2020-08-04:
Any dowside installing 64bit on a 4gb Rpi4? Asking because of 32 bit is recomended.

bogmaerke, 2020-08-04:
> Any dowside installing 64bit on a 4gb Rpi4? Asking because of 32 bit is recomended.

Well GPIO can't be controlled as far as I know.

McGiverGim, 2020-08-04:
> Any dowside installing 64bit on a 4gb Rpi4? Asking because of 32 bit is recomended.

In my case I have a little more CPU usage, but it can be for different reasons, this are the changes:
- From Hassio 4.11 32 bit to Hassio 5.0 64 bit
- From SD to SSD
- Adding the quirk to make it work

I have observed my CPU from 2% to 4%, and my CPU load 5 minutes from 0.3 to 0.5 more or less... Not a problem, but I have observed the same the two times that I updated.

At least I think I can confirm that the workaround with the quirk made my SSD stable. Or at least after 24h all is working, the first time, without the quirk, the system was unstable with only few use hours.

Peter750, 2020-08-04:
With 64 bit I could not get the razberry V2 to work on gpio port with pi4, 
on 32 bit with SSD no problem

RenzoBeux, 2020-08-04:
But on 32 bit you cant boot from ssd right ?

RenzoBeux, 2020-08-04:
Okay thank you ! @Peter750 i didnt undertood, you are saying you are running hassos on 32 bit with usb boot?

jodur, 2020-08-04:
@McGiverGim, i also running stable for a week ,since usb quirk is applied. Without the quirk i had no stable operation. With the original PiOS i didn’t needed the usb quirk. Maybe a bad implementation of xchi-uas in HassOS?

FaserF, 2020-09-21:
I think the issue can be closed? It works great with the latest beta release of homeassistant os 5.2

loganwedwards, 2020-09-24:
I have been running build 5.1 on a RPi 4 for over a month and it has been flawless. I have restarted a few times and have not noticed any lags on reboot.

maatx, 2020-09-24:
which is the best recommended SSD?

scorfman, 2020-09-24:
> which is the best recommended SSD?

Hard to say as everyone's use cases and specification needs are different. I can tell you what 
I have though. These are connected to a USB 3 port on a pi4:
[ELUTENG USB 3.0 SATA Adapter 2.5 Inch SATA to USB 3.0 Cable](https://www.amazon.com/gp/product/B0716JKJ68)
[Silicon Power 256GB SSD 3D NAND A55 SLC Cache Performance Boost SATA III 2.5" 7mm (0.28") Internal Solid State Drive (SP256GBSS3A55S25)]https://www.amazon.com/dp/B075RJS55D)

carras, 2020-10-05:
> Created a merge request which implements USB boot for RPi 4, see #775.
>
> I had to backport quite some patches to make PCIe/USB XHCI support work ontop of U-Boot 2020.07. Also, some changes to the [...]

Hi @agners, have you consider a bump to (the latest) [U-Boot 2020.10](https://github.com/u-boot/u-boot/tags) to help with booting problems? And is this or it's going to be REL-4 so we can boot from HassOS 4? Thanks.

agners, 2020-10-07:
@carras bump to U-Boot 2020.10 is planned for rel-5. In fact, I have done some testing with 2020.10-rc4 already, I'll need to update and complete that work.

There is no plan to backport to rel-4, but rel-5 eventually will get stable :-)

carras, 2020-10-07:
> @carras bump to U-Boot 2020.10 is planned for rel-5. In fact, I have done some testing with 2020.10-rc4 already, I'll need to update and complete that work.
> 
> There is no plan to backport to rel-4, but rel-5 eventually will get stable :-)

Thanks a lot for you clarification, just waiting to get USB boot for RP4 stable on HassOS.

VajraCool, 2020-10-11:
I have been going through the comments, few things aren't clear to me .. Here is what I have and what I want to do , can someone direct me how can I proceed?

I have a pi4 with 8GB RAM and an External SSD card already. I want to install HASSIO on Docker or if its recommended to use HomeAssistant OS I am fine with it too. I want to do USB-SSD boot instead SD boot. 

What should I do?

Thanks in advance.

FaserF, 2020-10-11:
@VajraCool 
First install the latest raspberry pi firmware using the official raspberry pi OS.
Then Go to the Homeassistant OS github releases page and download the latest 5.X release.
Than as usual plug in the usb drive to a pc, flash the image to this usb device using BalenaEtcher for example.
Plug in the usb drive to the raspberry pi, remove the MicroSD card and boot the raspberry pi.
First boot can take a long time.

VajraCool, 2020-10-11:
> @VajraCool
> First install the latest raspberry pi firmware using the official raspberry pi OS.
> Then Go to the Homeassistant OS github releases page and download the latest 5.X release.
> Than as usual [...]

@FaserF Thank you for your response. To make it more detail for me to understand .. 

 First install the latest raspberry pi firmware using the official raspberry pi OS. --> I will download from https://www.raspberrypi.org/downloads/ 
Then Go to the Homeassistant OS github releases page and download the latest 5.X release. --> I will download from https://github.com/home-assistant/operating-system/releases/download/4.13/hassos_rpi4-64-4.13.img.gz which is for pi4 8GB Than as usual plug in the usb drive to a pc, flash the image to this usb device using BalenaEtcher for example. --> as mentioned here: https://www.home-assistant.io/getting-started/#software-requirements instead of SD card, I will use USB(SSD) and flash the image 

I hope I mentioned all correct, the thing I am not clear is - or the pi OS should I use SD card ? if so after the first boot, should I remove SD card?

my sincere apologies, you did explain me, but I am still confused.

FaserF, 2020-10-11:
@VajraCool that's correct but you have linked the wrong Homeassistant OS.
The one you have linked is the latest 4.X release.
Here is the last 5 release:
https://github.com/home-assistant/operating-system/releases/tag/5.2

The raspberry pi OS should be on the MicroSD card, then install the latest firmware with the raspberry pi OS and enable USB boot.
Afterwards you can delete the raspberry pi OS from the MicroSD again and you have to remove the MicroSD from the pi as you only need the Homeassistant OS on the usb (ssd).

VajraCool, 2020-10-11:
@FaserF Thank you for your response. I will try this in a day or two.

I want to on a route of docker, in that case I will not have HomeAssistant OS instead HASS IO. Any recommendations or points to remember in this case?

carras, 2020-10-11:
> @FaserF Thank you for your response. I will try this in a day or two.
> 
> I want to on a route of docker, in that case I will not have HomeAssistant OS instead HASS IO. Any recommendations or points to remember in this case?

You need to update the EEPROM of your raspberry https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

I recommend you using the raspi-config command.

Then you use the HassOS 5.2 64 or later version. Not 4.xx.

Jpsy, 2020-10-14:
All steps to boot the RPi 4 from SSD including updating Pi's EEPROM are explained in detail great detail here:
https://community.home-assistant.io/t/installing-home-assistant-on-a-rpi-4b-8gb-with-ssd-boot/230948

When you follow that guide you should DEFINITELY use the latest Home Assistant OS version 5.3 (just released 2 days ago). It contains the latest version the boot loader U-Boot 2020.10 which HA OS is based on. This version should remove many problems when booting from USB. I will try it myself in a day or two and will update the linked article.

grilledbacon32, 2020-10-14:
> All steps to boot the RPi 4 from SSD including updating Pi's EEPROM are explained in great detail here:
> https://community.home-assistant.io/t/installing-home-assistant-on-a-rpi-4b-8gb-with-ssd-boot/230948
> 
> When you follow that guide you should DEFINITELY use the latest Home Assistant OS version 5.3 (just released 2 days ago). It contains the latest version of the boot loader U-Boot 2020.10 which HA OS is based on. This version should remove many problems when booting from USB. I will try it myself in a day or two and will update the linked article.

Quick question, I'm currently on 5.2. I'm guessing I need to take a snapshot of my current system, download and flash 5.3 and then import my snapshot? Or is there a way in HA to upgrade to 5.3 that I'm not seeing. Thanks in advance!

Jpsy, 2020-10-14:
> Or is there a way in HA to upgrade to 5.3 that I'm not seeing. 
This is a question I have myself too. I will try the `ha host update` command in HA's shell and will report here and in the article.

McGiverGim, 2020-10-14:
I've updated from 5.1 to 5.2 from the command line without problem some weeks ago.

jvalst, 2020-10-14:
I'll just wait for the official 5.x release. Because when that's out, you can just update the system with the UI right?

Jpsy, 2020-10-14:
> I'll just wait for the official 5.x release. Because when that's out, you can just update the system with the UI right?

Sure!

I just used the shell to update and it worked perfectly:

    ha os update --version 5.3

Before:

    ➜  ~ ha os info
    board: rpi4-64
    boot: A
    version: "5.2"
    version_latest: "4.14"

After:

    ➜  ~ ha os info                
    board: rpi4-64
    boot: B
    version: "5.3"
    version_latest: "4.14"

The `boot: A` and `boot: B` lines are interesting. 
Hope I will find some information about that.

Jpsy, 2020-10-14:
There seems to be absolutely no info about the  `boot: A` / `boot: B` line of `ha os info`  anywhere on the net.
Any ideas?

nicopag, 2020-10-14:
Is it possible to upgrade from 4.13 to 5.3 using that command?

Jpsy, 2020-10-14:
@nicopag, I think so but I would definitely save a snapshot and move it to some safe storage before I try that.

agners, 2020-10-14:
@nicopag as long as the layout of the partitions etc. hasn't been changed, then also update between major versions should work. But I agree with @Jpsy, doing a backup before that definitely make sense...

agners, 2020-10-14:
@Jpsy `boot: A/B ` tells you which boot partition has been used to boot. HAOS uses RAUC, which uses a so called A/B update system: At initial installation (e.g. when using Etcher) the A partition contains the initial image (the version you downloaded). When doing an update, partition B is written and then when rebooting the boot loader tries to boot into partition B. In the case that fails (e.g. do to a bad update), the boot loader has the option to go back and boot partition A. If B works fine, and another update is downloaded, then partition A will be used. So `ha os info` shows you which partition is currently used, and due to the mechanism described above it is perfectly normal that this changes on every update.

Teleportist, 2020-10-27:
I now have this working on 5.3. But only using the 64-bit. Is there any reason why it won't work for the 32-bit?
I was hoping to keep the use of my GPIO pins. If not I guess I'll have to offboard them to a spare ESP.

jodur, 2020-10-27:
GPIO pins are confirmed working under 64 bits with homeassistant now these days.

sanglt, 2020-10-28:
> GPIO pins are confirmed working under 64 bits with homeassistant now these days.

Thanks, I'm use alpha3 + SSD disk and GPIO still didn't work. I will try latest build.

jodur, 2020-10-28:
For feedback on GPIO see:
[https://community.home-assistant.io/t/hass-io-transfer-from-sd-card-to-ssd-or-usb/97452/438](https://community.home-assistant.io/t/hass-io-transfer-from-sd-card-to-ssd-or-usb/97452/438)

[https://community.home-assistant.io/t/hass-io-transfer-from-sd-card-to-ssd-or-usb/97452/444](https://community.home-assistant.io/t/hass-io-transfer-from-sd-card-to-ssd-or-usb/97452/444)

agners, 2020-11-03:
Since USB SSD boot is working now on rel-5 pre-releases closing this issue.

Please create a new issues in case USB SSD do not work with HAOS but do with Raspberry Pi OS.

Ideally also test with a powered hub as i had several times problems with power hungry SSD's directly attached to the RPi.

closed 2020-11-03 by agners