home-assistant/operating-system #754: Missing network drivers in vm image

TobbeJ, 2020-06-28:
**Description of problem:**

Downloaded hassos_ova-4.10 qcow2 image for use with libvirt and setup the network like i always do and found that pretty much most network drivers was missing so it is impossible to get the network up and running.

I use SRIOV assigned card to the vm and therefor need the vf driver for intel network cards.
One of the kernel modules under /lib/modules/kernelversion/kernel/drivers/ethernet/inte that ends with vf.ko

yuvalabou, 2020-07-02:
Thats weird, I just installed it yesturday on proxmox and it recieved network immidiately.

TobbeJ, 2020-07-02:
And you are using SRIOV assigned network card to the vm?

yuvalabou, 2020-07-02:
I have no idea what that means. I have just uploaded it to my machine using whiskerz method..
https://github.com/whiskerz007/proxmox_hassos_install
you may try too :)

TobbeJ, 2020-07-02:
SRIOV capable network card means the physical network card is designed with virtualization in mind where it allows you to assign it multiple times to different VMs.
This allow the vm to get a real physical (kind of) network card to work with, no emulation or anything else in the way.

BUT this works via virtual function driver and this module must be made available in the image or card wont be found (end result no network)
Since this is a prebuilt image i cannot do anything about the kernel modules built.

TobbeJ, 2020-07-04:
If someone can point me in the right direction to where kernel config comes from perhaps i can build my own image with the missing drivers.

TobbeJ, 2020-07-06:
Problem solved with attached patch

```diff
diff --git a/buildroot-external/board/intel/ova/kernel.config b/buildroot-external/board/intel/ova/kernel.config
index 86e28315..433231c3 100644
--- a/buildroot-external/board/intel/ova/kernel.config
+++ b/buildroot-external/board/intel/ova/kernel.config
@@ -38,3 +38,7 @@ CONFIG_HYPERV_STORAGE=y
 CONFIG_HYPERV_NET=y
 CONFIG_HYPERV_VSOCKETS=m
 CONFIG_FB_HYPERV=y
+
+CONFIG_IGBVF=m
+CONFIG_IXGBEVF=m
+CONFIG_I40EVF=m
```

TobbeJ, 2020-09-07:
well... this is an easy fix and would be nice to get it committed

agners, 2020-10-12:
This should be fixed with release 5.3.

closed 2020-10-12 by agners