home-assistant/operating-system #763: Hass.io doesn't boot when flashed on the emmc of the odroid H2+

tidalf, 2020-07-03:
Hello, 

I've tried to flash the intel nuc hass.img on the emmc of an odroid h2+, it doesn't work (stop waiting for root device because of the lack of mmc support).

Also the intel_nuc image doesn't provide the rtl8125 ethernet driver for the ports available in the device.

balloob, 2020-07-03:
ODroid H2 is not currently supported.

tidalf, 2020-07-03:
Hey Balloob,

it seems it's basically a cheap nuc, it works well with the above patch.

balloob, 2020-07-03:
Correct, but we're not currently planning on extending our support for different boards. Each board that we officially adopt will need to be tested on every release that we do, and we lack the resources for that currently. We're hiring (#751) so this might change in the future.

tidalf, 2020-07-03:
Oki !

closed 2020-07-03 by tidalf